weewx/weewx #349: add Evapotranspiration (ET) ( Potential evaporation (PE)) measurements in Weewx standard skin

yannick56, 2018-10-17:
Hello, 

Can you add Evapotranspiration (ET) measurements in a graphic in reports (daily, monthly, yearly ; and weekly if possible) for the Weewx standard skin ?
Maybe in Radiation graph if possible like this : http://www.boock.ch/meteo/evapotranspiration_48h.php

Best regards

tkeffer, 2018-10-17:
The issue list is for reporting and tracking bugs. Please direct your support questions to the [WeeWX User's Group](https://groups.google.com/forum/#!forum/weewx-user).

closed 2018-10-17 by tkeffer

tkeffer, 2018-10-18:
The graph you linked requires an aggregation type of 'cumulative,' which the present version of WeeWX does not have. However, Version 3.9 will. See Issue #302. 

It is unlikely that the Standard skin would ever include such a graph --- it's too specialized. But, once 'cumulative' becomes available, you can easily do it yourself.

yannick56, 2018-10-18:
>The graph you linked requires an aggregation type of 'cumulative,' which the present version of WeeWX >does not have. However, Version 3.9 will. See Issue #302.
OK, Thank you for this information.
Maybe it will be useful that you write documentation about aggregation type of 'cumulative,' with an example for people that want to use it.

>It is unlikely that the Standard skin would ever include such a graph --- it's too specialized. But, once >'cumulative' becomes available, you can easily do it yourself.
OK. I understand.
Then I will create a wiki article for this item like the specific "Monitor Computer Health" in the wiki (  https://github.com/weewx/weewx/wiki/cmon ) for people that need this feature.