weewx/weewx #357: Co/craft branded weather station from Clas Ohlson

gregersn, 2018-11-19:
Picked up, to test, a weather station branded Co/Craft, from Clas Ohlson in Norway 
(https://www.clasohlson.com/no/Tr%C3%A5dl%C3%B8s-v%C3%A6rstasjon-med-USB/36-7334)

Wondered if maybe it looked like something there was a driver for: 

```
Bus 001 Device 003: ID 1130:0829 Tenx Technology, Inc.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x1130 Tenx Technology, Inc.
  idProduct          0x0829
  bcdDevice            1.00
  iManufacturer           1 TENX
  iProduct                2 TMU313X USB R/W64
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      39
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

Anything else I can do to check if it is easily usable?

tkeffer, 2018-11-19:
Please direct your question to the [weewx-user](https://groups.google.com/forum/#!forum/weewx-user) group. The Issues List is for tracking bugs/issues in WeeWX.

closed 2018-11-19 by tkeffer

edombek, 2024-09-26:
Hi! Did anything work?