weewx/weewx #359: Crash on development branch

lylavoie, 2018-11-28:
Been running the development branch for a few days, so I can use the new Essential Values feature for pushing data to weather underground.  System had a crash.  Here are all the debug logs, in case this helps (not filling a specific bug).

Nov 27 01:43:18 raspberrypi weewx[7964]: restx: StationRegistry: wait interval (24300 < 604800) has not passed for record 2018-11-27 01:43:00 EST (1543300980)
Nov 27 01:43:18 raspberrypi weewx[7964]: wmr200: MainThread: D genLoop() phase getting live packets
Nov 27 01:43:18 raspberrypi weewx[7964]: wmr200: MainThread: D   Queuing live packet rx:16191 live_queue_len:1
Nov 27 01:43:18 raspberrypi weewx[7964]: reportengine: Running reports for latest time in the database.
Nov 27 01:43:18 raspberrypi weewx[7964]: reportengine: Running report 'SeasonsReport'
Nov 27 01:43:18 raspberrypi weewx[7964]: restx: PWSWeather: Published record 2018-11-27 01:43:00 EST (1543300980)
Nov 27 01:43:18 raspberrypi weewx[7964]: reportengine: Found configuration file /usr/local/weewx/skins/Seasons/skin.conf for report 'SeasonsReport'
Nov 27 01:43:18 raspberrypi weewx[7964]: restx: Wunderground-PWS: Published record 2018-11-27 01:43:00 EST (1543300980)
Nov 27 01:43:18 raspberrypi weewx[7964]: cheetahgenerator: using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo', 'weewx.
Nov 27 01:43:18 raspberrypi weewx[7964]: manager: Daily summary version is 2.0
Nov 27 01:43:19 raspberrypi weewx[7964]: wmr200: MainThread: D genLoop() Yielding live queued packet id:21962
Nov 27 01:43:20 raspberrypi weewx[7964]: wmr200: MainThread: D   Queuing live packet rx:16192 live_queue_len:1
Nov 27 01:43:21 raspberrypi weewx[7964]: wmr200: MainThread: D genLoop() Yielding live queued packet id:21963
Nov 27 01:43:29 raspberrypi weewx[7964]: cheetahgenerator: Generated 8 files for report SeasonsReport in 11.08 seconds
Nov 27 01:43:29 raspberrypi weewx[7964]: manager: Daily summary version is 2.0
Nov 27 01:43:30 raspberrypi weewx[7964]: wmr200: MainThread: D   Acknowledged control packet rx:5768
Nov 27 01:43:36 raspberrypi weewx[7964]: imagegenerator: Generated 14 images for SeasonsReport in 6.89 seconds
Nov 27 01:43:36 raspberrypi weewx[7964]: copygenerator: copied 0 files to /var/www/html/weewx
Nov 27 01:43:36 raspberrypi weewx[7964]: reportengine: Report 'SmartphoneReport' not enabled. Skipping.
Nov 27 01:43:36 raspberrypi weewx[7964]: reportengine: Report 'MobileReport' not enabled. Skipping.
Nov 27 01:43:36 raspberrypi weewx[7964]: reportengine: Report 'StandardReport' not enabled. Skipping.
Nov 27 01:43:36 raspberrypi weewx[7964]: reportengine: Report 'FTP' not enabled. Skipping.
Nov 27 01:43:36 raspberrypi weewx[7964]: reportengine: Report 'RSYNC' not enabled. Skipping.
Nov 27 01:43:44 raspberrypi weewx[7964]: wmr200: Thread-8: E read_device() USB Error Reason:[Errno 110] Operation timed out
Nov 27 01:43:44 raspberrypi weewx[7964]: wmr200: Thread-8: E USB device read error
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: MainThread: E USB polling thread unexpectedly terminated
Nov 27 01:43:45 raspberrypi weewx[7964]: engine: Main loop exiting. Shutting engine down.
Nov 27 01:43:45 raspberrypi weewx[7964]: engine: Shutting down StdReport thread
Nov 27 01:43:45 raspberrypi weewx[7964]: engine: StdReport thread has been terminated
Nov 27 01:43:45 raspberrypi weewx[7964]: restx: Shut down PWSWeather thread.
Nov 27 01:43:45 raspberrypi weewx[7964]: restx: Shut down Wunderground-PWS thread.
Nov 27 01:43:45 raspberrypi weewx[7964]: restx: Shut down StationRegistry thread.
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: MainThread: I USB polling thread expired
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: Thread-7: I Watchdog received shutdown
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: Thread-7: I Watchdog thread exiting
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: MainThread: I Watchdog thread expired
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: MainThread: I Received packet count live:16192 archive:4 control:5768
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: MainThread: I Received bytes:287224 sent bytes:18376
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: MainThread: I Packet archive queue len:0 live queue len:0
Nov 27 01:43:45 raspberrypi weewx[7964]: wmr200: MainThread: I Driver gracefully exiting
Nov 27 01:43:45 raspberrypi weewx[7964]: engine: Caught WeeWxIOError: USB polling thread unexpectedly terminated
Nov 27 01:43:45 raspberrypi weewx[7964]:     ****  Waiting 60 seconds then retrying...
Nov 27 01:44:45 raspberrypi weewx[7964]: engine: retrying...
Nov 27 01:44:45 raspberrypi weewx[7964]: engine: Using configuration file /usr/local/weewx/weewx.conf
Nov 27 01:44:45 raspberrypi weewx[7964]: engine: Debug is 1
Nov 27 01:44:45 raspberrypi weewx[7964]: engine: Initializing engine
Nov 27 01:44:45 raspberrypi weewx[7964]: engine: Loading station type WMR200 (weewx.drivers.wmr200)
Nov 27 01:44:45 raspberrypi weewx[7964]: wmr200: MainThread: I driver version is 3.3.5
Nov 27 01:44:45 raspberrypi weewx[7964]: wmr200: MainThread: I sensor map is {'outTempBatteryStatus': 'battery_status_out', 'outHumidity': 'humidity_1', 'rainRate': 'rain_rate', 'heatindex4': 'heatindex_5', 'rainTotal': 'rain_total', 'rainBatteryStatus':
Nov 27 01:44:45 raspberrypi weewx[7964]: wmr200: MainThread: I Created watchdog thread to poke for live data every 30 seconds
Nov 27 01:44:45 raspberrypi weewx[7964]: wmr200: MainThread: I Created USB polling thread to read block on device
Nov 27 01:44:45 raspberrypi weewx[7964]: wmr200: Thread-10: I USB polling device thread for live data launched
Nov 27 01:44:45 raspberrypi weewx[7964]: wmr200: MainThread: I Reset console device
Nov 27 01:44:45 raspberrypi weewx[7964]: wmr200: Thread-10: I USB polling device thread signaled to start
Nov 27 01:44:46 raspberrypi weewx[7964]: wmr200: MainThread: W Restart count: 5
Nov 27 01:44:46 raspberrypi weewx[7964]: wmr200: MainThread: D Configuration setup
Nov 27 01:44:46 raspberrypi weewx[7964]: wmr200: MainThread: D   Log sensor faults: True
Nov 27 01:44:46 raspberrypi weewx[7964]: wmr200: MainThread: D   Using PC Time: True
Nov 27 01:44:46 raspberrypi weewx[7964]: wmr200: MainThread: D   Erase archive data: True
Nov 27 01:44:46 raspberrypi weewx[7964]: wmr200: MainThread: D   Archive interval: 60
Nov 27 01:44:46 raspberrypi weewx[7964]: wmr200: MainThread: D   Archive threshold: 604800
Nov 27 01:44:46 raspberrypi weewx[7964]: engine: Loading service weewx.engine.StdTimeSynch
Nov 27 01:44:46 raspberrypi weewx[7964]: engine: Finished loading service weewx.engine.StdTimeSynch
Nov 27 01:44:46 raspberrypi weewx[7964]: engine: Loading service weewx.engine.StdConvert
Nov 27 01:44:46 raspberrypi weewx[7964]: engine: StdConvert target unit is 0x1
Nov 27 01:44:46 raspberrypi weewx[7964]: engine: Finished loading service weewx.engine.StdConvert
Nov 27 01:44:46 raspberrypi weewx[7964]: engine: Loading service weewx.engine.StdCalibrate
Nov 27 01:44:46 raspberrypi weewx[7964]: wmr200: Thread-9: I Started watchdog thread live data
[... 72 lines cut ...]
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****  'str' object is not callable
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****  Traceback (most recent call last):
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****    File "/usr/local/weewx/bin/weewx/engine.py", line 886, in main
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****      engine.run()
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****    File "/usr/local/weewx/bin/weewx/engine.py", line 160, in run
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****      self.dispatchEvent(weewx.Event(weewx.STARTUP))
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****    File "/usr/local/weewx/bin/weewx/engine.py", line 224, in dispatchEvent
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****      callback(event)
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****    File "/usr/local/weewx/bin/weewx/engine.py", line 516, in startup
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****      self._catchup(self.engine.console.genStartupRecords)
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****    File "/usr/local/weewx/bin/weewx/engine.py", line 631, in _catchup
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****      origin='hardware'))
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****    File "/usr/local/weewx/bin/weewx/engine.py", line 224, in dispatchEvent
[... cut ...]
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****      _weight = self._calc_weight(record)
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****    File "/usr/local/weewx/bin/weewx/manager.py", line 1582, in _calc_weight
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****      raise ValueError("Non-positive value for record field 'interval': %s" (record['interval'], ))
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****  TypeError: 'str' object is not callable
Nov 27 01:44:51 raspberrypi weewx[7964]:     ****  Exiting.
lines 16732-16774/16774 (END)

tkeffer, 2018-11-29:
There are two different things going on here.

First, there is a formatting error when creating the error message. This is what caused the `TypeError` exception. This has been fixed in commit https://github.com/weewx/weewx/commit/d47142b779a2ad68e0f01fca6f9f84cbe4221cda

The second problem is that, somehow, the `interval` field was either zero, or negative. We don't know which because of problem number one.

I suspect the problem is related to the very big "clock drift" reported on 01:44:50.

Does this error occur on every startup? Or, just once?

lylavoie, 2018-11-29:
So far, I've only seen this once, it ran for a day or so, then I noticed data wasn't being reported to Wunderground, so checked it and saw the error.   I'll pull the updated version and see if the issue arises again.  

I'm surprised about the drift, the Rasp Pi has ntp configured and operational and the station should be pulling time from the NIST WWVB radio clock.  I'll dig into that issue a little bit.  I also see a difference in the hardware archive interval and the 300 seconds configured in weewx.

tkeffer, 2018-12-05:
Anything new on this?

tkeffer, 2018-12-06:
The only patch I've done is one that fixes the error message. It didn't fix the underlying problem.

Still, I'll close the issue. We can open it again if it reoccurs.

closed 2018-12-06 by tkeffer