weewx/weewx #363: calcMissing() from weeimport should be a core wee_database function

poblabs, 2019-01-24:
One of the number one questions I get from the Belchertown skin is how to add `appTemp` to the schema. I understand the User Guide has this info, but I ended up [writing my own wiki](https://github.com/poblabs/weewx-belchertown/wiki/Adding-a-new-observation-type-to-the-WeeWX-database) focused specifically on `appTemp` and `windrun` so users can quickly copy/paste and get started. 

The 2nd question I get is how to backfill the appTemp data. 

You could export the database to CSV, import it to a second weewx system using `wee_import --calc-missing`, then export that new database again as CSV, trim it to the missing data that's needed, then run an UPDATE on the database with that CSV to add those new values for the `dateTime` record. Not the best.

It would be nice if `wee_database` had a similar feature of `--calc-missing` where it can calculate and back-fill the fields that are referenced in wee_import like appTemp, cloudbase, humidex, maxSolarRad, windrun. 

I've started some work on this (added the option to `wee_database` but need to update `manager.py` next I think), but wanted to get an appetite for the feature from the group before I invested time on it. This is new weewx territory for me.

biltmorelaker, 2019-01-24:
Pat,
The reason for back-filling is to display accurate results when showing record values (high, low, etc.) over a time period prior to the recording of some of these values. As you know, back-filling my appTemp data is something that is important to me. I'd much prefer your proposed option. The manual method you noted is pretty ugly.

tkeffer, 2019-01-25:
I assume we include only observation types that `StdWXCalculate` knows about?

poblabs, 2019-01-25:
I think that makes sense. My shoot-from-the-hip observations were those referenced from the comments of `wee_import`. 

It looks like `do_calculations` would only run through the observation types that `StdWXCalculate` knows about.

I'm still tackling the learning curve on figuring out how to methodically get records, then run them through `wxcalculate.do_calculations(record, 'archive')`, then back to the archive with `archive.updateValue(dateTime, obs, value)`. I know the bits and pieces I need - don't know how to get there yet.

tkeffer, 2019-01-25:
Frankly, I would make `--calc-missing` the default.

poblabs, 2019-01-25:
On `wee_database`?

gjr80, 2019-01-25:
As an aside we had the need for something similar with WeeWX-WD when we went to using a separate WeeWX-WD database when WeeWX 3.0.0 was released. This was before the innards of `StdWXCalculate` were as accessible as they are now (in fact I think it was before `StdWXCalculate` knew about `appTemp` and `humidex`. Was a very straight forward bit of code that should be even easier now with access to `WXCalculate`. I had a look back at the WeeWX-WD utility but it is not of much use, too different a context and too many things have changed since.

matthewwall, 2019-01-25:
> One of the number one questions I get from the Belchertown skin is how to add appTemp to the schema. I understand the User Guide has this info, but I ended up writing my own wiki focused specifically on appTemp and windrun so users can quickly copy/paste and get started.

We have had a wview-extended schema for a long time now, but it has not made it to the master branch

Other schemas we might consider are a power monitoring schema, or an all-in-one schema (power and weather)

tkeffer, 2019-04-08:
Pat, do you have a PR for this?

poblabs, 2019-04-08:
@tkeffer no, but I can put one together. Things have changed a bit in the last 3 months. Do you want it against the python3 branch or against the master branch?

tkeffer, 2019-04-08:
It's a new feature, so it should be against `development`.

poblabs, 2019-04-08:
FYI - #394 submitted

tkeffer, 2019-11-27:
Any reason why we can't close this issue?

poblabs, 2019-11-27:
I thought this was closed because #443 was addressing it. If that's true
then yes this can be closed.

Sent from mobile

closed 2019-11-27 by tkeffer