weewx/weewx #367: Allow the base to be specified in heating and cooling degree-days

tkeffer, 2019-02-02:
The idea is to allow tags like this:

```
$month.cooldeg(base=(65.0, 'degree_F')).sum
```
This will allow the tag to be used to calculate growing days.

See email thread https://groups.google.com/forum/#!topic/weewx-development/QxcqE-7Yd38

n7qnm, 2019-02-03:
Also been thinking about this, forgive the newbie question; but what I think you're saying is that we can have tags that would calculate any "degree days" with any base.   I'll take a look at that and see if  can figure out how to do that.

Then we could take the cooling and heating days out of the "base" code.   Also, turns out there are crop specific growing days that use different bases.

tkeffer, 2019-02-04:
Yes, that's the idea.

n7qnm, 2019-02-09:
Took a look at this - not quite sure how this would work.  Degree days seems to be a different kind of aggregation (not really a sum) - right now it's defined as a "synthetic stats type".  Not sure how we could do this without introducing a new aggregation.

tkeffer, 2019-02-09:
It would definitely require some Python-fu!

Make sure you review the section [*How tags work*](http://weewx.com/docs/customizing.htm#how_tags_work) in the Customizing Guide.

Take the tag sequence

```
$month.cooldeg.sum
```

The first part, `$month`, will create a `TimespanBinder` object. Cheetah will then use the `__getattr__()` on that object to evaluate `$month.cooldeg`. However, if we change it to a function call

```
$month.cooldeg(base=mybase).sum
```

that won't work, because `cooldeg` is no longer an attribute --- it's a function. So, you have to use a `__call__` method.

Using a broad brush, that's the idea, I think. This is off the top of my head.

n7qnm, 2019-02-09:
Yeah – but, the base for any given “degree day” calc doesn’t typically change.  There are three “major” types of degree days, cooling, heating and growing.   Cooing and heating actually use the same base, 65F, heating is degrees below, cooling is degrees above.  Growing is the outlying one and is normally 50F.   So, not sure we really need to embed the base in the tag definition, just in the method.    I looked at the How tags work section and it kinda made my head hurt; I need to play with it a bit.

tkeffer, 2019-06-26:
I experimented with allowing inline specifications, but decided it was too complicated to solve what was really a simple problem.

So, I simply followed your solution and added a new aggregation, `growdeg`. 

Commit https://github.com/weewx/weewx/commit/8cd9413818f2b05cc7aa52c160561ca10ee0ef71.

To appear in WeeWX V4.

closed 2019-06-26 by tkeffer