weewx/weewx #368: Rethink catchup strategy on start up of VPro2

tkeffer, 2019-02-05:
V3.9.0 introduced a [change](https://github.com/weewx/weewx/commit/6e7f3aac1b2926cacfa63229db2dbbd55f3f2cab#diff-5c5fa8bd3042a3e3815da70f6a58dd4e) where if software record generation is specified, it would not do a hardware catchup on startup. Is this the right strategy?

matthewwall, 2019-02-05:
this breaks expected behavior for any hardware with a logger.  the expectation has always been that the logger would be read on startup, then record generation would be either software or hardware, depending on which was specified.  for example, the recommended configuration for fine offset usb stations is to use record_generation=software, because hardware record generation is unreliable.  but reading records at startup is a good thing.  ws28xx stations are similar - you want software record generation during normal operation, but read the logger at startup.  i think the rainwise is that way too.

i think we need a 3.9.1 to revert this change.

what scenario instigated this change?

tkeffer, 2019-02-06:
It's not that bad. If you've specified hardware record generation, it won't do a catch up on startup, but it will at the end of the first archive interval.

So, why did I do this? I'll be damned if I remember. Seems like a bad idea now. Nevertheless, on the assumption that there may be someone out there who wants to avoid a hardware catch up, I've created an (undocumented) option `no_catchup`. If set to `True`, a startup catch up will not be attempted. Commit https://github.com/weewx/weewx/commit/8929d365ab5f878671bec8ca3a62c77f2e156f99

closed 2019-02-06 by tkeffer