weewx/weewx #369: Almanac labels from skin.conf ignored

stimpy23, 2019-02-09:
Hi!
Since v3.9, the almanac labels from my skin.conf are being ignored.
My definitions look just like in the skin.conf of your Smartphone-skin, except I translated them:

> [Almanac]
>     moon_phases = Neumond, Zunehmend (Sichel), Zunehmend (viertel), Zunehmend (halb), Vollmond, Abnehmend (halb), Abnehmend (viertel), Abnehmend (viertel)

It worked fine until v3.8.x and since 3.9, it's ignored and the original english labels are used, no matter what I defined in my skin.conf.

Here's my complete skin.conf:
https://gitlab.ambhost.net/stimpy/template_rabenwetter/blob/master/dist/i18n/de/skin.conf

And here's what it looks like in action:
https://www.rabenwetter.de/

Cheers,
  stimpy

matthewwall, 2019-02-09:
this is a result of a change to how skin parameters are defined and overridden.  the preferred approach is to set the values in weewx.conf, not in skin.conf.  that way you can do updates to a skin without losing your customizations.  however, if you make the changes in skin.conf, then you must ensure that there are no other values defined in weewx.conf that override them.

(this was not a capricious decision - we thought about many alternatives and implemented a few of them to test everything.  but this approach should lead to better i18n/l10n capabilities in one of the next weewx releases.)

please see the upgrade guide:

http://weewx.com/docs/upgrading.htm#From_V3.8_or_earlier

also, please see the i18n/l10n section of the user guide:

http://weewx.com/docs/customizing.htm#localization

your feedback would be greatly appreciated, and will help make the next release of weewx even better.

closed 2019-02-09 by tkeffer

stimpy23, 2019-02-10:
This even kinda makes sense.. Awesome, thanks! :)