weewx/weewx #370: log_success/log_failure is not working as designed

matthewwall, 2019-02-10:
while testing v3.9.1, i find that log_success and log_failure directives seem to work for restful services, but they are not working for report services, specifically the generators (imagegenerator, cheetahgenerator), nor are they working for the archive service.

i should be able to specify this in weewx.conf:

```
log_success = false
log_failure = true
```

to disable all output unless there is a failure.  but i still get this every archive interval:

```
Feb 10 09:10:26 7up weewx-sdr[10040]: manager: Added record 2019-02-10 09:10:00 EST (1549807800) to database 'weewx.sdb'
Feb 10 09:10:26 7up weewx-sdr[10040]: manager: Added record 2019-02-10 09:10:00 EST (1549807800) to daily summary in 'weewx.sdb'
Feb 10 09:10:30 7up weewx-sdr[10040]: imagegenerator: Generated 15 images for 4seasons in 2.04 seconds
Feb 10 09:10:30 7up weewx-sdr[10040]: copygenerator: copied 0 files to /var/www/html/4seasons
```

and i should be able to specify like this:

```
[StdReport]
    log_success = false
    log_failure = true
    ...

[StdArchive]
    log_success = false
    log_failure = true
    ...
```

but i still see the output from archive service and generators.

the specific use case is for some almost-embedded-systems - i want `log_success = true` while we configure and diagnose systems, but as soon as things are running properly we want `log_success = false` for EVERYTHING.  the log partitions are fairly small and we don't want useless chat filling up the partitions.

gjr80, 2019-02-10:
Noticed this myself last week when trying to cut down my logs. I have coded the changes and believe they are well behaved under the new override hierarchy. Will submit a PR later today if you want Matthew?

poblabs, 2019-02-27:
I'm seeing this with a restful service - specifically the MQTT extension. 

I have the following in weewx.conf:

```
# Whether to log successful operations
log_success = False

# Whether to log unsuccessful operations
log_failure = True

[StdReport]
    # Whether to log a successful operation
    log_success = False

    # Whether to log an unsuccessful operation
    log_failure = True
```

However in my syslog I see the following:

```
Feb 27 11:30:31 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:31 EST (1551285031)
Feb 27 11:30:33 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:34 EST (1551285034)
Feb 27 11:30:36 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:36 EST (1551285036)
Feb 27 11:30:39 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:39 EST (1551285039)
Feb 27 11:30:41 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:42 EST (1551285042)
Feb 27 11:30:44 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:44 EST (1551285044)
Feb 27 11:30:45 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:46 EST (1551285046)
Feb 27 11:30:46 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:47 EST (1551285047)
Feb 27 11:30:49 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:49 EST (1551285049)
Feb 27 11:30:51 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:52 EST (1551285052)
Feb 27 11:30:54 weewx weewx[20491]: restx: MQTT: Published record 2019-02-27 11:30:54 EST (1551285054)
```

tkeffer, 2019-02-27:
PR?

poblabs, 2019-02-27:
I'm not sure of a fix. Also don't know if it's isolated to the extension.

matthewwall, 2019-02-27:
as of weewx-mqtt v0.18:

the only logging specific to weewx-mqtt is when there is a failure, and in these cases it does not respect the log_success/log_failure parameters.

so i consider that part a bug in the weewx-mqtt extension, although it would be nice if any uploader could invoke a log function on self so it does not have to worry about the log_success/log_failure settings.  perhaps something like self.logmsg(xxx) instead of logmsg(xxx)?

the messages 'Published record ...' are coming from the superclass, so that would be a bug in RESTThread

gary fixed only the generators, not the restx stuff

poblabs, 2019-02-27:
Thanks for the insight. In the original message you mentioned "log_success and log_failure directives seem to work for restful services". This seems true still. I have other restful services enabled and am only receiving that "restx" log message from the weewx-mqtt v0.18 extension.

matthewwall, 2019-02-27:
the bug is entirely in weewx-mqtt

there is a log_success and log_failure member on the RESTThread, and weewx-mqtt ignores it - it neglects to set it in the ctor, and it neglects to use it in the uploading methods.

this is probably an issue for every uploader i have written...

poblabs, 2019-02-27:
Got it, thanks! I can take that offline since it's with the extension. Also to clarify something I said earlier, the other restx services are also ignoring the log_success setting. I hadn't noticed this previously because the MQTT extension was quite noisy. 😃

tkeffer, 2019-04-08:
Seeing as how this is an issue for extensions that do uploads, but not the core WeeWX uploaders, I'm closing this issue.

closed 2019-04-08 by tkeffer