weewx/weewx #372: Running ADS-WS1 getting error 'unexpected buffer length'

jezkerwin, 2019-02-11:
I'm running a Argent Data ADS-WS1 weather station on a raspberry Pi.

When looking at /var/log/messages, I see the following errors and then after a while the engine shuts down.
Occasionally a successful reading happens, but for the most post it's these messages.

`Feb 11 21:11:04 weewx[6897]: ws1: Failed attempt 1 of 5 to get readings: Unexpected buffer length 51
Feb 11 21:11:31 weewx[6897]: ws1: Failed attempt 2 of 5 to get readings: Unexpected buffer length 594
Feb 11 21:12:11 weewx[6897]: ws1: Failed attempt 1 of 5 to get readings: Unexpected buffer length 730
Feb 11 21:12:21 weewx[6897]: ws1: Failed attempt 2 of 5 to get readings: Unexpected buffer length 102
`

Does anyone know what this means, and how to fix?
Thanks.

jezkerwin, 2019-02-11:
Here is an excerpt from /var/log/syslog

```
Feb 11 21:18:09 . weewx[7001]: restx: AWEKAS: Posting not enabled.
Feb 11 21:18:09 . weewx[7001]: engine: Finished loading service weewx.restx.StdAWEKAS
Feb 11 21:18:09 . weewx[7001]: engine: Loading service weewx.engine.StdPrint
Feb 11 21:18:09 . weewx[7001]: engine: Finished loading service weewx.engine.StdPrint
Feb 11 21:18:09 . weewx[7001]: engine: Loading service weewx.engine.StdReport
Feb 11 21:18:09 . weewx[7001]: engine: Finished loading service weewx.engine.StdReport
Feb 11 21:18:09 . weewx[7001]: engine: Starting up weewx version 3.9.1
Feb 11 21:18:09 . weewx[7001]: engine: Station does not support reading the time
Feb 11 21:18:09 . weewx[7001]: engine: Starting main packet loop.
Feb 11 21:18:09 . weewx[7001]: manager: Daily summary version is 2.0
Feb 11 21:18:38 . weewx[7001]: ws1: bytes: '21 21 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 33 34 37 30 33 34 38 30 33 34 38 30 30 34 34 34 30 38 33 30 37 30 33 37 3C 36 3C 0D EA 21 21 30 30 30 37 30 30 46 44 30 30 30 30 30 30 30 34 32 32 41 42 3C 33 34 37 30 33 34 38 30 33 36 3C 30 3C 3B 37 36 3C 36 3B 37 37 38 B3 3F 3C 37 39 ED DB 21 21 30 30 30 37 30 30 46 44 30 30 30 30 30 30 30 34 32 32 41 61 30 33 34 37 38 33 34 3C 30 33 37 3E 34 3C 37 37 37 34 B6 31 3E 36 3C B3 FB FB 8A 6A 4F 21 21 30 30 30 39 30 30 46 44 30 30 30 38 38 30 30 36 32 32 41 72 36 33 37 37 3C 33 37 3B 34 BB B7 39 32 36 BF FA 9A FA 87 FA BB AA DA FF FF FF EA 7B FD 21 21 30 30 30 39 30 30 46 44 30 30 38 30 38 30 30 36 3A 3A 61 52 3B 33 37 37 36 33 53 B2 DA BB FB 92 BA FE FA 9A B2 FA FF FF FF DA DA BF FB EA EB 21 21 30 30 30 43 30 30 46 44 30 30 3A 32 36 38 34 34 3B 3B 71 7A 3F 3B 36 37 B7 3B 4D E2 DA BB EB 92 BA FE E7 E7 AB DA FF FF F3 9B DB FB FF 9A FE 21 21 30 30 30 43 30 30 46 64 30 30 37 34 3B 34 3C 9B A7 A7 E9 F5 B7 D3 D3 9F AB EF C2 DA B3 FF AF BA FE FF FF A3 FA FF FF FB DF F6 FF 21 21 30 30 30 43 30 30 46 44 30 30 3E 38 3A 34 3C 36 3B 3A 59 52 DF BB BA BA 9B BA FB C2 DE A7 FF CF BA FB FB FB AB FA FF 3F FF DF FB 21 21 30 30 30 45 30 3C 66 46 32 34 3B 37 3C 32 3C DB D6 8B E9 BF B7 DB D3 DF FF CF FA DE F3 E3 8A FF FF BF FA FF DA D3 FF 21 21 30 38 38 45 30 34 66 46 33 3C 3E 33 39 37 3E FB DF 6B ED 3F 76 D3 93 BF FB FA DA 9F EA FF FF DF FF 21 21 30 34 31 32 30 38 47 77 37 3C BF 4D CB D2 BB DB DB EB CD DF FF BB F6 BF FF F2 DF FF FF FF FE FF 21 21 30 34 34 45 30 3A 66 45 37 3C 3F DF 9D 9F B6 D7 DE 8B C9 DF FB BB EE E7 FF FB DE F7 FF FB 21 21 30 3C 31 36 38 3E 67 47 35 3C BF BB BB 3B 36 DF DF EA ED DB BF BB FA FF FF FF FE FF FE FF FF FF 21 21 30 38 36 65 38 3E 47 77 3E 3C BF F6 BF F2 FE DF DB EB ED FA F3 BB DE FF BF DE FB FF BF 21 21 30 3B 39 3B 38 B8 67 77 3F 36 BF DF FF BA BA FB DF CF ED FF 73 BB DB BF 21 21 38 3F 3D 3F 34 3A F7 F7 BA FA BB F3 FB BA FB FF FF 9F F5 7B FF DF FF 21 21 30 3E 31 3C 37 38 DF FF 3B BF FF FB AB F7 EB D3 FF DF 21 31 38 3C 4F BA B2 BE 77 AB 8A FF FF FF DF FE 4B FF EE FF F7 21 39 34 3E EF 9B B2 B2 CF BF FA FF BF 5F 21 31 30 38 6F DB BA B2 FF FA F6 BB FF'
Feb 11 21:18:38 . weewx[7001]: ws1: Failed attempt 1 of 5 to get readings: Unexpected buffer length 730
Feb 11 21:18:48 . weewx[7001]: ws1: bytes: '21 21 30 30 31 35 30 30 46 43 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 38 30 30 30 30 30 30 30 30 30 30 30 33 30 30 30 41 0D 0A'
Feb 11 21:18:48 . weewx[7001]: ws1: bytes: '21 21 30 30 31 33 30 30 46 43 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 38 30 30 30 30 30 30 30 30 30 30 30 33 30 30 30 42 0D 0A'
Feb 11 21:18:48 . weewx[7001]: ws1: bytes: '21 21 30 30 31 33 30 30 46 44 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 38 30 30 30 30 30 30 30 30 30 30 30 33 30 30 30 42 0D 0A'
Feb 11 21:18:48 . weewx[7001]: ws1: bytes: '21 21 30 30 31 33 30 30 46 44 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 38 30 30 30 30 30 30 30 30 30 30 30 33 30 30 30 42 0D 0A'
Feb 11 21:18:48 . weewx[7001]: ws1: bytes: '21 21 30 30 31 33 30 30 46 44 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 38 30 30 30 30 30 30 30 30 30 30 30 33 30 30 30 43 0D 0A'
Feb 11 21:18:48 . weewx[7001]: ws1: bytes: '21 21 30 30 31 30 30 30 46 44 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 38 30 30 30 30 30 30 30 30 30 30 30 33 30 30 30 43 0D 0A'
Feb 11 21:18:48 . weewx[7001]: ws1: bytes: '21 21 30 30 31 30 30 30 46 44 30 30 30 30 30 30 30 34 32 32 41 41 30 33 34 37 30 33 34 38 30 33 34 38 30 30 30 30 30 30 30 30 30 30 30 33 30 30 30 43 0D 0A'
Feb 11 21:18:48 . weewx[7001]: ws1: bytes: '21 21 30 30 30 45 30 30 46 44 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 38 30 30 30 30 30 30 30 30 30 30 30 33 38 38 30 43 0D 0A'
Feb 11 21:19:18 . weewx[7001]: ws1: bytes: '21 21 30 30 30 45 30 30 30 30 30 30 30 30 30 30 30 34 32 32 41 41 30 33 34 37 30 33 34 38 30 33 34 38 30 30 34 30 3C 30 38 36 36 36 30 33 34 36 34 43 8D CA 21 21 30 30 30 45 30 30 30 30 30 30 30 30 30 30 30 34 32 32 41 42 30 33 34 37 30 33 34 38 30 33 34 38 30 30 36 36 36 38 38 34 39 3A 30 33 3B 3B 37 43 ED EB 21 21 30 30 30 45 30 30 30 30 30 30 30 30 30 30 30 34 32 32 41 41 36 33 34 37 30 33 36 3A 30 33 35 3C 32 38 3C 37 37 33 34 3F 3A 3A 37 B3 FB F3 3A 35 AB 21 21 30 30 30 43 30 30 30 30 30 30 30 30 30 30 30 34 32 32 41 42 3C 33 34 37 30 33 36 3C 3C 33 34 3F 36 34 3E 3B 37 3B 36 BF B3 3F 99 A6 FB D2 23 35 EF 21 21 30 30 30 43 30 30 46 44 30 30 30 38 3C 3C 30 34 3A 3A 51 42 3C 33 37 37  [... cut ...]
Feb 11 21:20:20 . weewx[7001]: copygenerator: copied 5 files to /var/www/html/weewx
Feb 11 21:20:20 . weewx[7001]: reportengine: Report 'SmartphoneReport' not enabled. Skipping.
Feb 11 21:20:20 . weewx[7001]: reportengine: Report 'MobileReport' not enabled. Skipping.
Feb 11 21:20:20 . weewx[7001]: reportengine: Report 'StandardReport' not enabled. Skipping.
Feb 11 21:20:20 . weewx[7001]: reportengine: Report 'FTP' not enabled. Skipping.
Feb 11 21:20:20 . weewx[7001]: reportengine: Report 'RSYNC' not enabled. Skipping.
```

windcrusader, 2019-04-08:
I've seen similar issues from time to time on my WS1 also interfacing through a raspberry pi model B. 

I had a difficult time getting reliable serial interfacing with the unit. Initially I tried a cheap USB to serial converter and this didn't work at all. Eventually, I succeeded by building my own direct to raspberry pi GPIO UART interface using a MAX232 chip. This was a fun electronics project, but quite a bit of effort to go to for just getting a serial interface going. My connection is now pretty reliable though. I understand people have had more success with the USB serial converters based on the FDTI chip.

I think occasionally you also just get bad loop packets over the serial interface. For me, this seems to occur most often during startup or when trying to alter the unit parameters via the command prompt over serial interface (and being too lazy to shut weewx down during the process).

Have you debugged your serial connection e.g tried a different cable, shorter cable, etc etc?

I think this issue is not a weewx issue though, and more of a hardware issue. I would recommend the issue is closed. Agree?

closed 2019-04-08 by jezkerwin