weewx/weewx #374: Support Vantage LOOP2 loop packet

futzle, 2019-02-12:
Vantage consoles with firmware 1.90 or later support two types of loop packet:

- LOOP, which Vantage consoles have already supported, and which weewx has always handled.

- LOOP2, which omits some measurements (like extra*, leaf wetness, soil moisture) but includes some other measurements not in LOOP (wind chill, heat index, dewpoint, Temperature-Humidity-Sun-Wind index).  Currently some of these values are computed in software by weewx (wind chill, heat index, dewpoint), and some are not available at all (THSW index).

Support the LOOP2 packet for consoles which can produce it, and for users who want to use it.

This needs to be an opt-in feature for users because it will affect behaviour:

- Because the packet types alternate, LOOP packets will come in only every 5 seconds, not every 2.5 seconds.  This may be of concern to rapid-fire implementations.
- Rain bucket tips in the first packet of the month are lost because of the implementation; this may become the first two packets if LOOP2 is also sent.
- If prefer_hardware is set for wind chill, heat index and dewpoint (the default), these records will come from the LOOP2 packet and not be generated in software.  (Archive packets during catchup still need to be generated in software.)

Discussed in weewx-development list topic: https://groups.google.com/forum/#!topic/weewx-development/7wOOKjRnIII

futzle, 2019-02-12:
Affected by #115: there are new field values like THSW index.  What to call them in the schema?

ls4096, 2019-07-12:
I didn't see it mentioned already, but one more benefit to the Vantage LOOP2 packets is that they include both absolute sensor pressure and altimeter pressure, both reported by the hardware; it seems that currently for Vantage the sensor pressure and altimeter are calculated in software from other reported values.

tkeffer, 2019-08-21:
Added support for LOOP2 packets in the `development` branch. 

There is a new option in the `[Vantage]` stanza:

```ini
[Vantage]
  ... (as before)
  packet_request = 1 # 1=LOOP1, 2=LOOP2, 3=both
```

Setting `packet_request=3` will have the driver alternate between type 1 and type 2 packets. For variable types like `dewpoint`, this will cause the value to flip back and forth between the value in the type 2 packet and a value calculated in software. So, for these variables, you should specify `hardware` in `[StdWXCalculate]`.

Testers are welcome!!

tkeffer, 2019-08-22:
Changed name of option from `packet_request` to `loop_request`.

Documented it in the User's Guide.

tkeffer, 2019-08-22:
Closing this issue, but I really encourage anyone interested to give LOOP2 packets a try! We need to test this.

closed 2019-08-22 by tkeffer