weewx/weewx #375: Non-positive value for record field 'interval': 0 on WMR200

SuddenlyQ, 2019-02-15:
I fixed this by changing the value `if timestamp_packet_interval <= 0`  in /usr/share/weewx/weewx/drivers/wmr200.py from 0 to 1.

tkeffer, 2019-02-15:
Thanks, but I have 2 questions:

1. What problem did this fix? Were you having duplicate records? Something else?

2. The original line of code rejected values less than zero (not less than or equal to zero). What does the final code look like? This?

   ```python
                if timestamp_packet_interval <= 1:
                    loginf(('genStartup() Discarding received archive record that presented out-of-order; '
                            'current timestamp:%s; previous timestamp:%s')
                           % (weeutil.weeutil.timestamp_to_string(timestamp_packet_current),
                              weeutil.weeutil.timestamp_to_string(timestamp_packet_previous)))
   ```

SuddenlyQ, 2019-02-15:
1. weeWX was failing with an error about non-postive values for interval. I believe I changed it to be < 1 or <= 0.

tkeffer, 2019-02-15:
My apologies, but I'm still not understanding. If it encounters a non-positive interval, WeeWX should just reject it, not fail. Could you please post a system log where it was failing?

And, I would appreciate it if you could be more specific whether the comparison is

```python
if timestamp_packet_interval <= 0`
```
or
```python
if timestamp_packet_interval < 1`
```

SuddenlyQ, 2019-02-15:
Here's a syslog:
```
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: MainThread: I Using pc time adjusting archive record time by 38 sec 2018-01-26 20:51:00 PST (1517028660) => 2018-01-26 20:51:38 PST (1517028698)
Feb 15 08:44:06 raspberrypi weewx[17790]: manager: Added record 2018-01-26 20:51:38 PST (1517028698) to database 'weewx.sdb'
Feb 15 08:44:06 raspberrypi weewx[17790]: engine: Main loop exiting. Shutting engine down.
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: Thread-2: I USB polling device thread exiting
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: MainThread: I USB polling thread expired
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: Thread-1: I Watchdog received shutdown
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: Thread-1: I Watchdog thread exiting
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: MainThread: I Watchdog thread expired
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: MainThread: I Received packet count live:6 archive:1 control:2
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: MainThread: I Received bytes:816 sent bytes:56
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: MainThread: I Packet archive queue len:0 live queue len:6
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: MainThread: W Exiting with packets still in live queue cnt:6
Feb 15 08:44:06 raspberrypi weewx[17790]: wmr200: MainThread: I Driver gracefully exiting
Feb 15 08:44:06 raspberrypi weewx[17790]: engine: Caught unrecoverable exception in engine:
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****  Non-positive value for record field 'interval': 0
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****  Traceback (most recent call last):
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/engine.py", line 890, in main
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      engine.run()
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/engine.py", line 160, in run
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      self.dispatchEvent(weewx.Event(weewx.STARTUP))
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      callback(event)
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/engine.py", line 520, in startup
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      self._catchup(self.engine.console.genStartupRecords)
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/engine.py", line 635, in _catchup
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      origin='hardware'))
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      callback(event)
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/engine.py", line 601, in new_archive_record
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      dbmanager.addRecord(event.record, accumulator=self.old_accumulator)
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/manager.py", line 246, in addRecord
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      self._addSingleRecord(record, cursor, log_level)
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/manager.py", line 1212, in _addSingleRecord
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      _weight = self._calc_weight(record)
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****    File "/usr/share/weewx/weewx/manager.py", line 1582, in _calc_weight
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****      raise ValueError("Non-positive value for record field 'interval': %s" % (record['interval'], ))
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****  ValueError: Non-positive value for record field 'interval': 0
Feb 15 08:44:06 raspberrypi weewx[17790]:     ****  Exiting.
```

And here's the comparison:
```              
  if timestamp_packet_interval < 1:
                    loginf(('genStartup() Discarding received archive record that presented out-of-order; '
                            'current timestamp:%s; previous timestamp:%s')
                           % (weeutil.weeutil.timestamp_to_string(timestamp_packet_current),
                              weeutil.weeutil.timestamp_to_string(timestamp_packet_previous)))
```

tkeffer, 2019-02-15:
Fixed in commit https://github.com/weewx/weewx/commit/7ccb1345cc0831bae7df46266ee79f7b0b8c3b59

Thank you for your patience! I just wanted to make sure we were actually fixing the problem

superhet70, 2019-08-19:
I too have this problem with a WMR200. 
How does changing the timestamp_packet_interval in wmr200.py fix this issue when the error comes from manager.py? I changed wmr200.py a few months ago and the problem still pops up intermittently. I have now change line 1582 in manager.py and will wait and see if it happens again.

mdecker, 2019-09-15:
I have the same issue on WMR200. It happens very irregularly. Before a crash yesterday it ran for months.  I manually changed the code according to commit 7ccb134. That did not help, however. Just last night I had another crash. So I suspect that this does not fix the issue, unfortunately.

> Sep 14 17:56:54 vega weewx[31271]: engine: Main loop exiting. Shutting engine down.
> Sep 14 17:56:54 vega weewx[31271]: wmr200: Thread-4: I USB polling device thread exiting
> Sep 14 17:56:54 vega weewx[31271]: wmr200: MainThread: I USB polling thread expired
> Sep 14 17:56:54 vega weewx[31271]: wmr200: Thread-3: I Watchdog received shutdown
> Sep 14 17:56:54 vega weewx[31271]: wmr200: Thread-3: I Watchdog thread exiting
> Sep 14 17:56:54 vega weewx[31271]: wmr200: MainThread: I Watchdog thread expired
> Sep 14 17:56:54 vega weewx[31271]: wmr200: MainThread: I Received packet count live:21870 archive:971 control:1758
> Sep 14 17:56:54 vega weewx[31271]: wmr200: MainThread: I Received bytes:976 sent bytes:48
> Sep 14 17:56:54 vega weewx[31271]: wmr200: MainThread: I Packet archive queue len:0 live queue len:4
> Sep 14 17:56:54 vega weewx[31271]: wmr200: MainThread: W Exiting with packets still in live queue cnt:4
> Sep 14 17:56:54 vega weewx[31271]: wmr200: MainThread: I Driver gracefully exiting
> Sep 14 17:56:54 vega weewx[31271]: engine: Caught unrecoverable exception in engine:
> Sep 14 17:56:54 vega weewx[31271]:     ****  Non-positive value for record field 'interval': 0
> Sep 14 17:56:54 vega weewx[31271]:     ****  Traceback (most recent call last):
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/engine.py", line 890, in main
> Sep 14 17:56:54 vega weewx[31271]:     ****      engine.run()
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/engine.py", line 160, in run
> Sep 14 17:56:54 vega weewx[31271]:     ****      self.dispatchEvent(weewx.Event(weewx.STARTUP))
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
> Sep 14 17:56:54 vega weewx[31271]:     ****      callback(event)
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/engine.py", line 520, in startup
> Sep 14 17:56:54 vega weewx[31271]:     ****      self._catchup(self.engine.console.genStartupRecords)
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/engine.py", line 635, in _catchup
> Sep 14 17:56:54 vega weewx[31271]:     ****      origin='hardware'))
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
> Sep 14 17:56:54 vega weewx[31271]:     ****      callback(event)
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/engine.py", line 601, in new_archive_record
> Sep 14 17:56:54 vega weewx[31271]:     ****      dbmanager.addRecord(event.record, accumulator=self.old_accumulator)
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/manager.py", line 246, in addRecord
> Sep 14 17:56:54 vega weewx[31271]:     ****      self._addSingleRecord(record, cursor, log_level)
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/manager.py", line 1212, in _addSingleRecord
> Sep 14 17:56:54 vega weewx[31271]:     ****      _weight = self._calc_weight(record)
> Sep 14 17:56:54 vega weewx[31271]:     ****    File "/usr/share/weewx/weewx/manager.py", line 1582, in _calc_weight
> Sep 14 17:56:54 vega weewx[31271]:     ****      raise ValueError("Non-positive value for record field 'interval': %s" % (record['interval'], ))
> Sep 14 17:56:54 vega weewx[31271]:     ****  ValueError: Non-positive value for record field 'interval': 0
> Sep 14 17:56:54 vega weewx[31271]:     ****  Exiting.

tkeffer, 2019-09-16:
Michael, could you try this fix? https://github.com/cmanton/weewx/commit/47800a0a622b7931898faf5aa6b254cac943557a

mdecker, 2019-09-18:
thanks for the hint. I replaced my wmr200.py with the one from cmanton@47800a0 and will report

mdecker, 2019-09-20:
@cmanton there seems to be a typo or misconception in your fix. My weewx tripped into this last night. It looks to me like `timestamp_packet_previous += self.wmr200.time_drift` it should maybe be 
`timestamp_packet_previous += self.time_drift`?

```
Sep 20 05:32:21 vega weewx[15035]: engine: Caught unrecoverable exception in engine:
Sep 20 05:32:21 vega weewx[15035]:     ****  'WMR200' object has no attribute 'wmr200'
Sep 20 05:32:21 vega weewx[15035]:     ****  Traceback (most recent call last):
Sep 20 05:32:21 vega weewx[15035]:     ****    File "/usr/share/weewx/weewx/engine.py", line 890, in main
Sep 20 05:32:21 vega weewx[15035]:     ****      engine.run()
Sep 20 05:32:21 vega weewx[15035]:     ****    File "/usr/share/weewx/weewx/engine.py", line 160, in run
Sep 20 05:32:21 vega weewx[15035]:     ****      self.dispatchEvent(weewx.Event(weewx.STARTUP))
Sep 20 05:32:21 vega weewx[15035]:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
Sep 20 05:32:21 vega weewx[15035]:     ****      callback(event)
Sep 20 05:32:21 vega weewx[15035]:     ****    File "/usr/share/weewx/weewx/engine.py", line 520, in startup
Sep 20 05:32:21 vega weewx[15035]:     ****      self._catchup(self.engine.console.genStartupRecords)
Sep 20 05:32:21 vega weewx[15035]:     ****    File "/usr/share/weewx/weewx/engine.py", line 632, in _catchup
Sep 20 05:32:21 vega weewx[15035]:     ****      for record in generator(lastgood_ts):
Sep 20 05:32:21 vega weewx[15035]:     ****    File "/usr/share/weewx/weewx/drivers/wmr200.py", line 1940, in genStartupRecords
Sep 20 05:32:21 vega weewx[15035]:     ****      timestamp_packet_previous += self.wmr200.time_drift
Sep 20 05:32:21 vega weewx[15035]:     ****  AttributeError: 'WMR200' object has no attribute 'wmr200'
Sep 20 05:32:21 vega weewx[15035]:     ****  Exiting.

```

cmanton, 2019-09-20:
Hi Michael,

Thanks for testing...my wmr200 has died and I replaced with another
vendor/model.

The failure is WMR200 *is* the object scope, but then the code tries to
access wmr200 which is already available; so the variable 'path' is
incorrect.

I've included a patch to overlay on the one you've already applied to
address this issue.

thx,

Chris

tkeffer, 2019-11-20:
@cmanton Unfortunately, users are still getting this problem, so I'm reopening. Could you investigate?

cmanton, 2019-11-20:
Initial recommendation is to disable the archive feature on the wmr200 model and only process live packets.

erase_archive=true

Is there a recent bug report ?

tkeffer, 2019-11-20:
@cmanton Yes. Here's the thread: https://groups.google.com/forum/#!topic/weewx-user/WFDP08JI0E8

tkeffer, 2019-12-02:
Hopefully fixed for good!

closed 2019-12-02 by tkeffer

Adri1G, 2020-04-11:
I had this crash on my weewx installation, normally with the latest version of weewx (3.9.2).

This is from wee_debug --info
```
Using configuration file /etc/weewx/weewx.conf
Using database binding 'wx_binding', which is bound to database 'archive_sqlite'

System info
  Platform:       Linux-4.19.97-v7+-armv7l-with-debian-10.3
  Python Version: 2.7.16

Load Information
  1 minute load average:  0.08
  5 minute load average:  0.02
  15 minute load average: 0.01

General Weewx info
  Weewx version 3.9.2 detected.

Station info
  Station type: WMR200
  Driver:       weewx.drivers.wmr200

Driver info
[WMR200]
    # This section is for the Oregon Scientific WMR200

    # The station model, e.g., WMR200, WMR200A, Radio Shack W200
    model = WMR200

    # The driver to use:
    driver = weewx.drivers.wmr200

Currently installed extensions
Extension Name    Version   Description
cmon              0.16      Collect and display computer health indicators.

Archive info
  Database name:        weewx.sdb
  Table name:           archive
  Version               2.0
  Unit system:          1 (US)
  First good timestamp: 2020-04-05 15:58:00 CEST (1586095080)
  Last good timestamp:  2020-04-08 02:07:00 CEST (1586304420)
  Number of records:    3450
  weewx (weewx.conf) is set to use an archive interval of 300 seconds.
  The station hardware was not interrogated in determining archive interval.

Databases configured in weewx.conf
  Database name:        weewx.sdb
  Database driver:      weedb.sqlite

  Database name:        weewx
  Database driver:      weedb.mysql
  Database host:        localhost

  Database name:        cmon.sdb
  Database driver:      weedb.sqlite
```

This is from syslog
```
Apr  8 02:08:03 raspberrypi weewx[566]: wmr200: Thread-2: E read_device() USB Error Reason:[Errno 110] Operation timed out
Apr  8 02:08:03 raspberrypi weewx[566]: wmr200: Thread-2: E USB device read error
Apr  8 02:08:03 raspberrypi weewx[566]: wmr200: MainThread: E USB polling thread unexpectedly terminated
Apr  8 02:08:03 raspberrypi weewx[566]: engine: Main loop exiting. Shutting engine down.
Apr  8 02:08:03 raspberrypi weewx[566]: engine: Shutting down StdReport thread
Apr  8 02:08:03 raspberrypi weewx[566]: wmr200: MainThread: I USB polling thread expired
Apr  8 02:08:03 raspberrypi weewx[566]: wmr200: Thread-1: I Watchdog received shutdown
Apr  8 02:08:03 raspberrypi weewx[566]: wmr200: Thread-1: I Watchdog thread exiting
[... 64 lines cut ...]
Apr  8 02:09:09 raspberrypi weewx[566]:     ****  Non-positive value for record field 'interval': 0
Apr  8 02:09:09 raspberrypi weewx[566]:     ****  Traceback (most recent call last):
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/engine.py", line 894, in main
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      engine.run()
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/engine.py", line 160, in run
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      self.dispatchEvent(weewx.Event(weewx.STARTUP))
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      callback(event)
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/engine.py", line 520, in startup
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      self._catchup(self.engine.console.genStartupRecords)
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/engine.py", line 635, in _catchup
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      origin='hardware'))
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/engine.py", line 224, in dispatchEvent
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      callback(event)
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/engine.py", line 601, in new_archive_record
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      dbmanager.addRecord(event.record, accumulator=self.old_accumulator)
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/manager.py", line 246, in addRecord
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      self._addSingleRecord(record, cursor, log_level)
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/manager.py", line 1212, in _addSingleRecord
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      _weight = self._calc_weight(record)
Apr  8 02:09:09 raspberrypi weewx[566]:     ****    File "/usr/share/weewx/weewx/manager.py", line 1582, in _calc_weight
Apr  8 02:09:09 raspberrypi weewx[566]:     ****      raise ValueError("Non-positive value for record field 'interval': %s" % (record['interval'], ))
Apr  8 02:09:09 raspberrypi weewx[566]:     ****  ValueError: Non-positive value for record field 'interval': 0
Apr  8 02:09:09 raspberrypi weewx[566]:     ****  Exiting.
```

superhet70, 2020-04-12:
> 
> 
> Michael, could you try this fix? [cmanton@47800a0](https://github.com/cmanton/weewx/commit/47800a0a622b7931898faf5aa6b254cac943557a)

You should try this patch. it has worked for my WMR200.

tkeffer, 2020-04-12:
... or wait for V4.0, which will include the patch.

Adri1G, 2020-04-12:
OK. Thank you :-)