weewx/weewx #377: Weather Underground unsecure data (http not https)

mljenkinsdotcom, 2019-02-16:
Why does weewx send weather underground calls unsecure rather than using https?  On my deployment, I edited the weewx/restx.py file and changed the rf_url and pws_url under StdUnderground(StdRESTful) and changed http to https.  Is there a reason this project leaves it http?  I have had no issue using https since switching.

I searched around and could not find anything related to this in the forums or other issues, so I figured maybe I should bring it up.

tkeffer, 2019-02-16:
Good question. I vaguely recall a discussion about this, but I'll be darned if I can find anything either.

Would you be willing to submit a PR? It should be a very simple one.

tkeffer, 2019-02-24:
Fixed with PR #378

closed 2019-02-24 by tkeffer