weewx/weewx #379: CopyGenerator sub folder error

cyrilrey, 2019-02-22:
I would create a new template using Smentic Ui and Dygraph, but I have an issue.

**It seems that CopyGenerator can not handle with sub folder ?**

My folder is look like that :

```
cyril@meteo:/etc/weewx/skins/Semantic $ tree -d -f -i
.
./dygraph
./Semantic-UI-CSS
./Semantic-UI-CSS/components
./Semantic-UI-CSS/themes
./Semantic-UI-CSS/themes/default
./Semantic-UI-CSS/themes/default/assets
./Semantic-UI-CSS/themes/default/assets/fonts
./Semantic-UI-CSS/themes/default/assets/images
```
I try adding each folder individually in my skin.conf :

```
[CopyGenerator]

    # This section is used by the generator CopyGenerator

    # List of files to be copied only the first time the generator runs
    copy_once = ./Semantic-UI-CSS/themes/default/assets/images/*, ./Semantic-UI-CSS/themes/default/assets/fonts/*, ./Semantic-UI-CSS/themes/default/assets/*, ./Semantic-UI-CSS/themes/default/*, ./Semantic-UI-CSS/themes/*, ./Semantic-UI-CSS/components/*, ./Semantic-UI-CSS/*, ./dygraph/*

    # List of files to be copied each time the generator runs
    # copy_always =
```

But I get that error :

```
sudo wee_reports --config=../../weewx_test.conf
Using configuration file ../../weewx_test.conf
Generating for all time
Traceback (most recent call last):
  File "/usr/share/weewx/weewx/reportengine.py", line 239, in run
    obj.start()
  File "/usr/share/weewx/weewx/reportengine.py", line 273, in start
    self.run()
  File "/usr/share/weewx/weewx/reportengine.py", line 440, in run
    shutil.copy(_file, dest_dir)
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 21] est un dossier: './Semantic-UI-CSS/themes/default/assets/fonts'
```

It seems that that error appear at first folder with sub folder.

gjr80, 2019-02-22:
Copygenerator handles sub-directories, try removing './' from the front of each entry.

tkeffer, 2019-02-23:
Definitely a bug. The problem is when a top-level directory has another directory nested under it. The CopyGenerator does not recognize it as a directory, and, instead, treats it as a file.

Amazing that after nearly 10 years of using this, we've never nested things more than one level deep!

cyrilrey, 2019-02-23:
Hi,
As suggested by gjr80, I removed `./` in front of each entry. But I get same error message.
Yes, is really amazing that this bug goes about only now. That my first real bug I found in a software. So I was not confident before posting. Are you sure I did not do something wrong? Could you reproduce?

I think finding a solution may be not so easy ? Will you provide some kind of recursivity or just an `if` test for excluding folder ?

-----
By the way could, you give me some recommendation or guideline about the proper way to output CSV  file as reports instead of image.

I would like to show value, like the second example on that page :
[http://dygraphs.com/tutorial.html](http://dygraphs.com/tutorial.html)

That mean producing csv files for days, month, and years, similar to :
[dygraphs.com/temperatures.csv](http://dygraphs.com/temperatures.csv)

I saw a weewx module for csv, but purpose it not for reporting, but archiving.

Best regards
Cyril

tkeffer, 2019-02-23:
Don't worry, finding a solution will be easy.

There is a workaround: pick your wild cards ('`*`') so that they do not pick up any subdirectories. For example, 

```
./Semantic-UI-CSS/components/*.css
./Semantic-UI-CSS/themes/*.css
./Semantic-UI-CSS/themes/default/*css
./Semantic-UI-CSS/themes/default/assets/fonts/*.font
./Semantic-UI-CSS/themes/default/assets/images/*.png
```

Hopefully, you get the idea.

tkeffer, 2019-02-23:
Fixed in commit https://github.com/weewx/weewx/commit/2063dfa3aba65e747c1f47a222d43163b72d584a

closed 2019-02-23 by tkeffer

cyrilrey, 2019-02-26:
Problem solved in less than 24h !!
Thanks for your reactivity.