weewx/weewx #386: Any chance to add inHumidity to the plots?

chaoschris, 2019-03-08:
Any chance to add inHumidity to the plots?
I can add inHumidity together with inTemp and create a combined PNGs, this works fine.

But I didn't understood how to create single PNG only with the value from inHumidity.
Adding a new entry for a daily plot in skin.conf ends up in a failure and no plots were created.

working skin.conf:
....
[[day_images]]
[[[daytempin]]]
      [[[[inTemp]]]]
      [[[[inHumidity]]]]
....

not working skin.conf:
....
[[day_images]]
[[[dayhumidity_in]]]
      [[[[inHumidity]]]]
....

Mär 08 14:15:36 WEATHERCAM2 weewx[20330]: reportengine: Failed to read skin configuration file /etc/weewx/skins/Seasons/skin.conf for report… line 305.
Mär 08 14:15:36 WEATHERCAM2 weewx[20330]:         ****  Report ignored

Any chance to get a single plot for inside humidity?

Best regards,
Chris

tkeffer, 2019-03-08:
Hello, Chris
This issues list is for bug tracking. Please post your question to [weewx-user](https://groups.google.com/forum/#!forum/weewx-user) where it will have a better chance of getting answered.

Oh, and you should include more of your `skin.conf`. There's no way to know which line is line 305

closed 2019-03-08 by tkeffer

chaoschris, 2019-03-08:
Added a entry to the forum.