weewx/weewx #389: Clarify Wiki Instructions for running weewx as systemd service

pfletch101, 2019-03-11:
The wiki article on running weewx from systemd [https://github.com/weewx/weewx/wiki/systemd](url) says that: "The weewx distribution includes a systemd "unit" file called weewx.service that tells systemd how to run weewx." While it is possible, albeit with some difficulty, to find this file here on GitHub (called "systemd"), it doesn't appear to be included when you install the software (on a raspberry pi, in my case). Since the wiki article is from 2017, I was a bit concerned that things might have changed enough that the example file given in it would not work. As it turns out, the example unit file does work (provided you make the appropriate changes to the file paths in it). The article also doesn't, however, specify the changes you should make to disable/delete the weewx startup code in sysv, which I found (hopefully given correctly) in the user forum:

`sudo /etc/init.d/weewx stop

sudo update-rc.d weewx remove

sudo rm /etc/init.d/weewx`

It would be nice if the wiki article included a live link to the current version of the recommended .service file and instructions on disabling/deleting the sysv startup code.

tkeffer, 2019-03-11:
Good heavens. It's a Wiki! If you think the article can be phrased better, go for it!

closed 2019-03-11 by tkeffer

pfletch101, 2019-03-14:
I knew that it was a wiki and assumed that it was possible for me to revise it directly, but I would have preferred that someone who actually knew what (s)he was talking/writing about did so! If you think that is unlikely to happen, I will take a stab at it - I know a small amount about systemd and I (obviously) use weewx, but I know virtually nothing about sysv.