weewx/weewx #390: Document arguments to $current

tkeffer, 2019-03-15:
The tag `$current` has optional arguments `timestamp` and `max_delta`, which should be documented.

PeterQuinn925, 2019-04-14:
I'm interested in helping out and this looks like a good place to start. It might be easier to do this yourself than to answer my questions, but I'm game to try if you are.

I see in tags.py the code for current. 

It looks like it returns a CurrentObj for timestamp and max delta aren't defined and returns a a CurrentObj for the time in the timestamp if it's specified. I don't understand what it's doing with max_delta. I mean I see it's calling CurrentObj with the max_delta, but I don't know what max_delta means. It's a time delta? Like day or week? How does it know which obj to pull from that day or week?

Another dumb question. You say it should be documented. Documented meaning putting comments in the code? Or adding it to the user manual or something. I couldn't find any user docs on this, so I assume you mean add comments to the source. Is that right?

Lastly, this is a real noob question. I've been putting stuff in my own Github repositories but I've never worked on a shared project. I suppose I need to clone weewx, make the change and then issue a pull request? Or is there some other procedure to add some documentation?

Thanks for making it easy to get started here.

tkeffer, 2019-04-14:
In this case, "document" means add documentation to the Customizing Guide, specifically the documentation on [`$current`](http://weewx.com/docs/customizing.htm#_____________Time_period_$current_________). 

The present documentation documents the option `data_binding`. We need to add `timestamp` and `max_delta`. So, for example, the "most general tag" would look something like:

    $current($data_binding=binding_name, $timestamp=timestamp, $max_delta=max_delta).obstype[.optional_unit_conversion][.optional_formatting]

then you would add comments about what *`timestamp`* and *`max_delta`* are, much like the present documentation for *`binding_name`*.

Do you know HTML? This project is simple, but it does require a fair amount of HTML.

Yes, you would clone the weewx repository, make changes in the `development` branch, then submit a pull request.

PeterQuinn925, 2019-04-19:
Ok. I think I understand what to do. I have time today to look at it. 
In a nutshell,
If a timestamp is specified, use it to find the record and not the current time. 

max_delta is the largest difference in time that is acceptable (in seconds)

For example if you want to report the temperature on the hour, but your weather station only reports every 2 minutes, you would use something like:
$timestamp=(time.time() - (time.time() % 3600)) ##the last hour's time as unix epoch. Can you do this in a template?
$max_delta = 120 

Is this correct? Let me know and I'll write it up.

tkeffer, 2019-04-19:
Yes, you've got the right idea. 

You can write expressions like that in the template. 

    #set $timestamp = int(time.time() / 3600) * 3600

tkeffer, 2019-04-19:
PR #400

closed 2019-04-19 by tkeffer