weewx/weewx #404: Time calculation issue (division by zero)

LimitlessGreen, 2019-05-01:
Tested commit: 0c3c8c9c67cd104d71a4d52cdfcefbec90705d43

If I set the [StdArchive] -> archive_interval to a low number like 5, I got a division by zero error.

```
Traceback (most recent call last):
  File "./bin/weewxd", line 64, in <module>
    weewx.engine.main(options, args)
  File "/home/weewx/bin/weewx/engine.py", line 894, in main
    engine.run()
  File "/home/weewx/bin/weewx/engine.py", line 191, in run
    self.dispatchEvent(weewx.Event(weewx.NEW_LOOP_PACKET, packet=packet))
  File "/home/weewx/bin/weewx/engine.py", line 224, in dispatchEvent
    callback(event)
  File "/home/weewx/bin/weewx/engine.py", line 540, in new_loop_packet
    self.accumulator = self._new_accumulator(event.packet['dateTime'])
  File "/home/weewx/bin/weewx/engine.py", line 650, in _new_accumulator
    self.archive_interval)
  File "/home/weewx/bin/weeutil/weeutil.py", line 197, in startOfInterval
    m = int(time_tt.tm_min // interval_m * interval_m)
ZeroDivisionError: integer division or modulo by zero
```

tkeffer, 2019-05-01:
Just to clarify: are you trying to use a 5 second archive interval?

If so, then yes, it will not work. It's not documented, but the interval has to be evenly divisible by 60 seconds.

LimitlessGreen, 2019-05-04:
What's wrong about that? It's hard to debug things, if you get only one report by 60 seconds.
Is there a better way to debug weewx? So pls tell it me.

I'm working on a outgoing API https://github.com/severusstick/weewx-api (Name will be changed somewhere in the future). It pushes data to a separate website (In my case: http://wetter-ottendorf-okrilla.de/neu/wetter-aktuell/).

closed 2019-05-04 by tkeffer

tkeffer, 2019-05-04:
I'm not sure what you mean by "debug weewx." 

Make sure you understand the difference between LOOP packets and archive records. See the section [*LOOP packets vs archive records*](http://weewx.com/docs/customizing.htm#_____________LOOP_packets_vs._archive_records_) in the Customizing Guide.

If you need to see the LOOP packets, just run WeeWX [from the command line](http://weewx.com/docs/usersguide.htm#Running_directly) and watch what it prints.

Closing this issue as this is not a bug.

LimitlessGreen, 2019-05-04:
You're right. It's not a bug. It's an error!
So you have to catch or fix it...