weewx/weewx #410: Memory Leak

bryanmason, 2019-05-17:
weewx has a significant memory leak on my system.  It will consume ~2.5 GB of memory (RAM and swap) in about 5 days.  

OS: CentOS Linux release 7.6.1810 (Core) 
weewx version: weewx-3.9.1-1.rhel.noarch
python version: python-2.7.5-76.el7.x86_64

Nothing else of consequence is running on this system.  It's a VM with 1 GiB of RAM dedicated for weewx. I'm using the weewx-interceptor to capture data from an AcuRite bridge.

I created a log file with the command `(while /bin/true; do date; grep '^Vm' /proc/$(pidof python)/status; sleep 300; done) | tee weewx-memory.txt` and converted the data into a spreadsheet.  I'll upload the spreadsheet to show how memory use is growing over time.

Please let me know what additional data you need or if there are any diagnostics that I can do to figure this out.

bryanmason, 2019-05-17:
GitHub won't let me upload a spreadsheet, so here's a PDF of the chart in the spreadsheet.
[weewx-memory.pdf](https://github.com/weewx/weewx/files/3192410/weewx-memory.pdf)

tkeffer, 2019-05-17:
I know it's not very helpful, but weewx-interceptor is not part of WeeWX. Unless you can show that other parts of WeeWX are responsible for the leak, there isn't much we can do for you.

Having said that, I am aware of a memory leak on Debian 32-bit, that seems to involve either the exception machinery, or the sqlite API. It's been a nagging problem for me and my weather station. Perhaps something similar is happening under CentOS?

bryanmason, 2019-05-17:
Thanks for the quick response.  I understand the situation.  I'm happy to help bisect the problem to try and determine whether it's in weewx or weewx-interceptor.  Are there debug flags I can enable or lines of code to add to print memory usage within python? (I'm an experienced programmer and have used python from time to time, but nothing extensive).

tkeffer, 2019-05-17:
I have not had very good luck with all the Python memory leak tools. Kind of a Heisenberg Principle: they interfere with the thing being measured.

I use a small weewx service that records VSS , RSS, and SHR as `soilMoist1`, `soilMoist2`, and `soilMoist3` in the database, then have WeeWX plot them up as part of the regular reporting cycle. Then... patience. It can take weeks to nail down a leak. After a month of trying, I never did find the problem with the Debian leak.

Here's the WeeWX service:

```python
import os
import resource

import weewx
from weewx.wxengine import StdService

class Memory(StdService):
    
    def __init__(self, engine, config_dict):
        # Pass the initialization information on to my superclass:
        super(Memory, self).__init__(engine, config_dict)

        self.page_size = resource.getpagesize()
        self.bind(weewx.NEW_ARCHIVE_RECORD, self.newArchiveRecord)

    def newArchiveRecord(self, event):
        
        pid = os.getpid()
        procfile = "/proc/%s/statm" % pid
        try:
            with open(procfile) as fd:
                mem_tuple = fd.read().split()
        except IOError:
            return
         
        # Unpack the tuple:
        (size, resident, share, text, lib, data, dt) = mem_tuple

        mb = 1024 * 1024
        event.record['soilMoist1'] = float(size)     * self.page_size / mb 
        event.record['soilMoist2'] = float(resident) * self.page_size / mb
        event.record['soilMoist3'] = float(share)    * self.page_size / mb
```

closed 2019-06-24 by tkeffer