weewx/weewx #412: Please add support for the Davis WeatherLink Live device

kobuki, 2019-05-19:
As the title suggests, it's an enhancement request. Davis produced a new network-capable [device](https://www.davisinstruments.com/weatherlinklive/) called WeatherLink Live for their VP2 stations, and it uses a completely new, standards-based protocol for communication of local condition records (similar to archive records), while real-time data is also available through local UDP broadcasts (~similar to old loop records).

The API docs is available here: https://github.com/weatherlink/weatherlink-live-local-api

It would be nice to have an official WeeWx driver integrated into the code base.

tkeffer, 2019-05-20:
Looks interesting. I'll give them a call tomorrow and see if they'll send me a unit.

kobuki, 2019-09-30:
Did they by any chance respond favorably?

vinceskahan, 2020-02-20:
The WeatherFlow UDP driver works exactly this way - pulls JSON-formatted UDP broadcasts off the wire(less), maps them to weewx data elements, and lets weewx do the heavy lifting.  A quick glance at the Davis link mentioned above seems to say this would be a very doable thing to cook up based on that driver.

Or send me a Davis thingie and I'd be happy to take a crack at it.

jonotaegi, 2020-04-18:
I wrote a very basic driver that parses the HTTP requests described in the WeatherLink Live Local API documentation: https://weatherlink.github.io/weatherlink-live-local-api/

Nothing fancy but does the job until someone writes a better solution:

https://github.com/jonotaegi/weewx-wll

DavidCWGA, 2020-05-01:
Thanks @jonotaegi , I just set that up and it seems to work great!

lrosser2103, 2020-05-16:
I can't make this work.  I have followed your github to the letter with numerous linux packages, PI, Debian, Ubuntu, and triple checked the configs, etc.  @DavidCWGA what platform are you using and any details can you provide?

vinceskahan, 2020-05-16:
"I can't make this work" is not a problem report anybody can help you with.  You might want to use the weewx-users group to ask the question and provide logs etc. with what you did, and what you see.

FWIW, I have a different variant at https://github.com/vinceskahan/weewx-weatherlinklive-json but mine doesn't handle rain currently, if that is a show-stopper for you.

I've been meaning to steal/borrow/learn-from @jonotaegi variant one of these days to add rain to mine, but I haven't gotten any requests from my existing two users to do so, so I haven't gotten around to it.

lrosser2103, 2020-05-17:
@vinceskahan thanks for your feedback.  I think the problem is that the install is hitting some errors.  This occured with both installers (yours and @jonotaegi ):
[image]

vinceskahan, 2020-05-17:
The problem is you have a duplicate section name at line 616 :-)

Take a look at your weew.conf closely...you probably had a cut+paste
massacre.

lrosser2103, 2020-05-17:
That was it.  I edited my config for the WLL driver before running the install script.  The install script apparently adds the WLL config snip at the end of the config file.  Thank you so much for your help.  Hats off to the community for all you do to help us learners!

mykerinos1, 2020-05-25:
i just bought a Weatherlink live and i want to collect data in domotics with Weewx but i don't know how to do. i tried to install weewx but it doesnt allow connection. 
Does someone can explain me how to install it ? i'm also new on raspberry :(
Thanks a lot

captain-bodge, 2020-05-26:
Has anyone picked up feeding the UDP broadcasts into WeeWx yet?

FidgetyRat, 2020-05-26:
> Has anyone picked up feeding the UDP broadcasts into WeeWx yet?

Curious about this myself. I have a Live coming in the mail at some point soon but only see developer references to the HTML and JSON branches. Anything being officially supported yet?

kobuki, 2020-05-26:
This should help: https://weatherlink.github.io/weatherlink-live-local-api/

vinceskahan, 2020-05-26:
Get your Live shipped to me and I'll write it :-)

The WLLive box broadcasts UDP if you throw it a HTTP GET occasionally to turn broadcasts on.  It then emits the same JSON via UDP that you get if you do an occasional web get of the current conditions URL.

So what's needed is the WeatherFlowUDP driver basically tweaked to:
 * do the HTTP GET occasionally to keep UDP flowing
 * listen on the right port
 * and decode the Davis JSON (my driver does this already, not tough at all)

I worked with a very non-Linux user in writing my minimal driver and verified that the Davis API docs appear correct.  See "Format of Real – Time Broadcast Request" in that Davis API link for how to turn its UDP broadcast on.

jonotaegi, 2020-05-26:
I had a look back in the day and the UDP broadcast only contains rain and wind information. You still need the HTTP GET to get the rest of the sensor data.

vinceskahan, 2020-05-26:
Disagree.  Read the Davis docs again.  Pretty clearly shows the same JSON comes out of the UDP that you get from the current conditions via HTTP.

jonotaegi, 2020-05-26:
Disagree :P

Check the JSON carefully. In addition, in the first paragraph, it says (https://weatherlink.github.io/weatherlink-live-local-api/): 

> Additionally, you can also start a real-time 2.5 sec broadcast for **wind speed** and **rain** over UDP port 22222.

I tested it as well with my device and wind and rain is all I got.

vinceskahan, 2020-05-26:
it is entirely possible that I'm totally blind and/or decaffeinated, so I'll certainly defer on this one :-)

jonotaegi, 2020-05-26:
To be honest, I didn't realize of the missing fields until I tested it. Then I double-checked the documentation and it is all there :)

grebleem, 2020-07-01:
Hi,
inspired by the driver of jonotaegi I wrote a new wll driver, including the 2.5 seconds UDP wind/rain updates. So it pulls a full http report every given poll interval (min 10s) and in between is catches the UDP data. 

It can be found here: https://github.com/grebleem/weewx-weatherlinkliveudp

If you have the time, give it a try. I'm open for any suggestions.

captain-bodge, 2020-07-01:
> Hi,
> inspired by the driver of jonotaegi I wrote a new wll driver, including the 2.5 seconds UDP wind/rain updates. So it pulls a full http report every given poll interval (min 10s) and in between [...]

Ill give it a try later. 
I did fork the jonotaegi code and modified it to cater for multiple transmitters (I have rain on its own transmitter, wind and solar on another and temp/hum on a third) but life got in the way...

grebleem, 2020-07-01:
If you send a sample Json file ( /v1/current_conditions ) I can have a look at it.

captain-bodge, 2020-07-02:
I don’t have access to my running weewx until Sunday, but this is an early version of what I changed in many places in the code:

if "hum" in condition:  # most recent valid temperature **(°F)**
                            if tx_id == 4:
                                _packet.update({'outHumidity': condition["hum"]})
                                narrative = " (selected)"
                            else:
                                narrative = ""
                            loginf("<hum> found on TX %s : %s %s" % (tx_id,  condition["hum"], narrative))

The later version of code had an array in weewx.conf that had the sensor type and transmitter ID values, so the sensor / transmitters could be configured in any way.
E.g. 
WLLWind = 1
WLLTemp = 4
WLLRain = 8

I can either take your code and change it and submit a PR to bring the changes in, or if you prefer, you can replicate...

Output from current_conditions is this:

{"data":{"did":"xxxxxxxxxxxx","ts":1593697631,"conditions":[{"lsid":308870,"data_structure_type":1,"txid":1,"temp":null,"hum":null,"dew_point":null,"wet_bulb":null,"heat_index":null,"wind_chill":null,"thw_index":null,"thsw_index":null,"wind_speed_last":8.00,"wind_dir_last":326,"wind_speed_avg_last_1_min":5.56,"wind_dir_scalar_avg_last_1_min":331,"wind_speed_avg_last_2_min":4.87,"wind_dir_scalar_avg_last_2_min":322,"wind_speed_hi_last_2_min":11.00,"wind_dir_at_hi_speed_last_2_min":333,"wind_speed_avg_last_10_min":6.06,"wind_dir_scalar_avg_last_10_min":318,"wind_speed_hi_last_10_min":12.00,"wind_dir_at_hi_speed_last_10_min":309,"rain_size":1,"rain_rate_last":0,"rain_rate_hi":0,"rainfall_last_15_min":0,"rain_rate_hi_last_15_min":0,"rainfall_last_60_min":0,"rainfall_last_24_hr":0,"rain_storm":null,"rain_storm_start_at":null,"solar_rad":937,"uv_index":5.9,"rx_state":0,"trans_battery_flag":0,"rainfall_daily":0,"rainfall_monthly":0,"rainfall_year":0,"rain_storm_last":null,"rain_storm_last_start_at":null,"rain_storm_last_end_at":null},{"lsid":308871,"data_structure_type":1,"txid":4,"temp": 67.3,"hum":66.3,"dew_point": 55.7,"wet_bulb": 59.2,"heat_index": 67.2,"wind_chill": 67.3,"thw_index": 67.2,"thsw_index":null,"wind_speed_last":8.00,"wind_dir_last":326,"wind_speed_avg_last_1_min":5.56,"wind_dir_scalar_avg_last_1_min":331,"wind_speed_avg_last_2_min":4.87,"wind_dir_scalar_avg_last_2_min":322,"wind_speed_hi_last_2_min":11.00,"wind_dir_at_hi_speed_last_2_min":333,"wind_speed_avg_last_10_min":6.06,"wind_dir_scalar_avg_last_10_min":318,"wind_speed_hi_last_10_min":12.00,"wind_dir_at_hi_speed_last_10_min":309,"rain_size":2,"rain_rate_last":0,"rain_rate_hi":0,"rainfall_last_15_min":0,"rain_rate_hi_last_15_min":0,"rainfall_last_60_min":0,"rainfall_last_24_hr":0,"rain_storm":0,"rain_storm_start_at":null,"solar_rad":null,"uv_index":null,"rx_state":0,"trans_battery_flag":0,"rainfall_daily":0,"rainfall_monthly":0,"rainfall_year":49,"rain_storm_last":38,"rain_storm_last_start_at":1591298220,"rain_storm_last_end_at":1591617661},{"lsid":310225,"data_structure_type":1,"txid":8,"temp":null,"hum":null,"dew_point":null,"wet_bulb":null,"heat_index":null,"wind_chill":null,"thw_index":null,"thsw_index":null,"wind_speed_last":null,"wind_dir_last":null,"wind_speed_avg_last_1_min":null,"wind_dir_scalar_avg_last_1_min":null,"wind_speed_avg_last_2_min":null,"wind_dir_scalar_avg_last_2_min":null,"wind_speed_hi_last_2_min":null,"wind_dir_at_hi_speed_last_2_min":null,"wind_speed_avg_last_10_min":null,"wind_dir_scalar_avg_last_10_min":null,"wind_speed_hi_last_10_min":null,"wind_dir_at_hi_speed_last_10_min":null,"rain_size":2,"rain_rate_last":0,"rain_rate_hi":0,"rainfall_last_15_min":0,"rain_rate_hi_last_15_min":0,"rainfall_last_60_min":0,"rainfall_last_24_hr":0,"rain_storm":0,"rain_storm_start_at":null,"solar_rad":null,"uv_index":null,"rx_state":0,"trans_battery_flag":0,"rainfall_daily":0,"rainfall_monthly":7,"rainfall_year":330,"rain_storm_last":7,"rain_storm_last_start_at":1593594181,"rain_storm_last_end_at":1593694860},{"lsid":308865,"data_structure_type":4,"temp_in": 77.1,"hum_in":48.3,"dew_point_in": 56.1,"heat_index_in": 76.9},{"lsid":308864,"data_structure_type":3,"bar_sea_level":29.878,"bar_trend": 0.032,"bar_absolute":29.451}]},"error":null}

grebleem, 2020-07-04:
@captain-bodge  Thank you for the JSON very helpful. I can implement this, the problem with the weewx-wll driver is that is uses the 'rainstorm' from the device to count rain, but this rainstorm is only set after two ticks. I had a few times there was only 1 tick of rain and this was never registered by weewx. So I completely changed the rain collecting.
The approach of checking for Null values is good, but some people have more than one temp sensor, so the id has to be set for the correct one. I think I can get a beta out this Sunday for you to try.

uajqq, 2020-07-06:
Just setting up my brand new Vantage Vue with WeatherLink Live, and this driver seems to be working perfectly in terms of data collection. I have to wait to actually get the unit up on the roof to really test it, though--right now it's in my living room. I did have to make a small change to the stanza that got automatically added to weewx.conf (I opened a pull request with the change). 

Here's a sample of my living room weather JSON, in case it's helpful:
```
{
  "data": {
    "did": "001D0A71241A",
    "ts": 1594061320,
    "conditions": [
      {
        "lsid": 327216,
        "data_structure_type": 1,
        "txid": 1,
        "temp": 76.6,
        "hum": 53.9,
        "dew_point": 58.7,
        "wet_bulb": 63.3,
        "heat_index": 77,
        "wind_chill": 76.6,
        "thw_index": 77,
        "thsw_index": null,
        "wind_speed_last": 0,
        "wind_dir_last": 0,
        "wind_speed_avg_last_1_min": 0,
        "wind_dir_scalar_avg_last_1_min": 0,
        "wind_speed_avg_last_2_min": 0,
        "wind_dir_scalar_avg_last_2_min": 0,
        "wind_speed_hi_last_2_min": 0,
        "wind_dir_at_hi_speed_last_2_min": 0,
        "wind_speed_avg_last_10_min": 0,
        "wind_dir_scalar_avg_last_10_min": 0,
        "wind_speed_hi_last_10_min": 0,
        "wind_dir_at_hi_speed_last_10_min": 0,
        "rain_size": 1,
        "rain_rate_last": 0,
        "rain_rate_hi": 0,
        "rainfall_last_15_min": 0,
        "rain_rate_hi_last_15_min": 0,
        "rainfall_last_60_min": 0,
        "rainfall_last_24_hr": 0,
        "rain_storm": null,
        "rain_storm_start_at": null,
        "solar_rad": null,
        "uv_index": null,
        "rx_state": 0,
        "trans_battery_flag": 0,
        "rainfall_daily": 0,
        "rainfall_monthly": 0,
        "rainfall_year": 0,
        "rain_storm_last": null,
        "rain_storm_last_start_at": null,
        "rain_storm_last_end_at": null
      },
      {
        "lsid": 327215,
        "data_structure_type": 4,
        "temp_in": 80.9,
        "hum_in": 49,
        "dew_point_in": 60,
        "heat_index_in": 81.2
      },
      {
        "lsid": 327214,
        "data_structure_type": 3,
        "bar_sea_level": 30.034,
        "bar_trend": null,
        "bar_absolute": 29.958
      }
    ]
  },
  "error": null
}
```

uajqq, 2020-07-07:
I’ve had a chance to test with the unit on the roof for a little while now and I think the “bar_sea_level” value is actually supposed to map to `barometer` in weewx, not `altimeter`. This is confusing because the docs that are linked on that commit do say the WeatherLink Live module is using the altimeter method. However, this causes weewx to consistently report a few mbar higher than every other nearby source. Setting it to `barometer` makes it match exactly.

tkeffer, 2020-07-25:
There are now several wlink drivers available as extensions. Closing this issue. Feel free to continue your discussions on [weewx-development](https://groups.google.com/g/weewx-development).

closed 2020-07-25 by tkeffer