weewx/weewx #413: Restx threads silently quitting on certain errors

kobuki, 2019-05-20:
I had a momentary network misconfiguration that affected all clients on my LAN. It diverted all HTTPS traffic to a single internal IP by mistake - it was quickly fixed but some weird after-effects remained, such as this. It can happen in perfectly valid error situations as well, all it needs is a certificate mismatch, cert expiration or some other SSL certificate error on a REST target server.

Log excerpt:

```
May 20 22:58:14 weather weewx[12321]: restx: Weather365: Thread exiting. Reason: hostname 'channel1.weather365.net' doesn't match 'example.host.net'
May 20 22:58:14 weather weewx[12321]: restx: Wunderground-PWS: Unexpected exception of type <class 'ssl.CertificateError'>
May 20 22:58:14 weather weewx[12321]: restx: Wunderground-PWS: Thread exiting. Reason: hostname 'weatherstation.wunderground.com' doesn't match 'example.host.net'
```
My internal host has a valid certificate for example.host.net, obviously not a match for the original target hosts. It breaks the HTTPS function and makes all the restx threads silently exit, while WeeWx keeps running as if nothing happened.

At least 2 viable solutions would be either making WeeWx exit on such errors, or the threads retrying their respective REST/UPLOAD functions at the next possible chance instead of exiting. I think a certificate mismatch should simply be handled as any common upload error like a timeout, 5xx response, etc.

tkeffer, 2019-05-20:
Could you please post the full log around the failures, instead of just an excerpt? I'd like to see the stack traces.

kobuki, 2019-05-20:
Sorry, debug wasn't active so I can't post stack traces (I would have if I had any). The error could be replicated by redirecting a valid host name in the local DNS to the IP of another valid HTTPS host. Not all home routers or configurations are capable of that, though (mine is).

tkeffer, 2019-05-21:
I assume the exception occurred during the processing of the [`urllib2.urlopen()` expression](https://github.com/weewx/weewx/blob/master/bin/weewx/restx.py#L509).

We could catch and swallow any errors of the type `ssl.CertificateError`, but that seems like it could be ripe with unintended consequences.

Any suggestions?

kobuki, 2019-05-21:
While it's something that doesn't disappear by itself like a network outage, if the certificate is indeed fixed, operations can go on. An extra option could be ignoring expired or mismatched certs, though that's considered highly unsafe by today's standards. In any case, uploading operations silently dying for extended periods without notice is a pretty bad thing. I noticed it only because WU has a notification email service (it took 3 hours, though). If WeeWx dies with its child thread, it gets noticed pretty quickly and can be acted upon, but it has obvious drawbacks, too. Maybe if WeeWx had any kind of notification system, like email, or some service status indicator that can be templated and put on the generated web pages...

I think ignoring any certificate errors + logging them with at least a warning level might be sufficient as a quick fix.

tkeffer, 2019-06-24:
My inclination is to leave things as they are. This is the first time this issue has arisen and, even then, it was due to a networking configuration problem with your network. To the best of my knowledge, it has never happened in a more organic fashion.

But, if that changes, we can always reopen.

kobuki, 2019-06-24:
It will happen if a certificate expires where any of the modules or drivers connect (as one example). I think this is by principle that errors don't get silently ignored, causing unnoticed other issues. My setup and the configuration error may have been very specific, but it shed light on a more serious issue.

The nature of the problem is such it's very hard to detect. Maybe it happened many times before but people didn't notice the few lines of warnings in the log (as is normal for the average user). In these cases they simply restart weewx or their pc and done. The issue might have been handled at the other end.

tkeffer, 2019-06-24:
It's not silently ignored --- the error is logged. It's true that the user has to look at the log (or, better, use logwatch).

The alternative is to ignore the error and keep going. That just doesn't seem right to me.

kobuki, 2019-06-24:
> The alternative is to ignore the error and keep going. That just doesn't seem right to me.

From the user's perspective, this is exactly what's happening. You can't expect an average user to check the logs regularly, but they might not notice a stopped REST uploader for weeks. As a workaround, I can install/configure a log watcher on my server to alert me in case of such things happening, but I'm not the average user, either.

tkeffer, 2019-06-24:
You're wearing me down. Commit https://github.com/weewx/weewx/commit/52b1016b256ac0ca0f233ca8d7a93e057e0a85cf has a solution. It will wait one hour, then do a retry. It's in a new branch called `CertificateError` off `development`.

I have no idea how to test this. Perhaps you can pull the [new branch](https://github.com/weewx/weewx/tree/CertificateError), then test on your system?

Still not convinced I want to do this. It feels like extra complexity for a very unusual corner case.

kobuki, 2019-06-24:
> You're wearing me down. 

Sorry, that was not my intention :)

It's indeed a corner case, but the point is, other connection-related exceptions can happen that we can't foresee. I can test this, no problem. But what do you think about the following?

``` except (FailedPost, CertificateError) as e:```

instead of just:

``` except FailedPost as e:```

Would it be simpler to handle those the same way?

tkeffer, 2019-06-24:
`FailedPost` exceptions are retried right away. `CertificateError` exceptions are retried after a delay. The latter are treated in a manner similar to `BadLogin`.

kobuki, 2019-06-24:
> `FailedPost` exceptions are retried right away. `CertificateError` exceptions are retried after a delay. The latter are treated in a manner similar to `BadLogin`.

Yes, I understand that - I asked if it might be simpler for you to handle them in one block without any further changes.

A quick question: I see a lot of changes in your branch code. Should I install the whole package from your branch or can I just put a few files in place for test? If the former, it needs some more time to reconfig/test stuff. Anyway, I was able to reproduce the issue in the old (latest release, 3.9.1) code using a simple dnat rule in my router tailored to trigger it.

```
Jun 24 17:54:15 weather weewx[2382]: restx: Weather365: Thread exiting. Reason: hostname 'channel1.weather365.net' doesn't match either of 'autoconfig.xxx.tld', 'autodiscover.xxx.tld'
```

tkeffer, 2019-06-24:
The branch is off of `development`, the code base for the next version of WeeWX, V4.0. This version will run under either Python 2 or Python 3. Functionally, it is identical to V3.9, but the implementations can be quite different. 

You'll have to "install" the whole package. I put "install" in quotes, because all you really have to do is clone the repository. If you have `git`:

```shell
git clone --single-branch --branch CertificateError https://github.com/weewx/weewx.git
```

If you do not have `git`, then you can download a zip file from the link https://github.com/weewx/weewx/archive/CertificateError.zip

Whichever method you use, you can then run it normally:

```shell
cd weewx
./bin/weewxd /etc/weewx/weewx.conf
```

(adjust paths as necessary)

kobuki, 2019-06-24:
Git is fine. If I can just run the dev version instead of the stable 3.9.1 on the same db and config, that's fine and I can test it easily. I'll have a look this week.

tkeffer, 2019-06-24:
Yup. Should work fine using the same database and `weewx,conf` configuration file.

tkeffer, 2019-07-09:
Any progress on this?

kobuki, 2019-07-09:
Sorry, holidays and work deterred me from testing this. Will do in the next few days.

kobuki, 2019-07-14:
Sorry for the delay. I can confirm the fix works as intended. I lowered the retry timeout to 60 secs but this shouldn't change the logic.

```
Jul 14 21:45:15 weather weewx[15184]: restx: Weather365: Bad SSL certificate (hostname 'channel1.weather365.net' doesn't match either of 'autoconfig.example.net', 'autodiscover.example.net'); waiting 1.0 minutes then retrying
...
Jul 14 21:46:16 weather weewx[15184]: restx: Weather365: Published record 2019-07-14 21:46:00 CEST (1563133560)
```

tkeffer, 2019-07-15:
There already is a config setting. For example

```
[StdRestful]
  [[Wunderground]]
    retry_certificate = 7200
```

I've merged this into the `development` branch. To appear in V4.

closed 2019-07-15 by tkeffer

michaelm-github, 2021-04-07:
I am getting similar errors running  weewx v4.5.1 on a Raspberry Pi.

/var/log/user.log:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: Wunderground-PWS: Unexpected exception of type <class 'ssl.CertificateError'>
/var/log/user.log:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: *** CertificateError: hostname u'weatherstation.wunderground.com' doesn't match either of '*.X.X', 'X.X'
/var/log/user.log:Apr  6 06:15:43 weatherpi weewx[1698] CRITICAL weewx.restx: Wunderground-PWS: Thread terminating. Reason: hostname u'weatherstation.wunderground.com' doesn't match either of '*.X.X', 'X.X'

I tried retry_certificate = 7200, but then weewx seems never to post anything.

tkeffer, 2021-04-07:
1. Version of Python?
2. Can you post the error from `/var/log/syslog`, instead of `/var/log/user.log`? It will show any system issues, as well as WeeWX issues.
3. When you tried `retry_certificate=7200` did you restart WeeWX? 
4. You say "weewx seems never to post anything", but can we see the log?

michaelm-github, 2021-04-07:
1. Python 2.7.16 and Python 3.7.3 are installed. weewx seems to use 2.7 based on error messages below.

2. Here is the corresponding part.

/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: Wunderground-PWS: Unexpected exception of type <class 'ssl.CertificateError'>
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: *** Traceback (most recent call last):
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/share/weewx/weewx/restx.py", line 381, in run_loop
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     self.process_record(_record, dbmanager)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/share/weewx/weewx/restx.py", line 446, in process_record
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     self.post_with_retries(_request, data)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/share/weewx/weewx/restx.py", line 475, in post_with_retries
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     _response = self.post_request(request, data)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/share/weewx/weewx/restx.py", line 537, in post_request
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     _response = urllib.request.urlopen(request, data=data_bytes, timeout=self.timeout)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     return opener.open(url, data, timeout)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/urllib2.py", line 429, in open
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     response = self._open(req, data)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/urllib2.py", line 447, in _open
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     '_open', req)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     result = func(*args)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/urllib2.py", line 1241, in https_open
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     context=self._context)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/urllib2.py", line 1195, in do_open
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     h.request(req.get_method(), req.get_selector(), req.data, headers)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/httplib.py", line 1058, in request
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     self._send_request(method, url, body, headers)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/httplib.py", line 1098, in _send_request
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     self.endheaders(body)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/httplib.py", line 1054, in endheaders
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     self._send_output(message_body)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/httplib.py", line 892, in _send_output
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     self.send(msg)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/httplib.py", line 854, in send
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     self.connect()
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/httplib.py", line 1279, in connect
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     server_hostname=server_hostname)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/ssl.py", line 369, in wrap_socket
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     _context=self)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/ssl.py", line 599, in __init__
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     self.do_handshake()
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/ssl.py", line 836, in do_handshake
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***     match_hostname(self.getpeercert(), self.server_hostname)
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: ***   File "/usr/lib/python2.7/ssl.py", line 288, in match_hostname
[... cut ...] e, ', '.join(map(repr, dnsnames))))
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] ERROR weewx.restx: *** CertificateError: hostname u'weatherstation.wunderground.com' doesn't match either of '*.X.X', 'X.X'
/var/log/syslog.1:Apr  6 06:15:43 weatherpi weewx[1698] CRITICAL weewx.restx: Wunderground-PWS: Thread terminating. Reason: hostname u'weatherstation.wunderground.com' doesn't match either of '*.X.X', 'X.X'

3. Yes, service weewx restart.

4. I will work on it as soon as I have time,

michaelm-github, 2021-04-07:
Apr  7 09:49:49 weatherpi weewx[18930] CRITICAL __main__:     ****  __init__() got an unexpected keyword argument 'retry_certificate'

tkeffer, 2021-04-07:
The option is `retry_ssl`, not `retry_certificate`.

michaelm-github, 2021-04-07:
Thank you! So far, it is working.

michaelm-github, 2021-04-08:
It still fails the same way and has not retried.

tkeffer, 2021-04-08:
I'm not an expert at this stuff, but it appears to me the problem is on your end, not the server end. Your client is unable to find a certificate that matches the identity sent back by the WU. 

The error message in your earlier post references host `'*.X.X', 'X.X'`. Did it actually say `X.X`, or did you edit it to say that?

Something you could try is to refresh your certificates.
```
sudo apt update
sudo apt install ca-certificates
```

But, again, I'm not an expert at this.

kobuki, 2021-04-08:
When you get this error, what does a browser say when trying to view https://weatherstation.wunderground.com/ ? `404 page not found` would be the correct message without any other errors, cert or otherwise. You might have a proxy or other web filter between your server and the internet that's trying to filter https content and breaks it.

tkeffer, 2021-04-08:
Good idea. Especially if you can browse from the same computer that is getting the CA error.

michaelm-github, 2021-04-08:
The X.X is a substitution I made for my own domain name. The error makes me suspect that it could be a DNS resolution error because my resolv.conf does try to append and search my own domain.

I get the appropriate 404 error when trying to browse https://weatherstation.wunderground.com/ on the weewx computer. I already had the ca-certificate package installed.

I do not understand how it posts perfectly well for multiple times, even hours or over a day, and then sometimes fails with this certificate error and dies.

kobuki, 2021-04-08:
That's weird. I'd say it's prolly a bogus dns in your resolv.conf. If you have multiple dns entries in there and resolving a name falls through to the second for whatever reason, that might do things differently. Domain prefixes are only appended when the name is unqualified, so it's either always or never, for any given host name. At least assuming the services are working fine. You could try adding the problematic WU host name in your hosts file to see if the issue comes up or not. If it doesn't, you might have a buggy dns somewhere.