weewx/weewx #414: Wunderfixer fails with HTTP Error 403: Forbidden.  Conf file is correct and live data is being uploaded.

jshatch, 2019-05-21:
It has been a while since I had to use wunderfixer, but weewx did not backfill WU when it was restarted after several hours of being down.  It did packfill PWSweather though.  Tried wunderfixer and I get the following:

```root@weather:~# wunderfixer --config /etc/weewx/weewx.conf --verbose
Using configuration file /etc/weewx/weewx.conf.
Using database binding 'wx_binding', which is bound to database 'archive_mysql'
Weather Underground Station:   KUTEAGLE26
Date to check:                 2019-05-21
Number of archive records:     800
Unable to open Weather Underground station KUTEAGLE26  or  HTTP Error 403: Forbidden
Could not get Weather Underground data. Exiting.
```

jshatch, 2019-05-21:
Forgot to mention that live data is being uploaded to WU just fine.

jshatch, 2019-05-21:
Ran it through strace and I got a better idea of what we're getting back.
```send(5, "GET /weatherstation/WXDailyHistory.asp?ID=KUTEAGLE26&month=5&day=21&year=2019&format=1 HTTP/1.1\r\nAccept-Encoding: identity\r\nHost: www.wunderground.com\r\nConnection: close\r\nUser-Agent: Python-urllib/2.7\r\n\r\n", 204, 0) = 204
gettimeofday({1558467039, 541706}, NULL) = 0
poll([{fd=5, events=POLLIN}], 1, 10000) = 1 ([{fd=5, revents=POLLIN}])
recv(5, "HTTP/1.1 403 Forbidden\r\nServer: AkamaiGHost\r\nMime-Version: 1.0\r\nContent-Type: text/html\r\nContent-Length: 315\r\nCache-Control: max-age=0\r\nExpires: Tue, 21 May 2019 19:30:39 GMT\r\nDate: Tue, 21 May 2019 19:30:39 GMT\r\nConnection: close\r\nSet-Cookie: speedpin=4G; expires=Tue, 21-May-2019 20:00:39 GMT; path=/; domain=.wunderground.com; secure\r\nSet-Cookie: ci=TWC-Connection-Speed=4G&TWC-Locale-Group=US&TWC-Device-Class=desktop&X-Origin-Hint=dna&TWC-Network-Type=wifi&TWC-GeoIP-Country=US&TWC-GeoIP-Lat=40.3099&TWC-GeoIP-Long=-111.9316&Akamai-Connection-Speed=1000+&TWC-Privacy=exempt; path=/; domain=.wunderground.com; secure\r\nProperty-id: drupal-prod\r\nX-RequestSource: AkamaiProdProxy\r\nTWC-Privacy: exempt\r\nTWC-GeoIP-LatLong: 40.3099,-111.9316\r\nTWC-GeoIP-Country: US\r\nTWC-Device-Class: desktop\r\nTWC-Locale-Group: US\r\nTWC-Connection-Speed: 4G\r\nX-RequestSource: AkamaiDefaultDNA\r\n\r\n<HTML><HEAD>\n<TITLE>Access Denied</TITLE>\n</HEAD><BODY>\n<H1>Access Denied</H1>\n \nYou don't have permission to access \"http&#58;&#47;&#47;www&#46;wunderground&#46;com&#47;weatherstation&#47;WXDailyHistory&#46;asp&#63;\" on this server.<P>\nReference&#32;&#35;18&#46;ac1e8443&#46;1558467039&#46;d5de029\n</BODY>\n</HTML>\n", 8192, 0) = 1191
```

tkeffer, 2019-05-21:
Looks like the decided to completely change their API. For example, to access my station, it used to be

```
https://www.wunderground.com/weatherstation/WXDailyHistory.asp?ID=KORHOODR3
```

but now it's

```
https://www.wunderground.com/dashboard/pws/KORHOODR3
```

Thanks a lot, WU. I love you too.

jshatch, 2019-05-21:
For fun I modified wunderfixer to just upload anyway.  It went through all of my archive records and exited cleanly, however there are no backfilled records shown on WU.

jshatch, 2019-05-21:
Anything in here you could use?  https://docs.google.com/document/d/1eKCnKXI9xnoMGRRzOL1xPCBihNV2rOet08qpE_gArAY/edit

I just tried out a few of those and they might return the needed data, but looks like a rewrite of getDayTimeStamps.

paulkaras, 2019-05-28:
Thanks JShatch for highlighting this issue - I have it too after an interruption in data uploading whilst I had a few issues updating weewx software.  Do you know if there are any plans to try to fix it or should I try to hack a solution?

jshatch, 2019-05-28:
There's a discussion here about this problem:  https://groups.google.com/forum/#!topic/weewx-user/dq8xFrqGVMA

paulkaras, 2019-05-29:
Many thanks - I shall watch with interest - happy to test the new software on a Pi when you feel it's ready.

tkeffer, 2019-07-18:
I've ported wunderfixer to the new WU API. Commit https://github.com/weewx/weewx/commit/32c35ce380b766054b05a7b64d1d82ab113a5d73

Unfortunately, it looks like the WU no longer allows re-posting old records, so the utility may no longer be useful. I'd be interested in other people's experience.

Also, as I understand it, there are issues for people living east of Greenwich. I live west, so I wasn't able to check that. Other people's experiences would be welcome.

closed 2019-07-19 by tkeffer