weewx/weewx #415: Provide support for additional file name codes in template file names

gjr80, 2019-05-22:
Currently the following codes may be used in a template file name and are replaced as indicated at time of generation:

- `DD`. 2 digit zero padded day of month
- `WW`. 2 digit zero padded week of year
- `MM`. 2 digit zero padded month number
- `YYYY`. 4 digit year

A [user has asked](https://groups.google.com/d/msg/weewx-user/EKhs6Rudm_o/bmoPeOIDBAAJ) if month name to be provided. Some time ago I have needed abbreviated month names for some WeeWX-WD files and implemented abbreviated month names by overriding `weewx.cheetahgenerator.CheetahGenerator._getFileName()`. Never bothered to try to introduce it formally though.

The following additional codes could be added:

- `MMM`. Abbreviated month name eg Jan, Feb or May
- `MMMM`. Full month name eg January, February or May
- `YY`. 2 digit year.

As long as they are processed in order (longest through to shortest) I don't see any unacceptable conflict between `MM`, `MMM` and `MMMM`. 

Month names should be locale aware, though it might be useful to be able to turn of locale awareness, maybe through use of `XXX` and `XXXX`. We could also use case sensitivity (eg `Mmm`, `mmm` and `MMM`) to provide Jan, jan and JAN. Likewise for full month name.

@tkeffer if you are happy to have this implemented I guess it is a question of how far to go down the rabbit hole? It's an easy task and I am happy to do it now.

tkeffer, 2019-11-28:
Why not just use `strftime` encoding? Something like `NOAA-%Y-%m.txt.tmpl` is a legal filename, so I don't see why it wouldn't work.

The one exception is that you could not use a slash (`/`) in the name.

gjr80, 2019-11-28:
Sounds fine to me, my first thought was to extend what we had.

tkeffer, 2019-11-29:
I removed references to legacy `YYYY`, etc., codes, which allowed simplification of the Cheetah generator section. Also, renamed the file names in `Seasons` and `Standard` to reflect. 

Let's call it done.

Thanks for your help, Gary!!

closed 2019-11-29 by tkeffer