weewx/weewx #422: make deb-package fails

KoffieNu, 2019-06-15:
When trying to build a deb package, the sources are build from version 3.9.2a1 (appears in build dir), it's tarred to weewx_3.9.2a1.orig.tar.gz and then it tries to use weewx_3.9.1.orig.tar.gz to build the deb file.
debian/weewx/DEBIAN/control rontains the old version and isn't filled with the version from the sources when building.

matthewwall, 2019-06-15:
rebuilding the debian package will do nothing to address 'blocks differ', and reinstalling weewx will do nothing to address 'blocks differ'.

first of all, you don't have to rebuild the entire debian package to make a change to the fousb driver

secondly, a change to the fousb driver will probably do nothing to address the 'blocks differ' message.  the 'blocks differ' message happens all the time.  when you read data from the fine offset stations, you'll frequently get bogus data.  so we do multiple reads, and retain the data only when the reads are consistent.  normally 'blocks differ' is just spurious, infrequent, and recoverable.  if you see them all the time (like every read) then you probably have a hardware problem, such as underpowered power supply or bad usb cable or some kind of interference.

if you think you have some code that will improve the usb reads, then just modify fousb.py directly in your existing weewx installation.  if it works, then post a diff, or do it in a clone of the weewx repository and make a merge request.

matthewwall, 2019-06-15:
if you really want to build the debian package, then please post the steps you took to build it so that we can help you do it properly.

closed 2019-06-15 by matthewwall