weewx/weewx #426: Include largest regional weather network wetternetz-sachsen.de

tho-wa, 2019-06-21:
Is it possible to enable data transfer to largest regional weather network wetternetz-sachsen.de?
Upload is based on http direct post: 
e.g.
http://www.wetternetz-sachsen.de/get_daten_21.php?var=StaiosnID;DirctPostPasswort;WNSV2.1;weewex;20:15;20.11.2016;1;-2,4;-5,6;0,2;0,3;....

Parameter list is: 
NAME    ERKLAERUNG                                             EINHEIT   FORMAT
STA_ID  StationsID                                                       Zeichenkette, 6 Zeichen
STAKEN  Stations Kennung                                                 Zeichenkette, 12 Zeichen
TMPVER  Templet-Version                                                  Zeichenkette, formatlos
WSOVER  Version Wettersoftware                                           Zeichenkette, formatlos
ZEIT__  Zeit                                                             Uhrzeit, hh:mm
DATUM_  Datum                                                            Datum, TT.MM.JJJJ
UTCDIF  Abweichung zu UTC                                                Ziffer, ggf. mit Vorzeichen
T2AKT_  2m-Temperatur, aktuell                                 °C        Dezimalzahl, 1 Nachkommastelle
T2MIN_  2m-Temperatur, Minimum heute                           °C        Dezimalzahl, 1 Nachkommastelle
T2MAX_  2m-Temperatur, Minimum heute                           °C        Dezimalzahl, 1 Nachkommastelle
T2D1H_  2m-Temperatur, Änderung in der letzten 1 Stunde        K         Dezimalzahl, 1 Nachkommastelle
T5AKT_  5cm-Temperatur, aktuell                                °C        Dezimalzahl, 1 Nachkommastelle
T5MIN_  5cm-Temperatur, Minimum heute                          °C        Dezimalzahl, 1 Nachkommastelle
LFAKT_  2m-Relative Luftfeuchte, aktuell                       %         Ganzzahl
RRD05_  Niederschlagssumme, letzten 5 min                      mm        Dezimalzahl, 1 Nachkommastelle
RRD10_  Niederschlagssumme, letzten 10 min                     mm        Dezimalzahl, 1 Nachkommastelle
RRD1H_  Niederschlagssumme, letzte 1 Stunde                    mm        Dezimalzahl, 1 Nachkommastelle
RRD3H_  Niederschlagssumme, letzten 3 Stunden                  mm        Dezimalzahl, 1 Nachkommastelle
RRD24H  Niederschlagssumme, letzten 24 Stunden                 mm        Dezimalzahl, 1 Nachkommastelle
RRD1D_  Niederschlagssumme, heute                              mm        Dezimalzahl, 1 Nachkommastelle
WSAKT_  10m-Windgeschwindigkeit, aktuell                       km/h      Dezimalzahl, 1 Nachkommastelle
WRAKT_  10m-Windrichtung, aktuell                              °         Ganzzahl
WBAKT_  10m-Böengeschwindigkeit, aktuell                       km/h      Dezimalzahl, 1 Nachkommastelle
WSM10_  10m-Windgeschwindigkeit, 10min-Mittel                  km/h      Dezimalzahl, 1 Nachkommastelle
WRM10_  10m-Windrichtung, 10min-Mittel                         °         Ganzzahl
WSMX1H  10m-Windgeschwindigkeit, Maximum letzte Stunde         km/h      Dezimalzahl, 1 Nachkommastelle
WSMX1D  10m-Windgeschwindigkeit, Maximum heute                 km/h      Dezimalzahl, 1 Nachkommastelle
WBMX1D  10m-Böengeschwindigkeit, Maximum heute                 km/h      Dezimalzahl, 1 Nachkommastelle
WCAKT_  Windchill-Temperatur, aktuell                          °C        Dezimalzahl, 1 Nachkommastelle
WCMN1H  Windchill-Temperatur, Minimum letzte Stunde            °C        Dezimalzahl, 1 Nachkommastelle
WCMN1D  Windchill-Temperatur, Minimum heute                    °C        Dezimalzahl, 1 Nachkommastelle
LDAKT_  Relativer Luftdruck, aktuell                           hPa       Dezimalzahl, 1 Nachkommastelle
LDABS_  Absoluter Luftdruck, aktuell                           hPa       Dezimalzahl, 1 Nachkommastelle
LDD1H_  Luftdruckänderung, letzte 1 Stunde                     hPa       Dezimalzahl, 1 Nachkommastelle
LDD3H_  Luftdruckänderung, letzten 3 Stunde                    hPa       Dezimalzahl, 1 Nachkommastelle
LDD24H  Luftdruckänderung, letzten 24 Stunde                   hPa       Dezimalzahl, 1 Nachkommastelle
EVA1D_  Evapotranspiration, heute                              mm        Dezimalzahl, 1 Nachkommastelle
SOD1H_  Sonnenscheindauer, letzte 1 Stunde                     min       Ganzzahl
SOD1D_  Sonnenscheindauer, heute                               h         Dezimalzahl, 12 Nachkommastelle
BEDGRA  Bedeckungsgrad                                         Octa      Ganzzahl
SSAKT_  Globalstrahlung, aktuell                               W/m2      Dezimalzahl, 1 Nachkommastelle
SSMX1H  Globalstrahlung, Maximum letzte 1 Stunde               W/m2      Dezimalzahl, 1 Nachkommastelle
SSMX1D  Globalstrahlung, Maximum heute                         W/m2      Dezimalzahl, 1 Nachkommastelle
UVINDX  UV-Index, aktuell                                      UVI       Dezimalzahl, 1 Nachkommastelle
UVMX1D  UV-Index, Maximum heute                                UVI       Dezimalzahl, 1 Nachkommastelle
WOLKUG  Wolkenuntergrenze                                      km        Dezimalzahl, 1 Nachkommastelle
SIWEIT  Horizontale Sichtweite                                 km        Dezimalzahl, 1 Nachkommastelle
SNEHOE  Schneedeckenhöhe                                       cm        Dezimalzahl, 1 Nachkommastelle
SNEDAT  Datum der letzten Schneedeckenmessung                            Zeitpunkt, TT.MM.JJJJ hh:mm
SNEFGR  Schneefallgrenze                                       m         Dezimalzahl, 1 Nachkommastelle
T2M1M_  2m-Temperatur, Monatsmittel                            °C        Dezimalzahl, 1 Nachkommastelle
T2M1MA  2m-Temperatur, Abweichung zum langj. Monatsmittel      K         Dezimalzahl, 1 Nachkommastelle
RRDATU  Datum des letzten messbaren Niederschlagsereignisses             Dezimalzahl, 1 Nachkommastelle
RRGEST  Niederschlagssumme, Vortag                             mm        Dezimalzahl, 1 Nachkommastelle
RRD1M_  Niederschlagssumme, Montassumme                        mm        Dezimalzahl, 1 Nachkommastelle
RRD1MR  Niederschlagssumme, Abweichung zur langj. Montassumme  %         Dezimalzahl, 1 Nachkommastelle
RRD1A_  Niederschlagssumme, Jahressumme                        mm        Dezimalzahl, 1 Nachkommastelle
RRD1AR  Niederschlagssumme, Abweichung zur langj. Jahressumme  %         Dezimalzahl, 1 Nachkommastelle
EVAD1M  Evapotranspiration, Monatssumme                        mm        Dezimalzahl, 1 Nachkommastelle
EVAD1A  Evapotranspiration, Jahressumme                        mm        Dezimalzahl, 1 Nachkommastelle
SOD1M_  Sonnenscheindauer, Monatssumme                         min       Dezimalzahl, 1 Nachkommastelle
SOD1MR  Sonnenscheindauer, Abweichung zur langj. Montassumme   %         Dezimalzahl, 1 Nachkommastelle
SOD1A_  Sonnenscheindauer, Jahressumme                         min       Dezimalzahl, 1 Nachkommastelle
SOD1AR  Sonnenscheindauer, Abweichung zur langj. Jahressumme   %         Dezimalzahl, 1 Nachkommastelle
KLTSUM  Kältesumme                                             Kd        Dezimalzahl, 1 Nachkommastelle
WRMSUM  Wärmesumme                                             Kd        Dezimalzahl, 1 Nachkommastelle
GRASUM  Grünlandtemperatur, aktuell                            Kd        Dezimalzahl, 1 Nachkommastelle
GRADAT  Grünlandtemperatur, Datum des erreichens                         Datum, TT.MM.JJJJ     

In case of question administrator of the network is reachable by mail: 
silvio.wandel@wetternetz-sachsen.de

Thank you.

tkeffer, 2019-06-21:
We generally write and include an uploader only after there has been considerable interest demonstrated. Yours is the first mention of wetternetz.

Of course, you can always write an uploader yourself and offer it as a WeeWX extension.

closed 2019-06-21 by tkeffer

tho-wa, 2019-06-22:
Thank you for the quick response. Is there any documentation / reference for creating uploader?

tkeffer, 2019-06-22:
Unfortunately, no, just the notes at the top of the file weewx/restx.py.

However, there are many [uploaders](https://github.com/weewx/weewx/wiki#uploaders) listed in the Wiki. Just follow the example of one of them. For example, the [windy uploader](https://github.com/matthewwall/weewx-windy) is a recent one, which will work under Python 2 or 3.

tho-wa, 2019-06-30:
Thank you. We tried to adapt the existing code for "Wetternetz Sachsen" but we have problems as we have no Python knowledge. We have added line 1316 to 1518. If some would help us we will appreciate any kind of support. 
https://github.com/tho-wa/weewx/blob/master/bin/weewx/restx.py

tkeffer, 2019-06-30:
You're better off writing this as an extension, rather than modifying `restx.py`. Start with one of the uploaders listed in the Wiki. 

If you're looking for help, try the [weewx-user](https://groups.google.com/forum/#!forum/weewx-user) group.

tho-wa, 2019-06-30:
Thank you. It think with own extension its more difficult for us without Python knowledge. 
I think we will stop here.