weewx/weewx #428: Add additional method to calculate server uptime

JeffRossMT, 2019-06-27:
The ctpes method in station.py for FreeBSD does not work on OpenBSD.  Here's a patch:

```
--- bin/weewx/station.py.orig   Mon Jun 24 13:14:38 2019
+++ bin/weewx/station.py        Thu Jun 27 08:37:37 2019
@@ -118,7 +118,19 @@
                     libc.sysctlbyname("kern.boottime", ctypes.byref(buf), ctypes.byref(size), None, 0)
                     os_uptime_secs = time.time() - float(buf.value)
                 except (AttributeError, IOError, NameError):
-                    pass
+                    try:
+                        import subprocess
+                        from datetime import datetime
+                        cmd = ['sysctl','kern.boottime']
+                        proc = subprocess.Popen(cmd,stdout=subprocess.PIPE, stderr=subprocess.PIPE)
+                        o,e = proc.communicate()
+                        time =  o.decode('ascii').split()
+                        time_as_string = time[1] + " " + time[2] + " " + time[4][:4] + " " + time[3]
+                        os_time = datetime.strptime(time_as_string,"%b %d %Y %H:%M:%S")
+                        epoch_time = (os_time - datetime(1970, 1, 1)).total_seconds()
+                        os_uptime_secs = time.time() - epoch_time
+                    except:
+                        pass
 
         return weewx.units.ValueHelper(value_t=(os_uptime_secs, "second", "group_deltatime"),
                                        formatter=self.formatter,
```

tkeffer, 2019-06-27:
Fixed in commits https://github.com/weewx/weewx/commit/fb5224388fb126a7d13fb2ded723b3ef106de7aa and https://github.com/weewx/weewx/commit/0b668f23450083e98da2cb1d4da0c3b553b5f929. I renamed variable `time` to `time_t` so it won't collide with module `time`.

To appear in V4.0

Thanks, Jeff!

closed 2019-06-27 by tkeffer

JeffRossMT, 2019-06-27:
> Fixed in commits fb52243
> and 0b668f2

Thanks for weewx Tom!