weewx/weewx #430: weewx appears to fail silently under Debian Buster on Rasp Pi 3B

pfletch101, 2019-07-02:
Buster has just been officially released for the Pi, even though it is not officially 'stable', according to Debian. I am experimenting with it, and most things seem to work as in Stretch. However, weewx (using the simulator driver) seems to fail silently, without giving any clues as to what is going on. On my initial install, I used 'feet' instead of 'foot' as the key for my station height, and did not notice that weewx had failed to start, with an appropriate error report via systemctl:
`pi@PI3-Buster:~/services $ systemctl status weewx
● weewx.service - LSB: weewx weather system
   Loaded: loaded (/etc/init.d/weewx; generated)
   Active: active (exited) since Mon 2019-07-01 22:29:23 CDT; 18h ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 2077)
   Memory: 0B
   CGroup: /system.slice/weewx.service

Jul 01 22:29:23 PI3-Buster weewx[470]:     ****    File "/usr/share/weewx/weewx/engine.
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****      self.service_obj.append(weeutil.we
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****    File "/usr/share/weewx/weewx/wxservi
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****      engine.db_binder)
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****    File "/usr/share/weewx/weewx/wxservi
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****      self.altitude_ft = weewx.units.con
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****    File "/usr/share/weewx/weewx/units.p
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****      conversion_func = conversionDict[v
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****  KeyError: 'feet'
Jul 01 22:29:23 PI3-Buster weewx[470]:     ****  Exiting.
...skipping...
`
However, after correcting this, further attempts to start weewx appear to succeed:

`pi@PI3-Buster:~/services $ sudo /etc/init.d/weewx start`
`[ ok ] Starting weewx (via systemctl): weewx.service.
`
….but absolutely nothing happens! syslog contains nothing at all from weewx, and 'systemctl status weewx' continues only to report the **original** error (from the original install time). Has anyone else tried weewx under Buster?

gjr80, 2019-07-02:
Rather than just a ‘start’ did you try a ‘restart’ or a ‘stop’ then ‘start’?

pfletch101, 2019-07-03:
No, I didn't, but I should have! 'stop' then 'start' works (even though there would appear to be nothing running to stop). Very unintuitive, but all now looks fine, at least with the simulator driver. I have a few other things to do before I try to test it with my weather station.

closed 2019-07-03 by pfletch101