weewx/weewx #431: Past hour and 24 hours rain calculations sometimes "missing" .01 inch when reporting to CWOP

ls4096, 2019-07-12:
I've noticed an occasional inconsistency when rain data is reported to CWOP. It appears that sometimes the past hour and past 24 hours rain calculations are missing .01 inch. In contrast, the "since midnight" calculation does appear to be correct (this was cross-checked with data in the weewx SQLite database "archive" table as well as the information displayed on the weather station console).

Curiously, when the queries (as found in RESTThread.get_record()) are run directly in sqlite3 against the database, the calculated values appear to be correct!

Snippets of data sent to CWOP are shown below with the suspected incorrect values in **bold**.

111505z4332.80N/07937.26W_260/003g009t079r000p000P000b10091h79.weewx-3.9.1-Vantage
111510z4332.80N/07937.26W_227/003g011t078**r000p000**P001b10093h81.weewx-3.9.1-Vantage
111515z4332.80N/07937.26W_223/004g011t078**r003p003**P004b10088h82.weewx-3.9.1-Vantage
111520z4332.80N/07937.26W_215/003g012t077r008p008P008b10083h79.weewx-3.9.1-Vantage
111525z4332.80N/07937.26W_226/003g012t076r008p008P008b10077h73.weewx-3.9.1-Vantage 

...and again a few hours later...

111830z4332.80N/07937.26W_273/007g017t078r003p012P012b10071h72.weewx-3.9.1-Vantage
111835z4332.80N/07937.26W_236/006g018t075r014p023P023b10070h77.weewx-3.9.1-Vantage
111840z4332.80N/07937.26W_176/004g018t074r019**p027**P028b10067h81.weewx-3.9.1-Vantage

tkeffer, 2019-07-12:
What kind of weather station?

If the station emits it, weewx uses the emitted values for `hourRain`, `rain24`, and `dayRain`. Otherwise, it calculates them from the database.

So, it's possible the station is emitting bogus values. Running weewx from the command line and watching the values it prints, then comparing them to the values from the database, would tell you for sure.

ls4096, 2019-07-12:
@tkeffer: Station is a Davis VP2.

Here's what I saw in regards to rain info from a few loop packets coming from the station:
- dayRain: 0.28
- monthRain: 0.48
- rain: 0.0
- rainRate: 0.0
- stormRain: 0.28, stormStart: 1562817600
- yearRain: 0.48

Although I may be missing something, the above also seems to verify with what I see in the database (starting from dateTime 1562817600):
```
sqlite> select datetime,datetime(dateTime,'unixepoch'),rain,rainRate from archive where (dateTime >= 1562817600) and (rain != 0 or rainRate != 0);
dateTime    datetime(dateTime,'unixepoch')  rain                 rainRate  
----------  ------------------------------  -------------------  ----------
1562857800  2019-07-11 15:10:00             0.00999999999999998  0.0       
1562858100  2019-07-11 15:15:00             0.03                 0.38927152
1562858400  2019-07-11 15:20:00             0.04                 0.64059602
1562858700  2019-07-11 15:25:00             0.0                  0.11622516
1562859000  2019-07-11 15:30:00             0.0                  0.05794701
1562859300  2019-07-11 15:35:00             0.0                  0.03218543
1562861400  2019-07-11 16:10:00             0.00999999999999995  0.0       
1562869800  2019-07-11 18:30:00             0.03                 0.25662251
1562870100  2019-07-11 18:35:00             0.11                 1.22953642
1562870400  2019-07-11 18:40:00             0.05                 0.86529801
1562870700  2019-07-11 18:45:00             0.0                  0.14880794
1562871000  2019-07-11 18:50:00             0.0                  0.06417218
1562871300  2019-07-11 18:55:00             0.0                  0.04258278
1562871600  2019-07-11 19:00:00             0.0                  0.00026490

sqlite> select sum(rain) from archive where (dateTime >= 1562857800) and (rain != 0 or rainRate != 0);sum(rain) 
----------
0.28      
```

tkeffer, 2019-07-12:
Maybe I'm being dense, but I don't see what the problem is. A few things to remember:

The `r` value that goes to CWOP is the rain *in the last hour*. It is not the same as the `rain` field in the database. It is the sum of the `rain` field over the last 3600 seconds.

Same with `p`, except it's the sum of `rain` over the last 24 hours.

Similarly, `P` is the sum of `rain` since midnight.

Let me know if I'm missing something.

ls4096, 2019-07-12:
I understand what you're saying about the three values, and we're in agreement there.

Maybe let me put this another way then to clarify the issue better...

One would expect that if both `p` and `P` are zero at local midnight (in my case, this is currently 0400Z, since it's relevant to the data I've posted), then until the next local midnight these two values ought to be equal at any given moment regardless of what rain does or doesn't fall during that time interval. Hopefully we're in agreement about this assumption so far.

Now, given that, in my CWOP data I see instances where this assumption is violated (where I bolded the values). As for `r` being incorrect in some cases, I just meant to highlight the fact that this value is sometimes *also* (as with the value of `p`) inconsistent with what one would expect based on the current and recent/changing values of `P` (which I've determined to be "correct" based on the data that the hardware is providing).

tkeffer, 2019-07-12:
There might be a rounding problem. In your database, instead of 0.01 for `rain` we are seeing 0.00999999999999998. 

Because the formatting for CWOP is done as

    "%03d" % (rain * 100)

this would give a value of `r` of  `000`, when `001` would be more accurate. It should be formatted as

    "%03d" % int(rain * 100 + 0.5)
   
There would be similar rounding issues with `barometer` and other types.

Commit https://github.com/weewx/weewx/commit/88995afa7df16c4ad704597a6d5f4a35a73667ce fixes the problem. Try downloading its version of [restx.py](https://raw.githubusercontent.com/weewx/weewx/master/bin/weewx/restx.py) and see if that fixes the problem.

Thanks for your patience!

ls4096, 2019-07-12:
@tkeffer: Thanks! I really appreciate the investigation and fix with such a quick response time.

Just a few small comments I left on the commit where I think there may be an issue (for example, when humidity is >=99.5 but <100.0), but the change is otherwise working well for me (verified on original data -- I see the expected .28" rainfall value being set for `p` and `P` now for that time interval).

ls4096, 2019-07-13:
I consider this issue fixed now, and it can be closed.

@tkeffer: Also wanted to say I really enjoy WeeWX. Thanks again!

closed 2019-07-13 by ls4096