weewx/weewx #433: wunderground again (sigh)

RPeters52, 2019-07-17:


gjr80, 2019-07-17:
Suggest you have a search of the forums for ‘wunderfixer’ over the past few months. I believe Leon may have done some work and created a workaround; however, I don’t believe any of that work has yet been committed here (and hence the yet to be released next version of WeeWX you have does not include the workaround).

RPeters52, 2019-07-17:
Thanks!  I will check it out.

closed 2019-07-17 by tkeffer

RPeters52, 2019-07-17:
So, I downloaded the development branch wunderfixer and tried it
File last mod:
1.4.1 05/14/2019
Made WunderStation class consistent with restx.AmbientThread parameters

and I ran it to get:
wunderfixer -e250 -t
Using configuration file /etc/weewx/weewx.conf.
Using database binding 'wx_binding', which is bound to database 'archive_sqlite'
Traceback (most recent call last):
  File "/usr/bin/wunderfixer", line 512, in <module>
    main()
  File "/usr/bin/wunderfixer", line 223, in main
    softwaretype = "wunderfixer-%s" % __version__)
TypeError: __init__() got an unexpected keyword argument 'q'

The section that fails is:

    # Get a WunderStation object so we can interact with Weather Underground
    wunder = WunderStation(q=None,  # Bogus queue. We will not be using it.
                           manager_dict=dbmanager_t,
                           station=options.station,
                           password=options.password,
                           server_url=weewx.restx.StdWunderground.pws_url,
                           protocol_name = "wunderfixer",
                           essentials=_essentials_dict,
                           softwaretype = "wunderfixer-%s" % __version__)

Oh well....

gjr80, 2019-07-17:
I think you will find that none of the workarounds that make wunderfixer work with the ‘new’ WU API have been committed to the WeeWX github repo. The issue you have struck with the development branch may have to do with 3.9.2 incompatibility with the development branch version of wunderfixer, it doesn’t look like a problem with WU per se. That is one of the dangers you run into using development. You may get better mileage asking in [this thread](https://groups.google.com/forum/m/#!topic/weewx-user/dq8xFrqGVMA/discussion).

RPeters52, 2019-07-17:
Thanks for the response!!  I guess that I can wait until someone figures out this mess...  

One of the rules that I had when writing software is that "interfaces" change only if absolutely necessary.  One can add new/modified interfaces but one should try to keep the old ones viable...  Otherwise chaos....