weewx/weewx #434: MQTT with auth only sending UV

jdetmold, 2019-07-27:
Running weewx 3.9.1 with mqtt extension 0.19 with Mosquitto
everything works great when I do not use mqtt authentication however as soon as I add auth things go a bit sideways. logs show that mqtt is being sent however all that is received is UV as soon as I remove auth from the config everything starts working again.

tkeffer, 2019-07-27:
Sorry, but mqtt is not part of weewx. You'll have to ask the author. Or, try the [weewx-user group](https://groups.google.com/forum/#!forum/weewx-user).

closed 2019-07-27 by tkeffer

jdetmold, 2019-07-27:
how does one find the author? I thought the extension was part of weewx because it's documentation is in the wiki https://github.com/weewx/weewx/wiki/mqtt
I would recommend if extensions are made by external people but in the wiki perhaps putting a link to the extensions github in the docs?

thanks