weewx/weewx #435: Update Wunderground uploader to include new parameters

tkeffer, 2019-07-29:
The recently republished protocol for the WU includes a few new parameters. https://feedback.weather.com/customer/en/portal/articles/2924682-pws-upload-protocol?b_id=17298

tkeffer, 2019-08-22:
Some of the new fields have to do with air quality, which are not part of the wview schema used by WeeWX. Nevertheless, I'd like to standardize the names. Looking around at some of the user-supplied drivers and services, I'm seeing lots of variants. I propose we settle on the pattern set by the Weather Underground:

```
AqNO - [ NO (nitric oxide) ppb ]
AqNO2T - (nitrogen dioxide), true measure ppb
AqNO2 - NO2 computed, NOx-NO ppb
AqNO2Y - NO2 computed, NOy-NO ppb
AqNOX - NOx (nitrogen oxides) - ppb
AqNOY - NOy (total reactive nitrogen) - ppb
AqNO3 -NO3 ion (nitrate, not adjusted for ammonium ion) UG/M3
AqSO4 -SO4 ion (sulfate, not adjusted for ammonium ion) UG/M3
AqSO2 -(sulfur dioxide), conventional ppb
AqSO2T -trace levels ppb
AqCO -CO (carbon monoxide), conventional ppm
AqCOT -CO trace levels ppb
AqEC -EC (elemental carbon) – PM2.5 UG/M3
AqOC -OC (organic carbon, not adjusted for oxygen and hydrogen) – PM2.5 UG/M3
AqBC -BC (black carbon at 880 nm) UG/M3
AqUV-AETH  -UV-AETH (second channel of Aethalometer at 370 nm) UG/M3
AqPM2.5 - PM2.5 mass - UG/M3 
AqPM10 - PM10 mass - PM10 mass
AqOZONE - Ozone - ppb
```

Any comments?

matthewwall, 2019-08-23:
there are three particulate numbers: 1.0, 2.5, and 10.0.  usually you see 2.5 and 10.0, but there are some sensors that do 1.0.

in the 'schema' branch i used the names pm1_0, pm2_5, and pm10_0.

https://github.com/weewx/weewx/blob/schema/bin/schemas/wview_extended.py

these are also being tracked in the rtl_433 codebase, but there they have units in the names, for example: pm10_0_ug_m3

https://github.com/merbanan/rtl_433/commit/e9452cfcdf3f2472b39b6dd41c839e0ad3243699

tkeffer, 2019-09-02:
I ended up going with the names in the existing extended schema. So, we have:

```python
        'co': 'AqCO=%f',
        'no2': 'AqNO2=%f',
        'o3': 'AqOZONE=%f',
        'pm10_0': 'AqPM10=%.1f',
        'pm2_5': 'AqPM2.5=%.1f',
        'so2': 'AqSO2=%f',
```

Closing issue.

closed 2019-09-02 by tkeffer