weewx/weewx #438: Davis Vantage Vue wind direction lost in accumulator, ok in LOOP

HQJaTu, 2019-08-01:
I'm new to WeeWx-project and my Vantage Vue has been running for roughly two weeks.

For some reason my wind direction data gets archived incorrectly as 337.5 degrees.
```
$ sqlite3 /var/lib/weewx/weewx.sdb 'select `dateTime`,`outTemp`,`windSpeed`,`windDir`,`windGust`,`windGustDir` from archive'
1564650000|56.2|3.0|337.5|10.0|337.5
1564651800|56.4|3.0|337.5|11.0|0.0
1564653600|56.8|3.0|337.5|8.0|337.5
1564655400|56.4|4.0|337.5|10.0|22.5
1564657200|56.5|2.0|337.5|8.0|0.0
1564659000|56.2|4.0|337.5|8.0|0.0
1564660800|56.6|4.0|337.5|10.0|0.0
1564662600|56.4|5.0|337.5|12.0|0.0
1564664400|56.1|4.0|337.5|11.0|315.0
1564666200|56.1|3.0|337.5|11.0|0.0
1564668000|56.3|2.0|337.5|7.0|0.0
1564669800|56.6|2.0|337.5|9.0|0.0
1564671600|56.2|4.0|337.5|10.0|0.0
1564673400|56.4|3.0|337.5|8.0|0.0
1564675200|56.6|3.0|337.5|8.0|0.0
1564677000|56.1|4.0|337.5|9.0|337.5
1564678800|55.9|4.0|337.5|10.0|0.0
1564680600|55.9|3.0|337.5|7.0|0.0
```

At the same time the loop packets indicate a realistic value for wind direction:
```
Aug 01 21:00:42: windDir: 330.0
Aug 01 21:00:42: windDir: 319.0
Aug 01 21:00:50: windDir: 328.0
Aug 01 21:00:58: windDir: 309.0
Aug 01 21:01:06: windDir: 318.0
Aug 01 21:01:14: windDir: 325.0
Aug 01 21:01:22: windDir: 307.0
Aug 01 21:01:30: windDir: 314.0
```

Something weird happens and the direction information is lost during accumulation/archiving. Given a new station, I don't have much archive data, but it is impossible for the wind direction to be exactly same on all records.

Any ideas what to look for?

gjr80, 2019-08-01:
A few thoughts to start off with. At the moment you appear to be comparing apples and oranges. The archive values for windDir are average values over an archive period. In your case your archive period is set to 1800 seconds or 30 minutes (the out of the box Davis default). The loop data you present covers 48 seconds. A proper comparison needs to look at the loop windDir values over a full archive period at least.

Also, what sort of archive record generation are you using; software (which uses the accumulator to synthesise an archive record from loop data) or hardware (where the station provides the archive record). I suspect you are using the default hardware record generation in which case the accumulator has nothing to do with the windDir data stored in the archive.

Further, archive records generated by Davis stations provide discrete windDir values that are multiples of 22.5 degrees (not 1 degree as per loop packets). So if your station is seeing predominantly north west’ish wind (as evidenced by your loop data) it is quite possible that hardware generated archive records will have all or predominantly 337.5 degree values. I would be more concerned if you had predominantly easterly wind and you had 337.5 in the archive for windDir 

At the moment I see nothing wrong here.

Gary

PS. Probably not something that justifies an issue being raised (at this stage), github issues are usiually kept for dealing with bugs/changes to the WeeWX code base, user question are more better placed in the weewx-user google group.

closed 2019-08-01 by tkeffer

HQJaTu, 2019-08-02:
Thank you for your input.

On wxforum.net I wrote:
> I calculated the average wind direction from all LOOP-packages I have logged and it is 227 degrees.
This differs quite a lot from what I get archived. I'm getting a constant 337.5.

Your complaint about the archived value of
```
1564680600|55.9|3.0|337.5|7.0|0.0
```
and the missing LOOP-values is valid. If I would have added all the appropriate log entries, it would have been obvious to you that the average direction from those values does not add up as 337.5 degrees.

Unfortunately my system's journald doesn't contain the LOOP-values for above archive ranges anymore so it is impossible to provide you the proof at this point of time.

Meanwhile, I'll keep debugging the problem. Something funny happens with accumulation.

tkeffer, 2019-08-02:
Jari,

1. You still have not told us whether your installation uses hardware or
software record generation. As Gary says, if you are using hardware record
generation (the default), then the values you are seeing are coming
directly from your console logger and have nothing to do with the
accumulators. Stop weewx, set debug=1, restart, let it run through the
first reporting cycle, then post the resultant log. The log will record
what kind of record generation you are using.

2. Run weewx directly from the command line. It will print out the LOOP
packets and the archive records. What value does it show for windDir in the
archive records?

-tk

HQJaTu, 2019-08-02:
Well, the ticket was already closed when I first saw any comments in it. So, let’s keep it that way.

tkeffer, 2019-08-02:
Ok. But I'd still like to resolve this.

I just don't think it's a bug.

tkeffer, 2019-08-11:
Jari?

Did you resolve this?