weewx/weewx #443: Some missing fields may not be calculated during a `wee_import` session

gjr80, 2019-09-15:
During an import if calculation of a missing field relies upon historical data (eg `ET`, `pressure`) that exists in the tranche but is not yet saved to archive, the field will not be calculated and will be set to `None`. PR #440 attempted to fix this by saving the tranche to archive before the missing fields are calculated. However, this approach will fail as the WeeWX API `addRecord()` method uses an SQL `INSERT` statement to add the record and any subsequent attempt to 'add' a record with the same timestamp (`dateTime` field) will result in a duplicate primary key error.

gjr80, 2019-09-15:
One approach is to keep the current code but force the tranche size to 1 if a missing field that requires historical data is to be calculated. This will make any such imports very slow.

tkeffer, 2019-09-16:
I wonder if it would be simpler to do things in two, independent passes.

The first pass would insert all the independent data.

A second, later, pass would then do updates on ET and pressure. (Incidentally, one can do updates on more than one variable in a single `UPDATE` statement).

gjr80, 2019-09-16:
After looking through the original PR the first thing I thought of was a two pass solution (anything less than a two pass solution was going to involve obtaining historical data from both the archive and tranche, simple enough in theory but I suspect complex in implementation). I did start coding the two pass approach but when I realised I would need to write/rewrite some code to update a record and recalculate (some) daily summaries I stopped to consider whether there may be a better approach. So far the only approaches I have though of are:

1. Force tranche size = 1. Easy to code but expect this will be very slow for moderate to large imports.
2. Two pass solution. Need new/updated `updateValue()` method to handle multiple field updates at once as well updating the relevant daily summaries. More complex to code, likely faster than 1 but slower than 3.
3. One pass approach. No need to change the underlying code for inserting data into the database but will need a different approach to calculating the missing data. Either a new set of missing data calculations for `wee_import` (wasteful duplication) or sub-classing `WXCalculate()` and overriding some methods (more palatable but still some duplication). Probably fastest execution but complex and some serious duplication drawbacks.

I think 1. is the approach of last resort. Both 2. and 3. require development of some significant new functionality but I think 2. would be simpler and the new `updateValue()` code may have other utility.

Will go back to my two pass code and see what I can do. The approach I was thinking of was:

1. rework the `updateValue()` method to handle multiple fields at once
2. give class `DaySummaryManager()` an `updateValue()` method that is multi field and daily summary aware.

tkeffer, 2019-09-17:
Think of the second pass of the two-pass approach as something that `wee_database` could do. 

It could be implemented as a separate method (much like rebuild-daily), which could then be kicked off by either `wee_database` or `wee_import`.

tkeffer, 2019-09-17:
> The --calc-missing action in wee_import would then result in a call to this new wee_database 
> --calc-missing action with the span of the action being set to the period of the import.

I'm not sure what you mean by an "action." They should share the same code --- either could call the necessary functions. But, `wee_import` should not kick off a `wee_database` process.

gjr80, 2019-09-17:
Each WeeWX utility has actions and options (refer Utility Guide structure for each utility), not sure there is a clear definition other than the obvious. Perhaps a poor choice of words on my part.

I envisioned a new class subclassed from `weecfg.database.DatabaseFix` (let's call it `weecfg.database.CalcMissing`) that would calculate missing derived obs, basically you give it a config dict and a timespan and it calculates whatever missing derived obs that it can over the timespan saving any calculated obs to archive and daily summaries as necessary.

Case 1. User performs an import without `--calc-missing`. The entire import is performed with no calculation of missing derived obs.

Case 2. User performs an import with `--calc-missing`. The entire import is performed with no calculation of missing derived obs. Once the basic import is complete `wee_import` obtains a `weecfg.database.CalcMissing` object and calls its `run` method which then causes any missing derived obs to be calculated and saved to database.

Case 3. User wants to add any missing derived obs from an earlier import. The user runs `wee_database` with the `--calc-missing` action and any `--date` or `--from` and `--to` options as necessary. `wee_database` then obtains a `weecfg.database.CalcMissing` object and calls its `run` method which then causes any missing derived obs to be calculated and saved to database.

In case 2 I don't see `wee_import` kicking off a `wee_database` process as such, rather `wee_database` and `wee_import` are sharing a common action through common code.

tkeffer, 2019-09-17:
Exactly what I'm thinking.

gjr80, 2019-09-17:
OK, sounds like a plan. Will get the `wee_import` documentation updated to reflect the new WU API usage and then see where this takes me.

gjr80, 2019-09-20:
Have started from the `wee_database` side and implemented a `--calc-missing` action. My first cut code appears to be working, no idea about speed yet due to the small data sets I am using but it has thrown up one interesting question.

The `wee_database --calc-missing` action is only meaningful for fields that exist in the current schema. In other words if `humidex` is not in your schema then there is no point in running `wee_database --calc-missing` to populate `humidex` as `humidex` is not going to be saved to database. When I extract a record from the archive with `genBatchRecords()` any fields that have never had a numeric value saved are returned as `None`, for example `appTemp`. If I now have `class WXCalculate` calculate any missing derived fields for that record `appTemp` is skipped because the field already exists, albeit with a value of `None`. Now I could force `WxCalculate` to calculate any fields it knows about that are `None`, but how do you distinguish between a field that has never been used (ie one you do want to calculate and backfill) and one that legitimately had `None` stored (ie you don't want to calculate and backfill).

I suspect that when coming from `wee_import` a similar issue will occur because we are saving records to archive then augmenting.

I guess we are only talking about derived fields so it's not likely you would have saved 'None/null' unless pre-requisites are missing. If they are still missing we will still get/store `None/null` otherwise we get a value. So maybe the appropriate action is to force `WXCalculate` to calculate all derived fields it knows about. Worst case we save `None/null` in place of an existing `None/null`, best case we obtain an actual derived value.

tkeffer, 2019-09-20:
There is a precedence for this: right now, `StdWXCalculate` will calculate a derived field not only if it is missing completely, but also if it has a value of `None`. 

I don't see any problem with trying to fill in as many `None` values as possible all the time. Indeed, I could see a case where a better formula comes along, so the user wants to recalculate all derived variables. So, s/he sets all the target values in the database to `None`, then runs `--calc-missing`.

gjr80, 2019-09-20:
> I don't see any problem with trying to fill in as many `None` values as possible all the time. Indeed, I could see a case where a better formula comes along, so the user wants to recalculate all [...]

I think I had too many things spinning in my head and actually writing it down helped me come to the same realisation.

> There is a precedence for this: right now, `StdWXCalculate` will calculate a derived field not only if it is missing completely, but also if it has a value of `None`.

Yes, um, ah, slightly red faced, a logic error was masking what was really gong on. Was looking at what additional fields were added to the record by `WXCalculate` when I should have been looking at the fields whose keys are `WXCalculate.calculations.keys()`. Works as expected now.

Speed is the issue now, 2 days of 5 minute interval took 90 seconds to `--calc-missing`, need to look at the update query transactions, that is the bottle neck.

gjr80, 2019-10-09:
Philosophical question. 

Both `wee_import` and `wee_database` use a `--to` and `--from` command line parameter. In both cases these parameters essentially do the same thing; allow the user to specify a limited time domain over which the utility will operate. There is a difference though, `wee_import` uses the format `YYYY-mm-dd[THH:MM]` whereas `wee_database` uses the format `YY-mm-dd`. Not a problem as the two utilities perform different and unconnected functions.

However, the revised calc missing arrangement will see `wee_import` and `wee_database` both be able to calculate missing derived fields but at the moment the `--to` and `--from` formats will be different. Calculating missing derived fields does need the fine control of `THH:MM`, do we give `wee_database` the ability to accept an optional `THH:MM` or leave `wee_database` as is? I think the former; `THH:MM` is optional so no problems there, it will have no effect on rebuilds plus it keeps the same format in both utilities. Of course the other approach is to drop `--calc-missing` from `wee_database`

tkeffer, 2019-10-09:
I agree. If adding `THH:MM` does not compromise the mission of `wee_database`, I don't see any problem with allowing it.

gjr80, 2019-10-19:
Text of an email exchange brought regarding `lastUpdate` meta data value when performing a daily summary rebuild after an import:

From: Thomas Keffer
To: Gary Roderick 
Cc: mwall 
Subject: Re: wee_import and WU
Date: Fri, 18 Oct 2019 16:49:03 -0700

I have a curious feeling of deja vu all over again. Didn't we have a similar problem when we refactored backfill_day_summary() to allow partial rebuilds? We must have had some good reason to enforce the semantics of lastUpdate.

Another option (which you eluded to) is to use Manager.addRecord(), which would not touch the daily summaries. The danger is that it can leave the database out of whack, with no indication that anything is wrong.

Still another option, which would be available if this back calculation will only be available in V4, is to change the backfill_day_summary() API so there's an option to not touch lastUpdate

On Fri, Oct 18, 2019 at 3:07 PM Gary Roderick  wrote:
> I wouldn't worry about it. Python 2.6 is becoming very rare.

Yep, too hard and not enough time.

I mentioned I've run into an issue, it's to do with the lastUpdate meta value. In order to divorce the 'calc missing' aspects of an import from the actual import of the data I have a new CalcMissing class that does two things: firstly it calculates any missing derived fields over a timespan and secondly it calls manager.DaySummaryManager.backfill_day_summary() to recalculate the daily summaries over the timespan concerned. wee_import now imports all records (without calculating any missing derived fields) then the CalcMissing class works on the import timespan to calculate any missing fields and rebuild the daily summaries.

The problem is the import process uses manager.DaySummaryManager.addRecord() to add records to the archive, this also has the effect of updating the daily summaries and setting the lastUpdate meta value. If the span of records imported is before the timestamp of the last good record when CalcMissing comes to rebuild the daily summaries the lastUpdate meta value will almost certainly be less than the last good timestamp and this causes manager.DaySummaryManager.backfill_day_summary() to raise a violated precondition error since it is likely rebuilding over a span.

Rebuilding the entire daily summaries with wee_database takes care of the daily summaries but we have now rebuilt all daily summaries rather than just the span of imported records.

Ways around:
1. rebuild the entire daily summaries rather than just the import span
2. fudge the lastUpdate meta value in class CalcMissing knowing that the rebuild will then proceed and manager.DaySummaryManager.backfill_day_summary() will set lastUpdate as required
3. when adding imported record to the archive and if missing derived fields are to be calculated somehow only add records to the archive rather than adding to the archive and updating the daily summaries (initially thought this would be easy to do by calling a parent function but I am not sure this will be so easy.

I think it is reasonable to be able to limit the rebuild rather than always doing an entire rebuild so not keen on 1 though it is easiest to implement. 2 feels wrong but is also easy to implement and I can't see it causing a problem if something goes wrong, a full rebuild will always get things back to normal. 3 is probably the best solution (will give a speed improvement as well) but I don't see an easy way to implement it.

Have I missed anything here?

Gary

gjr80, 2019-10-19:
> Still another option, which would be available if this back calculation will only be available in V4, is to change the backfill_day_summary() API so there's an option to not touch lastUpdate

This will only be used under version 4 (the solution to the immediate WU API import for v3 users is to clone the development branch and use wee_import - which is essentially the same as running v4) so I think we are good on that point.

Thinking more about this approach, on entry `backfill_day_summary()` does a check on `lastUpdate` and under certain conditions a `ViolatedPrecondition` exception is raised and the rebuild aborted. If the rebuild proceeds `backfill_day_summary()` updates `lastUpdate`. So does 'not touch `lastUpdate`' mean skip both the `lastUpdate` check and updating of `lastUpdate` in `backfill_day_summary()` or just the latter? Just doing the latter would not appear to achieve much as by the time `backfill_day_summary()` is called the `lastUpdate` damage has already been done.

If we leave `lastUpdate` untouched are there any corner cases where we are setting ourselves up to fail? I don't think so, if `lastUpdate` was already set to last good timestamp then thatis fine. If `lastUpdate` was set to some early timestamp then leaving it as it is not making the situation any worse/different. A '`lastUpdate` issue` can always be fixed with a complete rebuild.

tkeffer, 2019-10-19:
As I recall, we introduced `lastUpdate` to guard against the possibility of an aborted backfill leaving the database half backfilled.

So, back to your problem: it's a little different. We are just filling in the middle part of the daily summaries. So long as each day's update is done as a single transaction, it should be OK. 

But, we might be missing something.

gjr80, 2019-10-19:
OK, thanks, will see what I come up with.

gjr80, 2019-10-24:
Just to update. As of commit a02488122f71c8ac070303d8deb5c399a2535b2f `wee_database` supports recalculation of missing derived observations using the `--calc-missing` action. To support the timespan granularity required by the `--calc-missing` the `--to` and `--from` options have been reworked to support the `yyyy-mm-dd[THH:MM]` format.

Have also slightly modified the behaviours of the `--to` and `--from` options such that if only `--to` is provided the action concerned is applied from the first good record through to the date-time specified by the `--to` option. Similarly, if only the `--from` option is specified the action concerned is applied from the date-time specified by the `--from` option through to the last good record.

gjr80, 2019-10-30:
Updated `wee_database` section of `Utilities Guide` at commit 07c15fd1ae6073bb9d145e28ed02f7d4458a51c7. Documentation of the usage of the various `--date`, `--from` and `--to` combinations across `--rebuild-daily` and `--calc-missing` is somewhat wordy but necessary to so that users understand what days/records are affected. Would welcome review of the `--rebuild-daily` and `--calc-missing` paras of the `wee_database` section of the `Utilities Guide`.

gjr80, 2019-11-16:
Verifying `calc-missing` operation with the xtypes code has caused me to have to review which fields `calc-missing` will calculate. My concern is for fields marked as `hardware`; if a field is marked as `hardware` should we calculate the field? The `StdWXCalculate` approach is to only calculate if such fields are missing or == None. At the moment `calc-missing` will calculate such fields irrespective of its appearance or value. I believe we should mimic the `StdWXCalculate` approach?

tkeffer, 2019-11-17:
Damn, you ask some hard questions! :-)

I assume in this context the `hardware` option means accept whatever value appears in the import data stream? If that's the case, then, yes, we should mimic the `StdWXCalculate` approach.

gjr80, 2019-11-17:
> Damn, you ask some hard questions! :-)

Sorry, it felt somewhat more acceptable putting this to you via an issue rather than emailing direct. Doesn't change the question though.

> I assume in this context the `hardware` option means accept whatever value appears in the import data stream? If that's the case, then, yes, we should mimic the `StdWXCalculate` approach.

I think it is slightly more complicated than that. We now have this standalone (well separately callable via `wee_database` and called during an import via `wee_import`) calc-missing capability that will look at a record and calculate derived obs. So the record that is being operated on could be an existing archive record from year dot that has some missing derived obs or the record could be an imported record with some missing derived obs. Either way I agree we should mimic the `StdWXCalculate` approach.

I will make it so.

gjr80, 2019-11-25:
Another philosophical question/statement. As far as I am aware whenever any of the WeeWX utilities makes a change to the database we confirm the user's intent before proceeding. I expect that calc missing is not different even it is possible no changes will be made.

tkeffer, 2019-11-25:
Agree.

tkeffer, 2019-11-27:
Can you do the honors of closing this if you're satisfied with the results?

gjr80, 2019-11-27:
I'm happy that the `wee_database --calc-missing` action works fine (#363) but give the recent changes to that code I would like to go back to `wee_import` and make sure it is still functioning correctly. It was before and it should still but it neds to be checked. Also need to check the `wee_import` section of the Utilities Guide is still correct, again already touched up but some example outputs may need refreshing.

Couple of days....

tkeffer, 2019-11-27:
I appreciate your diligence!

I would strongly suggest coming up with some sort of test suite to check wee_import. Create a synthetic database (you can use `gen_fake_data()` for this), export to CSV, then import, compare the results.

If you want to get fancy, you can use `mock` to create a fake WU website, which you then import.

gjr80, 2019-11-27:
> I would strongly suggest coming up with some sort of test suite to check wee_import. Create a synthetic database (you can use `gen_fake_data()` for this), export to CSV, then import, compare the results.

Yep, already have that on the list, just wanted to try to clear this and other WeeWX issues first.

> If you want to get fancy, you can use `mock` to create a fake WU website, which you then import.

Hmm, sounds like the Rolls Royce solution, might have to start with the Morris Minor solution first :)

gjr80, 2019-12-06:
Having an issue with `lastUpdate` metadata when rebuilding the daily summaries after an import. When the rebuild is commenced `DaySummaryManager` complains about `lastUpdate` being less then the timestamp of the last good record.

I think what is happening is that when records are imported that all earlier than the most recent record in the archive the `lastUpdate` metadata field is left set to the timestamp of the most recent record that was imported. When the daily summary rebuild occurs this causes the rebuild to abort.

Why is this happening now and not before? Previously we did not explicitly rebuild the daily summaries, we relied on `addRecord` and friends to update the daily summaries as we went. So `lastUpdate` never affected us in the same way (though it would have been left set to the same timestamp of the most recent record imported - this may or may not cause a problem depending on what the user did next - if they just carried on running WeeWX then no problem and `lastUpdate` would be fixed). Now we do the import (which sets `'lastUpdate` and implicitly incrementally updates the daily summaries), then we do the calc-missing then we do an explicit rebuild of the daily summaries which is where we get bitten.

How to fix? One solution would be to have `wee_import` set the `lastUpdate` metadata field to `max(lastUpdate, last good timestamp)` once the import finishes. Another approach could be to modify the logic up front of `backfill_day_summary` but that sounds dangerous, risky and messy.

Any thoughts?

Oh, and by the way I have now satisfied myself that `wee_import` and the `calc-missing` code is indeed working together as it should (had a nagging doubt it was not). So really just need to fix this and finish the Utilities Guide updates.

tkeffer, 2019-12-06:
It's kind of hazy history now, but as I recall, the sole job of `lastUpdate` was to detect aborted backfills: if its value was less than the last good record, then the user must have aborted. Makes me wonder if we should have treated it as a temporary variable to be deleted after a successful rebuild. I'm not prepared to start doing that, but it does help frame its role.

I didn't completely follow your description of the import process, but it sounds like the right thing to do is to have `wee_import` set `lastUpdate` to the last record when it's done.

gjr80, 2019-12-06:
I was racking my brain trying to remember where I was last caught by `lastUpdate`. Having thought overnight I think it was when we were patching the database with the `windSpeed` fix or the interval weighting fix. Can't remember the details though.

The revised `wee_import` is basically a three step process. 

First step is to save the imported data to archive. This uses the `addRecord` method to add/commit tranches of records to archive. As part of the `addRecord` call the `lastUpdate` field is updated. If I read the code correctly once the import to archive is finished the `lastUpdate` field will take on the value of the timestamp of the youngest record imported. 

The second step (if requested by the user) is to calculate any missing derived obs. This step consists of two sub-steps; firstly the derived obs are calculated and respective archive records updated (via a new 'update' method so `lastUpdate` is not touched). The second sub-step is to rebuild the daily summaries via a call to `backfill_day_summary`. This is were we run into problems because of the value that was left in `lastUpdate` during the initial import.

I will have `wee_import` set `lastUpdate` on completion, thinking some more I will set it to the greater of the timestamp of the youngest record imported and the value of `lastUpdate` when `wee_import` was started.

gjr80, 2019-12-14:
A trap for young players. Seems when I first started down this path I decided to perform calc-missing on tranches of whole days at a time (rather than tranches of so many records at a time). I can only surmise this was to align with daily summaries. I found that 10 days at a time worked fine. Well that decision came back to bite when I tried importing some Cumulus monthly log files that had 1 minute intervals. Instead of calculating missing obs for 10 * 288 = 2880 records per db transaction (5 minute interval) I was doing 10 * 3600 = 36000 per transaction. Things ground to a halt and a full day of processing still did not complete importing 140000  odd records. I did check htop and admittedly only looked at CPU usage and it was not bad, can't recall memory usage but nothing really stood out. Perhaps something else was going on on my desktop.

Anyway dropped 10 days to 1 day and much improved. Even though it is a fairly simple chnage to go from tranches of days to tranches or records I will leave the 1 days tranche in there for now for the 4.0 release, it works fine, plus I don't have the time to go back and do all the testing from scratch that would be required for such a fundamental change. The change to tranches of records can wait for a later release. Will raise another issue so I don't forget.

tkeffer, 2019-12-15:
I think that's fine. For people with five minute archives, the difference in processing time would be very small. Besides, it's a one time cost.

gjr80, 2019-12-16:
OK, despite a couple of last minute python3 issues with WU imports (I blame WU, I think they have changed their behaviour since I last worked on the WU code) I think I have this one complete. I have tested under current 4.0.0b5 using both python2 and python3 and all four supported import sources (CSV, WU, Cumulus and WD). In summary:

- `--calc-missing` action implemented in `wee_database`
- `wee_database` documentation updated
- `wee-import` now imports all data then calls `calc-missing` (and then rebuilds the daily summaries)
- `wee_import` documentation updated
- change log has been updated (you might want to check the entries)

I am happy for this issue to be closed.

tkeffer, 2019-12-16:
Seems like only yesterday you opened it! :-)

closed 2019-12-16 by tkeffer