weewx/weewx #445: wunderfixer fails to work correctly for today for stations in east of Greenwich when local date is ahead of the GMT date.

rodyager, 2019-09-18:
There is a problem with the WU API which only affects stations that lie east of the Greenwich meridian, and only while the local date is ahead of the GMT date. (eg. I am in Sydney Australia, which is 10 hours ahead of GMT so this problem only occurs between midnight and 10am local time for me.)

A patch for wunderfixer (development branch) to fix this is attached.

Rod
[wunderfixer.txt](https://github.com/weewx/weewx/files/3624464/wunderfixer.txt)

tkeffer, 2019-09-18:
Thanks, Rod! Fixed in commit https://github.com/weewx/weewx/commit/d1c3abee6f7357119276907b0de8d949a71587dc

closed 2019-09-18 by tkeffer

rodyager, 2019-09-18:
It looks like I took long to submit this patch as WU has now fixed the behaviour of their API for stations east of Greenwich, and the patch is no longer necessary.

You should revert the change in the interests of code simplicity and readability.

Sorry for this bother.

Rod

Thanks, Rod! Fixed in commit d1c3abe<https://github.com/weewx/weewx/commit/d1c3abee6f7357119276907b0de8d949a71587dc>

tkeffer, 2019-09-18:
Really? If you're sure, I'll revert the change.

rodyager, 2019-09-18:
I’d been running this patch for quite a while, and duly applied it to my local copy when I upgraded to 4.0.8a yesterday. That was what prompted me to submit it.

It was only after I had submitted it that it ocurred to me that I should perform the sanity check of verifying that it was still required. (The history was that WU had more substantial problems with this API that were fixed, leaving just the case of stations east of Greenwich that had already transitioned to the next day as the one edge case they had not addressed. This morning’s testing shows that this has now also been addressed.)

Rod

tkeffer, 2019-09-18:
OK. Reverted in commit https://github.com/weewx/weewx/commit/04d1985f7ea83f3db33dba1b012e54ed29c6c937

Thanks for your diligence, Rod!