weewx/weewx #446: WMR300 excessive heartbeat delay with full data logger

poobeth, 2019-09-19:
Upgraded weewx from 3.8.2 (I think) to 3.9.2,  

wmr300: driver version is 0.19rc6+f3+l2

Didn't get any current data and syslog has this: 

> Sep 19 12:01:49 odroid weewx[26051]: wmr300: Excessive heartbeat delay: 90s, restarting
> Sep 19 12:01:49 odroid weewx[26051]: wmr300: No loop data in heartbeat interval,  restarting
> Sep 19 12:01:49 odroid weewx[26051]: wmr300: history buffer at 100.00% (32735)
> ...
> Sep 19 12:03:19 odroid weewx[26051]: wmr300: Excessive heartbeat delay: 89s, restarting
> ...
> Sep 19 12:52:55 odroid weewx[26168]: wmr300: Clearing records since 2019-09-19 12:52:55 > BST (1568893975) (last_index=32734 history_end_index=32735)
> Sep 19 12:52:55 odroid weewx[26168]: wmr300: get history complete: count=0 last_index=32734 history_end_index=32735
> Sep 19 12:54:25 odroid weewx[26168]: wmr300: Excessive heartbeat delay: 89s, restarting

Left it for nearly an hour and it continued complaining about heartbeat delay.

On a hunch I cleared the data logger from the console and restarted weewx. No more excessive heatbeat delay and I'm getting data.

CameronD73, 2020-01-02:
I should probably comment here since I think some of that code started as mine.
It is not clear at this stage whether the problem is relating to the 100% full buffer, or that was just incidental. Running with some older versions of the driver would simply fill the history buffer, but as far as I know the system would still report loop data to other users.

I don't recall ever testing the history clear operation at 100% full, so there might be an off-by-one issue, or some other quirk in the operation of the WMR300 - and there are plenty of quirks around there.

Maybe the initial attempt to automatically clear the buffer has a bug that caused the loop packet failures.
I can have a look at it at some stage by disabling the automatic clear and waiting for the buffer to fill - I can at least check that the current code clears a full buffer.

Alternatively, :
-  I have in the past experienced extensive heartbeat failures during checks of the RAID drives - at which time genlooppackets was just never given a chance to return anything
- The WMR300 console itself has periods when it just stops sending loop packets - never solved that, but they  can be around a minute, occasionally longer. Whether these can be caused by wireless dropouts between the console and the external station I don't know.

CameronD73, 2020-01-23:
I can confirm the bug still exists in WMR300 driver v0.32 - it can clear history from less than 100%, but if it gets to 100% then the console basically locks up and refuses to talk any more until the "restarting" packet . It then sends 1 or 2 loop packets, at which time the code decides it is time to try to clear history again ... and around we go.

The good news is that the bug is not present in the original code I developed using libusb1. The history read code got slightly simplified in the transfer and it may be that this was one of the obscure cases where the more complicated code was required.

I shall transfer the current working code to the WMR300 driver and test it.  This will take another 3 weeks to refill the history buffer

closed 2020-02-19 by tkeffer