weewx/weewx #450: Discarding out-of-order archive packets causes loss of data

tgomas, 2019-10-07:
On an Oregon WMR200, after the station being disconnected for one week, archive packets are discarded because some newer packets were received before.

Not sure if real time packets were received before all archive ones or if the archive packets were not ordered by their creation date.

tgomas, 2019-10-07:
Here are the logs of packets being discarded:

```
oct. 07 13:43:11 raspberrypi weewx[31215]: wmr200: MainThread: I Using pc time adjusting archive record time by -26 sec 2019-10-01 16:39:00 CEST (1569940740) => 2019-10-01 16:38:34 CEST (1569940714)
oct. 07 13:43:11 raspberrypi weewx[31215]: wmr200: MainThread: I genStartup() Discarding received archive record that presented out-of-order; current timestamp:2019-10-01 16:38:34 CEST (1569940714); previous timestamp:2019-10-07 12:36:00 CEST (1570444560)
oct. 07 13:43:12 raspberrypi weewx[31215]: wmr200: MainThread: I genStartup() Still receiving archive packets cnt:0 len:1
oct. 07 13:43:12 raspberrypi weewx[31215]: wmr200: MainThread: I Using pc time adjusting archive record time by -26 sec 2019-10-01 16:40:00 CEST (1569940800) => 2019-10-01 16:39:34 CEST (1569940774)
oct. 07 13:43:12 raspberrypi weewx[31215]: wmr200: MainThread: I genStartup() Discarding received archive record that presented out-of-order; current timestamp:2019-10-01 16:39:34 CEST (1569940774); previous timestamp:2019-10-07 12:36:00 CEST (1570444560)
oct. 07 13:43:12 raspberrypi weewx[31215]: wmr200: MainThread: I genStartup() Still receiving archive packets cnt:0 len:1
oct. 07 13:43:12 raspberrypi weewx[31215]: wmr200: MainThread: I Using pc time adjusting archive record time by -26 sec 2019-10-01 16:41:00 CEST (1569940860) => 2019-10-01 16:40:34 CEST (1569940834)
oct. 07 13:43:12 raspberrypi weewx[31215]: wmr200: MainThread: I genStartup() Discarding received archive record that presented out-of-order; current timestamp:2019-10-01 16:40:34 CEST (1569940834); previous timestamp:2019-10-07 12:36:00 CEST (1570444560)
```

tkeffer, 2019-11-04:
I am not a WMR200 expert, but I think the problem is that the only way to calculate the archive interval is to take the difference in timestamps between the current archive record and the last one. So, if they present out-of-order, this can't be done.

I think we'll have to live with this one.

closed 2019-11-04 by tkeffer

garberw, 2023-04-26:
here is perhaps the same problem
[weewx-sdr driver modified for rtl-sdr plus extra ttyACM1 input; getting gaps](https://groups.google.com/g/weewx-user/c/2pRcvkLAzZc)