weewx/weewx #453: Acurite display 01036A1 not found automatically

seanocaster123, 2019-10-13:
Hi,

The Acurite 01036A1 is connected to the pi via USB and it works.

Then i initiate a sudo reboot to the pi which restarts. When it's back up i ssh into it and run `sudo service weewx status` only to find Errno 110 due to not finding the device via USB.

Then i change USB Mode from wherever it is (either 3 or 4) to the the other (either 3 or 4), run `sudo service weewx restart` and then `sudo service weewx status` and it is all working again.

Note: Acurite 01036A1 remaings powered on at all times.

Any ideas how i can fix this so that weewx just works after the pi has been rebooted ?

tkeffer, 2019-10-13:
From your limited description, it's not clear whether this is a bug nor not. It's possible the USB system is not ready when WeeWX starts up. It would help if you could include the log.

seanocaster123, 2019-10-13:
Thanks for the reply.

For some random reason, i removed the USB mouse and USB keyboard that were connected still from the initial setup of the RPi. That means only the acurite display is connected via usb.

Now when i reboot the RPi, weewx starts up and finds the display every time. Did it 4 or 5 times and worked every time. Could it be that simple? That the driver was getting confused with all the other USB devices plugged in?

matthewwall, 2019-10-13:
weewx is not confused.  it searches for acurite stations by product/vendor codes, not by mapping to /dev/ttyXXX (as is common for usb-serial devices)

your usb subsystem was not fully initialized when weewx started looking for the station.  the other devices that were plugged into the bus probably caused usb initialization to be slower (the log file would indicate this).

please see the suggestions in the google groups topic:

https://groups.google.com/forum/#!topic/weewx-user/0_x5JuAyfVU

closed 2019-10-13 by matthewwall

seanocaster123, 2019-10-13:
Thanks very much for the help. I'm trying my best to understand the command line but it is undeniably new to me.