weewx/weewx #454: Forecasting extension KeyError: 'U'

m90att, 2019-10-21:
After installing the forecasting extension I am seeing the following error in the syslog.

`Oct 21 11:05:31 weatherpi weewx[5977]: cheetahgenerator: Generate failed with exception '<type 'exceptions.KeyError'>'
Oct 21 11:05:31 weatherpi weewx[5977]: cheetahgenerator: **** Ignoring template /etc/weewx/skins/exfoliation/forecast.html.tmpl
Oct 21 11:05:31 weatherpi weewx[5977]: cheetahgenerator: **** Reason: 'U'
Oct 21 11:05:31 weatherpi weewx[5977]: ****  Traceback (most recent call last):
Oct 21 11:05:31 weatherpi weewx[5977]: ****    File "/usr/share/weewx/weewx/cheetahgenerator.py", line 332, in generate
Oct 21 11:05:31 weatherpi weewx[5977]: ****      fd.write(str(compiled_template))
Oct 21 11:05:31 weatherpi weewx[5977]: ****    File "/usr/lib/python2.7/dist-packages/Cheetah/Template.py", line 1045, in __str__
Oct 21 11:05:31 weatherpi weewx[5977]: ****      rc = getattr(self, mainMethName)()
Oct 21 11:05:31 weatherpi weewx[5977]: ****    File "_etc_weewx_skins_exfoliation_forecast_html_tmpl.py", line 336, in respond
Oct 21 11:05:31 weatherpi weewx[5977]: ****    File "_etc_weewx_skins_exfoliation_forecast_html_tmpl.py", line 113, in __errorCatcher7
Oct 21 11:05:31 weatherpi weewx[5977]: ****    File "<string>", line 1, in <module>
Oct 21 11:05:31 weatherpi weewx[5977]: ****  KeyError: 'U'
`
@matthewwall fyi

Apologies if I'm adding this to the wrong place.  I am new round here!

Thanks,

Matt

tkeffer, 2019-10-21:
Hello, Matt
The forecasting extension is not actually part of weewx. Could you please post your problem on [weewx-user](https://groups.google.com/forum/#!forum/weewx-user)? Thanks!

closed 2019-10-21 by tkeffer

m90att, 2019-10-21:
Will do.  Thanks