weewx/weewx #455: Bresser 6in1

f4gqk, 2019-10-22:
Hello,

I bought this weather station:

[https://www.bresser.de/en/Weather-Time/Weather-Center/BRESSER-PC-Weather-station-with-6-in-1-outdoor-sensor.html](url)

What is the driver Weewx for the weather station ?

Thanks

Sebastien

tkeffer, 2019-10-22:
Hello,

It's a Fine Offset WH1080 clone. Use the Fine Offset driver.

In the future: this issues list is for bug tracking. Please direct your general questions to [weewx-user](https://groups.google.com/forum/#!forum/weewx-user).

closed 2019-10-22 by tkeffer

gus70, 2020-03-01:
The Bresser brand is not a Fine Offset clone. Totally different manufacturer.

gus70, 2020-03-01:
Ambient and Froggit sell not only Fine Offset but they also sell this Bresser stuff with their brand on it. But if it looks like a Bresser but instead is from Ambient or Froggit then it isn't Fine Offset even though it is branded Ambient or Froggit. Yes it is confusing.

tkeffer, 2020-03-01:
Well, it uses the exact same vendor and product ID as the Fine Offset stations: 0x1941, 0x8021.

gus70, 2020-03-01:
The same product ID,  as in what shows up in Device Manager in Windows is because these Chinese manufacturers use the same parts bin building blocks. But then they are differentiated in firmware and how they actually send data.

gus70, 2020-03-01:
Here is another way to look at it. Fine Offset does not sell to the public, they are the manufacturer. Fine Offset lets any company rebrand their stuff. These resellers often then sell the Fine Offset stuff but with their brand on it. These resellers are not limited to just selling the Fine Offset stuff. They then go to Bresser and UCTECH which are yet two other manufacturers of totally different stuff and these resellers then sell stuff from more than one manufacturer. This stuff from these different manufacturers is not compatible with each other. You can only pair up same with same, as in sensors and consoles and software. 

Ecowitt at first may seem just like any of the other resellers because they sell Fine Offset stuff to consumers. But Ecowitt is not the same as Ambient and Froggit and Misol...etc. This is because Ecowitt only sells Fine Offset stuff. Because Ecowitt is nothing more than a subsidiary of Fine Offset created to sell to consumers. Therefore if a model doesn't show up on Ecowitt's website then it isn't a Fine Offset station nor is it compatible with anything Fine Offset. 

For that matter just simply looking at the Bresser stuff, you'll notice that it also doesn't show up on Fine Offset's website FOSHK.com.