weewx/weewx #457: Hex RGB colours wrong way around i.e. BGR

Zanderama, 2019-10-23:
Hi, 

Configuring colors for day_night (and possibly others) shows that the Hex 'RGB' is the wrong way around.  e.g. 0xFF0000 should be Red but shows up Blue, whereas 0x0000FF should be Blue but shows up Red.

Of course, grey colours and green don't matter so much.

I believe this is on weewx 3.9.2 on a reasonably recent raspberry pi.

Cheers

Alex

Zanderama, 2019-10-23:
In addition, this may explain why I get a 'rainbow' effect transitioning from a simple light blue to grey as part of the day_night chart effect - it could be the colours getting confused perhaps?

[image]

gjr80, 2019-10-23:
Did you read the comments in the [ImageGenerator] stanza of skin.conf? 
```
    # Colors can be specified any of three ways:
    #   1. Notation 0xBBGGRR;
    #   2. Notation #RRGGBB; or
    #   3. Using an English name, such as 'yellow', or 'blue'.
    # So, 0xff0000, #0000ff, or 'blue' would all specify a pure blue color.
```

I think you will find this is by design and has something to do with colours accepted by PIL.

Zanderama, 2019-10-23:
Thanks! I was just reading through the weewx code too - and yes, seems to be something to do with PIL's preferred format.  I didn't spot explanation in skin.conf - but no mention of BBGGRR in HTML documentation I believe.  Would be better to include "#RRGGBB examples in HTML docs and some explanation ideally.

However, still doesn't explain the rather unusual rainbow banding for day_night?  Would have thought that would just go from light-blue to grey-blue to grey rather than via bright cyan, green, yellow and red somehow?

gjr80, 2019-10-23:
> I didn't spot explanation in skin.conf - but no mention of BBGGRR in HTML documentation I believe. Would be better to include "#RRGGBB examples in HTML docs and some explanation ideally.

I think I can guess Tom’s thoughts - something along the lines of ‘PRs are always welcome’ :)

> However, still doesn't explain the rather unusual rainbow banding for day_night? Would have thought that would just go from light-blue to grey-blue to grey rather than via bright cyan, green, yellow and red somehow?

Not something I have ever looked at closely, but if you are delving into the code you will finder the underlying blend routines at the bottom of `bin/weeplot/genplot.py`

Zanderama, 2019-10-23:
LOL, I've added a pull request for some documentation to try and help today he seemed to like :)  I'll see if I can help with colours a bit too.

Does seem there might be something 'off' with the HSL blending for day_night.  HSL blending looks a lot prettier on this example page, using identical start-end colours.  I'll try to see what's happening.

http://www.workwithcolor.com/mm-color-blender-01.htm?cp=AED9FF&ch=208-100-84&cp2=999999&ch2=360-0-60&im=6&cm=2

[image]

Zanderama, 2019-10-23:
Hi, I've improved blend() with HSL colour wheel direction and grey detection.  Hopefully looks better now :)  Pull request submitted.

* Uses shortest travel distance around colour wheel to minimize extra colours in blend.
* Detects grey colours to minimize travel around HSL colour wheel
* Fixes math to use 255.0 rather than 256.0

Example image before :-

[image]

Example image using shortest distance :-

[image]

Example image using grey detection and shortest distance (pretty similar to a simple RGB blend) :-

[image]

Zanderama, 2019-10-25:
Hi, 

As pretty much discussed, I've cancelled pull request #459 

https://github.com/weewx/weewx/pull/459

And replaced all 0xbbggrr .conf and .htm content with "#RRGGBB" equivalents in #460 

https://github.com/weewx/weewx/pull/460

Hope that helps everyone with their colors/colours ;)

Zanderama, 2019-10-25:
Colors and blends all fixed now, so I'll close this issue thanks :)

closed 2019-10-25 by Zanderama