weewx/weewx #461: Help to hack driver for OurWeather station

Strykar, 2019-10-27:
The station is an ESP based custom SoC - https://shop.switchdoc.com/products/ourweather-complete-weather-kit. Upon searching for weewx support I came across this example on the `weewx-user` newsgroup to poll the JSON interface via HTTP on https://groups.google.com/d/msg/weewx-user/x1P9BX6_blQ/oFNBLPmpAAAJ

This is what my JSON looks like at http://192.168.1.9/FullDataString:
```
{"FullDataString": "25.60,89.30,29.42,101926.00,592.16,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,1,2019-10-17 18:39:19,NIBM,0,-1,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,V:1,WXLMB ,0,,,0,,,0", "id": "1", "name": "OurWeather", "hardware": "esp8266", "connected": true}
```

This station supports Air Quality and Lightning detection sensors which I've currently mapped (to ignore) as `extraTemp` and `soilTemp` for now.
The trouble from the JSON is multi-part:
1. After `part14` which is `dateTime`, the user-configured Station Name is sent for which I don't see an option to specify in https://github.com/weewx/weewx/blob/master/bin/schemas/wview.py
2.  Same issue for these bits towards the end of the JSON: `V:1,WXLMB ,0,,,0,,,0", "id": "1", "name": "OurWeather", "hardware": "esp8266", "connected": true` which I'm happy to ignore as of now.

Here's the output of me trying to troubleshoot it:
```
$ PYTHONPATH=/home/weewx/bin python2 /home/weewx/bin/user/ourweather.py
Traceback (most recent call last):
  File "/home/weewx/bin/user/ourweather.py", line 65, in <module>
    for packet in driver.genLoopPackets():
  File "/home/weewx/bin/user/ourweather.py", line 23, in genLoopPackets
    {"FullDataString": "25.60,89.30,29.42,101926.00,592.16,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,1,2019-10-17 18:39:19,NIBM,0,-1,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,V:1,WXLMB ,0,,,0,,,0", "id": "1", "name": "OurWeather", "hardware": "esp8266", "connected": true}
NameError: global name 'true' is not defined
```
How do I ignore sections of the JSON?
This is what my current `ourweather.py` looks like, any pointers are appreciated - https://gist.github.com/Strykar/bd7c77c1229af6c8d5bd7ff551a5c541

P.S. I don't know Python, just some Bash.

tkeffer, 2019-10-27:
Your immediate problem is that in Python, the keyword is `True`, not `true`. But, it's not clear to me what the dictionary on line 23 is doing at all.

In any case, this issues list is for bug tracking, not for general support questions. Those should be directed to [weewx-user](https://groups.google.com/forum/#!forum/weewx-user).

closed 2019-10-27 by tkeffer

Strykar, 2019-10-27:
@tkeffer Alright I will post there, thank you.