weewx/weewx #464: Bogus tweet line contains the formatting in some occasional cases

cryptax, 2019-11-01:
Hi, 

I am using **weewx 3.9.2** with T**witter extension 0.12**. In weewx.conf, my format is the following:

```       
[[Twitter]]
   ...
   format = {dateTime:%H:%M} Temp: {outTemp:%.1f} °C; Humidité: {outHumidity:%.0f}%; Pression: {barometer:%.0f} hPa; Pluie 1h: {rain:%.1f} mm; Vent: {windSpeed:%.1f} km/h; Dir: {windDir:%03.0f}
```

This works quite well, and I generally get nice tweets e.g. `14:40 Temp: 19.3 °C; Humidité: 64%; Pression: 1013 hPa; Pluie 1h: 0.0 mm; Vent: 2.7 km/h; Dir: 182`

However, **occasionally, I get bogus tweets**: `22:10 Temp: 14.4 °C; Humidité: 81%; Pression: 1019 hPa; Pluie 1h: 0.0 mm; Vent: {windSpeed:%.1f} km/h; Dir: {windDir:%03.0f}`, **as you can see it prints the formatting for wind speed and wind dir, instead of formatting it.**

My intuition is that this occurs when for some reason, windSpeed and WindDir are not defined, or null, or something like that. Perhaps this test should be added to the twitter python script so that this error does not happen (I'm not totally sure where the error lies here).

Thanks

matthewwall, 2019-11-01:
hi!  this space is for reporting bugs.  your request it better suited to the weewx user group - please post on that forum so that others can benefit from whatever answers you receive.

m

closed 2019-11-01 by matthewwall

matthewwall, 2019-11-02:
btw, the twitter extension leaves the variable name unreplaced (in this case, 'windSpeed' and 'windDir') if the variable does not exist in the packet.  when you post to the weewx user list, please report the type of weather station.  that will help confirm the cause of the behavior.