weewx/weewx #473: CSV imports will not accept a `usUnits` mapping

gjr80, 2019-12-19:
Source data unit information for a CSV import can be specified either by specifying individual units against each source field in the the field map or by specifying a mapping to the WeeWX `usUnits` field. At present the individual field mapping approach works but mapping to WeeWX field `usUnits` does not. [This thread](https://groups.google.com/d/topic/weewx-user/LzPOrPMaqCo/discussion) refers.

Issue #425 was raised to address the `wee_import` CSV field map instructions.

This issue has been raised to fix the broken code.

tkeffer, 2020-01-02:
Gary, how are we doing on this issue? Does it affect V4?

gjr80, 2020-01-03:
> Gary, how are we doing on this issue? Does it affect V4?

I haven't done anything further with this, though I don't believe it is an impediment in releasing 4.0. `wee_import` will fail if a `usUnits` mapping is used though there is a workaround through including mappings for all imported fields. In anticipation of not being able to address this issue in time for the 4.0 release I included words in the 4.0 `wee_import` documentation to the effect that for a CSV import the mapping of field `usUnits` is not currently supported.

I am also not sure the target audience here will be that big, I think the only folks who would do a CSV import with a `usUnits` mapping would be WeeWX or potentially wview users who have CSV exported their data and now wish to import that data. Again the workaround mentioned above will allow the data to be imported.

gjr80, 2020-01-03:
Well this is sort of embarrassing, after much head scratching, what if'ing and trying to get the logic straight in my head I decided I needed to sit down, re-create the issue and work through the steps logically to find the cause. Well using the OPs config and data file and it just works, well at least on the current development branch and under python3. Perhaps there was an issue that was somehow fixed in the calc-missing work, not sure and too hard to find out now.  I did find an error that would cause a CSV import with a `usUnits` mapping to fail if there were multiple different `usUnits` values in the source data. However, that has been fixed at commit 46873169fe7baa37b4c7dc65c2e21ea5115ae068. I just need to confirm python2 works as well and go back and fix the documentation.

I trust that since I have not seen the merge from development into master this will still make v4.

gjr80, 2020-01-03:
python2 is fine. Just the `Utilities Guide` now.

gjr80, 2020-01-03:
Documentation done, thanks for the kick in the pants on this @tkeffer, turned out to be easier than I expected.

closed 2020-01-03 by gjr80

tkeffer, 2020-01-03:
Thanks, Gary! Just trying to clean up as many issues as possible!