weewx/weewx #482: Python3

kskenyon, 2019-12-29:
I tried converting this to Python3 to run along with weewx4.0.0b5 which I have running.

I got stuck on importing the weewx python3 module.  I know it's there but running python3 mqtt.py doesn't find it.  I fixed several other issues related to python3 but not this one.  Any help? 

Kevin

python3 mqtt.py 
Traceback (most recent call last):
  File "mqtt.py", line 80, in <module>
    import weewx
ModuleNotFoundError: No module named 'weewx'

tkeffer, 2019-12-29:
This is a PYTHONPATH problem. Try:

```shell
cd /home/weewx
PYTHONPATH=bin python3 bin/drivers/mqtt.py
```

Incidentally, the issues list is for bug tracking. For support questions on beta versions use [weewx-development](https://groups.google.com/forum/#!forum/weewx-development).

closed 2019-12-29 by tkeffer

kskenyon, 2020-01-04:
Thanks, all sorted.