weewx/weewx #491: Allows StdWXCalculate to calculate rain from total rain

tkeffer, 2020-01-15:
The goal here is to make "total rain" a first class observation type. User would be able to get `rain` from, say, `dayRain`.

With the introduction of the aggregate type `diff` in V4.0, it should also be possible to use `dayRain` in plots.

NB: It is way too late to squeeze this in V4.0. It will have to wait until V4.1.

matthewwall, 2020-05-05:
more generally:

* get a delta value from a total value (requires state - need the last value - and wraparound detection)
* get a total value from a delta value (requires the definition of a timespan)

you definitely want these at the reporting level, where you display cumulative values with incremental values, for example.

but are they necessary at the calculation level, where you insert the result into a packet, and thus possibly into the data storage?

tkeffer, 2020-05-05:
I'm thinking of the former, more than the latter.

I am seeing drivers calculate rain from rain totals so often, that it struck me it's a lot like having drivers calculate `dewpoint`. Getting `rain` from a rain total makes it a derived quantity.

I should also add that, if for some reason your driver does not emit `rain`, and you really don't want it in your database, you either leave it out of `[[Calculations]]`, or set its option to `None`. Or, not include it in the database schema.

The new xtypes facility also has a `diff` aggregate, which would allow you to use `rain` in the reporting layer even if all you have is rain totals in the database.

gjr80, 2020-05-14:
Just trying to understand what is required here, if I understand correctly it is as simple as (1) creating a 'rain' xtype [let's call it `RainPourer` - not as good as `PressureCooker` but on par with `RainRater` :)] that calculates `rain` from a cumulative rain field and (2) adding `rain` to `StdWXCalculate` via the new `RainPourer` xtype. State would be stored in the `RainPourer` xtype.

A couple of questions. 
- Is the intent for the chosen cumulative rain field (1) to be user specified (2) chosen from a list of likely suspects (eg `dayRain`, `monthRain`, `yearRain` etc) or (3) a single hardwired field ?
- the reference to using `dayRain` in plots using `diff`, understand this is a done deal and already implemented, ie nothing more to do ?

tkeffer, 2020-05-14:
You've got the right idea. 

1. The user should be able to pick the source total (`dayRain`, etc.). This is because different devices emit different totals, so we have to be flexible.
2. I probably should not have mentioned `diff`. It was just a thought that given a field like `totalRain`, we can do plots of `rain`. But, that doesn't take away from the need to treat `rain` as a derived quantity. 

Personally, I like the name `rainMaker` :-)

tkeffer, 2020-05-14:
Oh, one other thought: right now we have scalar options such as `wind_height` and `rain_period`, as well as a separate `[[algorithms]]` subsection. Not only does that not follow the pattern used by other services, it also flattens what should be a hierarchical structure.

Think about a structure closer to:

```ini
[StdWXDerive]
  [[ET]]
    wind_height = 10, meter
    source = software
  [[rainRate]]
    rain_period = 900
    source = prefer_hardware
  [[rain]]
    total = yearRain
    source = prefer_hardware
```

and so on. It will make it easier to add future options.

We would allow a `[StdWXCalculate]` section for backwards compatibility. It would basically fill out the structure I outlined above, to be overridden by a `[StdWXDerive]`.

I'm not wedded to this idea, but I am getting concerned about how options are getting separated from their algorithms. Just something to think about.

tkeffer, 2020-05-14:
Still another postscript: this should definitely be off the `development` branch, with its own branch name.

gjr80, 2020-05-15:
> 2. I probably should not have mentioned `diff`. It was just a thought that given a field like `totalRain`, we can do plots of `rain`. But, that doesn't take away from the need to treat  `rain` as a derived quantity.

OK, spent a good while trying to work out if I was missing some basic connection between `diff` and getting `rain` from `dayRain`. Will set `diff` aside.

> Personally, I like the name `rainMaker` :-)

Well what can I say, outdone again, but in my defence I am an engineer not an artist so catchy names are not my strong point :)

> We would allow a `[StdWXCalculate]` section for backwards compatibility. It would basically fill out the structure I outlined above, to be overridden by a `[StdWXDerive]`.

So I can add a new service `StdWXDerive` to the task, with backwards compatibility with `StdWXCalculate` or do you want a bit more development of the concept first? I am happy to go ahead and implement it and see where it takes us; you and Matthew do the concepts best, I am still stuck at coming up with catchy names :)

> Still another postscript: this should definitely be off the `development` branch, with its own branch name.

Yep, that one I had worked out.

tkeffer, 2020-05-15:
> So I can add a new service StdWXDerive to the task, with backwards compatibility with StdWXCalculate or do you want a bit more development of the concept first? 

We definitely need to do some experimentation. One option is to stay with `StdWXCalculate`, just because of all the `weewx.conf` files out there that specify it as a service, but allow it to read two different `[StdWXCalculate]` stanzas: the "old" style, as well as the new style I outlined above. It would need some kind of heuristic to determine the difference. Perhaps the presence of stanzas other than `[[Calculations]]`?

gjr80, 2020-05-15:
How about I implement a revised/restructured `StdWXCalculate` leaving aside support for the 'old' stanza format for now (but keeping it in mind as we will need to support both 'new' and 'old') and we can see what it looks like. If we don't like the looks of it we can tread a different path.

tkeffer, 2020-05-15:
Perfect!

gjr80, 2020-05-16:
That was pretty easy. Have:
- refactored `StdWXCalculate` to use the new `[StdWXCalculate]` structure
- added field `rain` as a derived obs
- `wind_height` can now be specified using any (altitude) unit

As an aside I also extended the simulator to be able to emit cumulative rainfall obs.

A couple of points. 
- The restructured `[StdWXCalculate]' means we need to check for existence of a derived obs before obtaining its settings. It seemed to flow from this that if there was no `[[rain]]` or no `[[rainRate]]` then there was no point in instantiating a `RainMaker` or a `RainRater` object respectively. Not sure if I have missed some other requirement that means these must be instantiated. Given the extra complexity that woudl be required in the code I saw no point in doing the same for `PressureCooker` and `WXXTypes`.
- Did not see any reason to change the the `StdWXCalculate` startup log output to reflect the revised `[StdWXCalculate]` structure.

tkeffer, 2020-05-16:
> It seemed to flow from this that if there was no [[rain]] or no [[rainRate]] then there was no point in instantiating a RainMaker or a RainRater object respectively.

Yes, that's right. I'm trying to get away from "special" types that are hardwired into the code. So, if a derived type isn't mentioned in `[StdWXCalculate]`, then it should not be created, nor should its accompanying machinery.

Is the code on a branch somewhere?

gjr80, 2020-05-16:
> Is the code on a branch somewhere?

Yep, branch `issue_491`.

gjr80, 2020-05-17:
> Yes, that's right. I'm trying to get away from "special" types that are hardwired into the code. So, if a derived type isn't mentioned in `[StdWXCalculate]`, then it should not be created, nor [...]

How far do you want to take this, can give `PressureCooker` and `WXXTypes` the same treatment but the if statement is going to get pretty long for `WXXTypes`.

tkeffer, 2020-05-17:
I think this is fine. Yes, these types are treated as "special," in that they appear in the code, but last winter I looked mighty hard at avoiding that, and didn't come up with anything I liked better. It was looking like each one would need a service, or would need to specify a "driver", in order to make sure it got instantiated and used. I didn't like that idea.

So, I came to think of them as a "package" that calculates weather-related types --- after all, the service is called `StdWXCalculate`. If the user is not using WeeWX in a weather-related application, s/he can always remove the service and put in something else. Indeed, that's what I do for my power monitoring application.

One note: none of these types should be instantiated by default. They have to be specified in `weewx.conf`. Note that in the old-style default configuration, they are all commented out. 

Now what's missing is a shim that checks `[StdWXCalculate]` for a `[[Calculations]]` section. If it finds one, it translates old-speak into new-speak. 

Additionally, we could have the upgrade process massage old-speak `weewx.conf` files into new-speak. But, even if we do this, we should be able to detect old-speak at runtime.

gjr80, 2020-05-17:
> One note: none of these types should be instantiated by default. They have to be specified in `weewx.conf`. Note that in the old-style default configuration, they are all commented out.

Yep, noted, they are only in the default at the moment as a hangover from development/testing.

> Now what's missing is a shim that checks `[StdWXCalculate]` for a `[[Calculations]]` section. If it finds one, it translates old-speak into new-speak.

Yes, that is the next part.
 
> Additionally, we could have the upgrade process massage old-speak `weewx.conf` files into new-speak. But, even if we do this, we should be able to detect old-speak at runtime.

Thought about this too but have not dealt with the upgrade process and modifying `weewx.conf` before. Will get the old-to-new translation happening first and then see what happens.

tkeffer, 2020-05-17:
Let's see how we like this new style before investing much more energy. 

Have you had a chance to try it? What do you think?

Thanks, Gary!

gjr80, 2020-05-17:
> Let's see how we like this new style before investing much more energy.

Will do the translation and stop there for a 'review', after all its just a bit of dictionary juggling. I guess Matthew has been silent on this latest development so his views would be good to hear.

> Have you had a chance to try it? What do you think?

Have only used on a VM during development. I guess there are two parts to this, allowing `rain` to be derived and restructuring the config stanza. At present none of my in-use machines will make use of derived `rain`, though the vast majority of drivers that are being written now natively use/emit a cumulative rain value; guess the simplification and standardisation of calculating rain for such drivers would be a good thing.

Config stanza wise, it does make more sense having everything for a derived obs in one place. I don't think we have payed too high a price for that. I do feel though that `ignore_zero_wind` appears a bit of an orphan, thought it would be better under `[[windDir]]` or something but that does not work.

Thinking aloud, is there any value to generalising `RainMaker` a step further so that a user can easily add a new derived 'delta' obs by simply adding a `[[new_obs]]` stanza to `[StdWXCalculate]`. Am thinking of my solar driver where energy is analogous to rain, I currently take a cumulative energy value from the inverter and calculate a delta value in the driver. If I want to move that out of the driver I will need to write my own `EnergyMaker` xtype and then write some code to add it to the list of xtypes, might as well leave it in the driver in that case. Such an arrangement may be of use for that fellow who was discussing multiple stations recently (in weewx-development I think).

tkeffer, 2020-05-17:
Hmm, great idea! That suggests that perhaps the organization should be:

```ini
[StdWXCalculate]
  [[DeltaMaker]]
    [[[rain]]]
      total = totalRain
      source = prefer_hardware
```

or even
```ini
  [DeltaMaker]
    [[rain]]
      total = totalRain
      source = prefer_hardware
```
but now we're sliding towards making each facility like `DeltaMaker` be its own WeeWX service.

matthewwall, 2020-05-17:
maybe i'm too late to the game, but why not make the 'delta maker' a property of the observation?

this would be how to you specify a delta, and how you specify a cumulative (kind of the inverse of a delta).
```
[StdWXCalculate]
    [[rain]]
        # create observation 'rain' that is the delta between 'rain_total' observations
        type = delta
        observation_source = rain_total
        source = prefer_hardware
    [[snow_total]]
        # create observation 'snow_total' that is the cumulative of 'snow' observations
        type = cumulative
        observation_source = snow # the 'snow' readings are deltas
        source = prefer_software
```

are there other types?

i think there is a formal name for delta/cumulative...

matthewwall, 2020-05-17:
i started generalizing elements in StdWXCalculate so that you could specify a function, but that leaves you with another implementation of what is already in [StdCalibrate].  i think that there should be functions in StdWXCalculate, so that, for example, you can say "calculate max solar radiation using formula x".

tkeffer, 2020-05-17:
> i think that there should be functions in StdWXCalculate, so that, for example, you can say "calculate max solar radiation using formula x".

That's the intention of the XTypes system. It allows new functions to be introduced that can create new derived types. So, it solves that problem, but creates two new problems:
1. If the intention is to get the type into the record, then someone has to invoke the new function.
2. Many derived types need state (such as the last rainfall total). The XTypes system allows this because new functions are subclasses of `weewx.xtypes.XType`, which can hold the state.

It's the job of `StdWXCalculate` to declare which types to put in the record, and to create the instance of `XType`, thus solving these two problems.

> maybe i'm too late to the game, but why not make the 'delta maker' a property of the observation?

I like this idea, but given the above, I'm not sure it's going to be possible. Maybe Gary can experiment and report back...?

tkeffer, 2020-05-17:
Had a long walk and a chance to think about this. The trouble with my original choice of a hierarchy by type
```ini
[StdWXCalculate]
  [[ET]]
    wind_height = 10, meter
    source = software
  [[rainRate]]
    rain_period = 900
    source = prefer_hardware
```
is that either `StdWXCalculate` has to know about all the types, or each type must list what facility can calculate it. So, if we want to make this extensible, we should list by facility. It's up to each facility to interpret its stanza. Something like:
```ini
[StdWXCalculate]
  [[Delta]]
    [[[rain]]]
      observation_source = rain_total
      source = prefer_hardware
    [[[snow]]]
      observatoin_source = snow_total
      source = prefer_hardware
  [[Cumulative]]
    [[[total_hail]]]
      observation_source = hail
      source = prefer_hardware
  [[PressureCooker]]
    altitude = 400, meter
    [[[QNH]]]
      source = prefer_hardware
      algorithm = aaASOS
```

gjr80, 2020-05-17:
The hierarchy makes sense though I think the likes of `[[PressureCooker]]` and (I presume) `[[WXXTypes]]` would need to change, not too intuitive. But we know names is not my strong point :).

gjr80, 2020-07-08:
I have come back to this. As things stand at present I have the `issue_491_generalised` branch that supports the 
```
[StdWXCalculate]
  [[Delta]]
    [[[rain]]]
      ...
    [[[snow]]]
      ...
  [[Cumulative]]
    [[[total_hail]]]
      ...
  [[PressureCooker]]
    [[[QNH]]]
    ...
```
structure. Have implemented all the 'maker' classes. The `Cumulative` class seems to me to be missing an implied parameter that specifies when the cumulative value is reset, eg midnight, 0900 daily, midnight on the last of the month, 0900 on the first of the month.... Not sure how to approach this; a list of key words, a multi-part parameter etc. Starts to get very messy/limiting very quickly.

The `issue_491_generalised` branch is probably a little behind development but I don't think that matters too much just now considering this was a suck it and see branch.

Would appreciate some more guidance.

tkeffer, 2020-08-02:
Sorry for taking so long to get back to you.

I like where you're headed: the results are much more modular. 

How would one extend the set of calculators beyond `Delta`, `Cumulative`, `Rater`, `PressureCooker`, and `WXXTypes`? Right now, the instantiation of the appropriate calculator is hardwired into `WXCalculate`. Instead of the present `xtype_factory()`, could we do something like:

```python
   def xtype_factory(self, svc_dict, section):
        """Factory method to create XType objects to support derived obs."""

        if svc_dict[section]:
            #  Get the class
            klass = weeutil.weeutil.get_object(svc_dict[section])
            # Instantiate
            return klass(svc_dict[section])
```

Of course, this would require the section names get changed from `[Delta]` to `[weewx.wxservices.Delta]`. I'm sure I've left out lots of other details, but hopefully you get the idea.

Which brings me to another thought: it feels like `StdWXCalculate` is starting to replicate a lot of what the engine already does. Totally radically different idea: what if we got rid of `StdWXCalculate` and, instead, broke up what it does into separate services? That is, we would have new services `StdDelta`, `StdCumulative`, etc?

gjr80, 2020-08-03:
> How would one extend the set of calculators beyond `Delta`, `Cumulative`, `Rater`, `PressureCooker`, and `WXXTypes`?

That's a good question and a feature that needs to be included. You've created a bit of a factory fanatic since you introduced me to factory methods in `wee_import`, so I like the sound of using a factory method. Do we really need to use dotted section names? Could we keep the section name as the class name only and push the location detail down, maybe with a default of `weewx.wxservices`?

Separate services is a radical idea. Thinking aloud I guess there is no great overhead in doing that and certainly keeps things modular. I wonder though if we are opening ourselves (or the unsuspecting user) up to having such services spread throughout the services being run and some possibly unexpected results. If we keep everything within `StdWXCalculate` then all the packet/record augmentation happens at once rather than risk some happening before 'service X' and some after. Probably mitigated somewhat by `StdWXCalculate` being on the end of the list of `process_services`.

I guess its a fairly easy jump from where I am now to implement either approach, how about I go down the former and see what that looks like?

tkeffer, 2020-08-03:
Nah. We are, or rather, I am, over-engineering this. Let's stick with what we've got. For one thing, the upgrade would be vastly simpler.

Another way to find the class implementing an extension is to require any extensions declare the location of the implementing class. Something like:

```ini
[StdWXCalculate]

    [[Calculations]]
    
        [[[MyExtension]]]
            class = user.myextension.MyExtension
```
The default for `class` would be `weewx.wxservices.NAME` where `NAME` is the name given in the `weewx.conf` stanza.

gjr80, 2020-08-03:
So are you saying stick with what I have done so far and not make it extensible? Was looking forward to that. I don't know about over-engineering, seems to me that retaining `StdWXCalculate` and making it extensible is not not that much of a stretch and a useful bit of future proofing. Though this is probably something that will not be extended too many times; there are few enough extension authors now and I expect even fewer that would extend `StdWXCalculate`.

I would have thought the upgrade would not be too bad, we are going from a well known, tightly structured arrangement to something less so. Granted it will be easier to do it once only rather than go down one path now and change path later.

tkeffer, 2020-08-03:
Sorry, no, that's not what I meant. I referring to my idea of breaking up `StdWXCalculate` into separate WeeWX services.

Keep going with your idea! Just try to make it more extensible. For example by doing what I outlined in my earlier note.

gjr80, 2020-08-03:
Good, will do, need a break from Ecowitt!

tkeffer, 2020-08-06:
Been mulling this around in my head the last couple of days. I'm concerned that we are inventing yet another way to extend WeeWX, when we already have a rich set of models to choose from. 

I'm thinking the `StdRESTful` model might be appropriate. One base class, many specialized uploaders. Each is run as a separate service. There is one `[StdRESTful]` clause, with many sub-clauses under it, one for each uploader.

This is very similar to what we are trying to do with `StdWXCalculate`.

Let me think this over a bit.

tkeffer, 2020-09-13:
This is what I'm thinking. Introduce a new service, `StdCalculate`, not to be confused with
`StdWXCalculate`, the existing service. The user would use one or the other. `StdCalculate` is
similar to `StdRESTful`, in that it's not actually a WeeWX services *per se*, but rather a collection of
services that make use of some common machinery, and share a section in `weewx.conf`.

The `weewx.conf` stanza would look like:

```ini
[StdCalculate]
    # Source can be overridden by individual calculations
    source = prefer_hardware
    # Data binding can be overridden by individual calculations
    data_binding = wx_binding
    [[Rates]]
        [[[rainRate]]]
            input = rain
            period = 900        # Rain rate window
            retain = 930        # How long to retain rain events. Should be >= rain_period + archive_delay
    [[Pressures]]
        altitude = None            # None means get from station information
        max_delta_12h = 1800       # When looking up a temperature in the past, how close does the time have to be?
        [[[pressure]]]
        [[[altimeter]]]
            algorithm = aaASOS
        [[[barometer]]]
    [[Deltas]]
        [[[rain]]]
            input = dayRain
    [[Wind]]
        ignore_zero_wind = True     # If windSpeed is zero, should windDir be set to None?
    [[WX]]
        [[[appTemp]]]
        [[[cloudbase]]]
        [[[dewpoint]]]
        [[[ET]]]
            et_period = 3600
            wind_height = 2.0   # In meters
        [[[heatindex]]]
        [[[humidex]]]
        [[[inDewpoint]]]
        [[[maxSolarRad]]]
            atc = 0.8       # For solar radiation RS
            nfac = 2        # Atmospheric turbidity (2=clear, 4-5=smoggy)
            algorithm = RS
        [[[windChill]]]
        [[[windrun]]]   
```

In this example, there are five new services, `Rates`, `Pressures`, `Deltas`, `Wind`, and `WX`.
They all have their own substanzas, where pertinent configuration options are specified, using a
hierarchical style, similar to the rest of WeeWX. For example, specifying `source=prefer_hardware`
at the top level makes it apply to everything underneath, but it can be overridden for individual
derived types.

The services `Rates` and `Deltas` are general ones, which can be used to calculate the rate or deltas
of any variable. (NB: there may be complications if the types need access to the database.)

Each of these new services has to instantiated in order to be run by the engine. Analogous to
how`StdRESTful` works, this is done by a new service group, `calc_services`, which is specified
under `[Engine]`:

```ini
[Engine]
    [[Services]]
        prep_services = weewx.engine.StdTimeSynch
        data_services = ,
        process_services = weewx.engine.StdConvert, weewx.engine.StdCalibrate, weewx.engine.StdQC, weewx.wxservices.StdWXCalculate
        calc_services = weewx.calcservices.Rates, weewx.calcservices.Pressures, weewx.calcservices.Deltas, weewx.calcservices.Wind, weewx.calcservices.WX
        archive_services = weewx.engine.StdArchive
        restful_services = weewx.restx.StdStationRegistry, weewx.restx.StdWunderground, weewx.restx.StdPWSweather, weewx.restx.StdCWOP, weewx.restx.StdWOW, weewx.restx.StdAWEKAS
        report_services = weewx.engine.StdPrint, weewx.engine.StdReport
```

Just like `StdRESTful`, it is easy to add new calculations by writing a new service, then adding it
to the list.

I think it's also possible that all these new types can be introduced through the XTypes system, making the types available for other things even if they don't appear in the database schema.

gjr80, 2020-09-14:
Hmm, going to need to get my brain back into 'Calculate' mode. Can see this approach being better from an extensibility viewpoint. 

So ideally one would run just a suite of 'calc' services. You said 'the user would use one or the other' but the example `[Engine]` stanza includes both `StdWXCalculate` and some 'calc' services. Was this this intentional, are there any use cases where one might run `StdWXCalculate` and one or more 'calc' services? I would imagine that the existing `StdWXCalculate` would readily lend itself to being mapped to one or more 'calc' services during an upgrade and `StdWXCalculate` could be retired. I guess the unknown is what any power users may have done with `StdWXCalculate` unbeknown to us.

Do you want me to code this in a new branch? I guess Git saves everything but it might be handy to keep the existing '#491' related branches readily accessible as is.

tkeffer, 2020-09-14:
Oops. That was unintentional. I've edited the post and removed `StdWXCalculate`.

My biggest concern is whether this is a case of the perfect being the enemy of the good. Yes, this solution is probably more elegant and extensible, but we already have something that works. Introducing something new means upgrading users.

gjr80, 2020-09-14:
> Yes, this solution is probably more elegant and extensible, but we already have something that works. Introducing something new means upgrading users.

Can't be any worse than python2 v python3 or view v view_extended can it? Seriously, `StdWXCalculate` is not that sophisticated now so I would have thought we could handle upgrading easily enough.

In any case let me see what I can put together and then I guess we will have 2 or 3 models to choose from.

tkeffer, 2020-10-24:
These commits allow one to put the following in `weewx.conf`:

```ini
    [StdWXCalculate]
        [[Calculations]]
            rain = prefer_hardware
        [[Delta]]
            [[[rain]]]
                input = totalRain
```

In this case, if `rain` is emitted by the hardware, it is used. Otherwise, it will be calculated from differences in `totalRain`.

closed 2020-10-24 by tkeffer

prchal, 2022-11-04:
Thanks, this works, either fileparse.