weewx/weewx #495: WeeImport : UV index without decimals is an issue when importing data from wlk file

FredSau, 2020-01-29:
Hello,

I've installed weewx some days ago (https://www.famille-sauvage.be/weewx/index.html) and would like to import 8 years of history (from my vantage pro 2) by the upload of converted wlk files.

In my first try (missing data from january) I had a problem because I've uploaded UV index as I've received if from wlk, as an integer (in january 30 to 80 values). In fact it is really 0.3 to 0.8.

I'had updated the loaded data by a sql request and changed my "Import Excel sheet" to convert for example 80 to 0.8.

BUT WEEIMPORT want an integer without decimal - as if the error message doesn't say that.

Are you thinking it will be a good correction to do ? Allow decimal value as uv_index ?

It is easy to test. Try with a 0 in the data. No message, import ok. Try  0.0 as uv_index value and weeimport give an errormessage (talking about my firt field (the date)) - to be corrected at the same time ?)

I can send you the files.

Thank for reading me and excuse my poor english.
KR;
Fred.

gjr80, 2020-01-29:
Hi,

Let me see if I can work through your issue. First up I would like to understand what exactly you are importing with `wee_import`. You say you are importing 'converted wlk files', since `wee_import` does not directly support wlk file imports I expect you are using an intermediate format, the most likely choice would be csv format?

If you are importing data that contains UV data with values from 30 to 80 then `wee_import` will certainly reject them if `qc = True` in your import config file. This setting applies the WeeWX quality control limits to the imported data and a UV index of 30 or more would be rejected. This is expected behaviour. It sounds to me like this is a source data export issue and one that you will address before using `wee_import`.

`wee_import` should import integer values or values with decimal places without problem. So 0.0 and 0 should both be imported as 0.0, similarly 1.0 and 1 should both be imported as 1.0 and so on. I just did a trial import of csv data using WeeWX 3.9.2 and `wee_import` handled UV integer fields and fields with decimal places as expected.

Can you please provide a screen capture of your `wee_import` session showing the exact commands entered and the `wee_import` output/errors.

Feel free to post in your native language if you wish, we will work out what you mean.

Gary

FredSau, 2020-01-31:
Hi,

Here is a zip file with some informations.
Hope there will help you.

Fred.
[weewx_Issue_495.zip](https://github.com/weewx/weewx/files/4139794/weewx_Issue_495.zip)

FredSau, 2020-01-31:
Sorry - bad click ...

tkeffer, 2020-02-02:
Gary, do you have a sense yet whether this is a bug?

gjr80, 2020-02-03:
No, no bug but rather two dodgy unicode characters that managed to find their way into the December 2019 csv file and attach themselves to the start of the header row. So the header row appeared like it had a column `DateTimeObs` but it was in fact `\ufeffDateTimeObs` and hence the error.

I thought I could fix it by saving the file via a text editor but no combination of spells in the text editor I was using would remove them. I ended up copying the the header row from the 2nd character onwards, deleting the entire header row, typing the first character (D) and pasting the remaining header row. Saved the file and it imports just fine, following (verbose) output refers:
```
gary@buster47:~$ /home/weewx/bin/wee_import --import-config=/var/tmp/csv.conf --dry-run --verbose
Using WeeWX configuration file /home/weewx/weewx.conf
Starting wee_import...
A CSV import from source file '/var/tmp/Import_2019_12.csv' has been requested.
The following options will be used:
     config=/home/weewx/weewx.conf, import-config=/var/tmp/csv.conf
     source=/var/tmp/Import_2019_12.csv, from=None, to=None
     dry-run=True, calc_missing=True, ignore_invalid_data=True
     tranche=250, interval=1, date/time_string_format=%Y-%m-%d %H:%M:%S
     rain=cumulative, wind_direction=[-360.0, 360.0]
     UV=True, radiation=True
Using database binding 'wx_binding', which is bound to database 'weewx.sdb'
Destination table 'archive' unit system is '0x01' (US).
Missing derived observations will be calculated.
This is a dry run, imported data will not be saved to archive.
Starting dry run import ...
Obtaining raw import data for period 1 ...
The following imported field-to-WeeWX field map will be used:
     source field 'DateTimeObs' --> WeeWX field 'dateTime'
     source field 'Barometre' in units 'hPa' --> WeeWX field 'barometer'
     source field 'TempInt' in units 'degree_C' --> WeeWX field 'inTemp'
     source field 'TempExt' in units 'degree_C' --> WeeWX field 'outTemp'
     source field 'HumidInt' in units 'percent' --> WeeWX field 'inHumidity'
     source field 'HumidExt' in units 'percent' --> WeeWX field 'outHumidity'
     source field 'VitVent' in units 'km_per_hour' --> WeeWX field 'windSpeed'
     source field 'DirVent' in units 'degree_compass' --> WeeWX field 'windDir'
     source field 'RafVent' in units 'km_per_hour' --> WeeWX field 'windGust'
     source field 'DirVentMax' in units 'degree_compass' --> WeeWX field 'windGustDir'
     source field 'Pluie' in units 'mm' --> WeeWX field 'rain'
     source field 'RadSol' in units 'watt_per_meter_squared' --> WeeWX field 'radiation'
     source field 'UV' in units 'uv_index' --> WeeWX field 'UV'
Raw import data read successfully for period 1.
Mapping raw import data for period 1 ...
Mapped 44640 records.
Raw import data mapped successfully for period 1.
44640 records identified for import.
Unique records processed: 44640; Last timestamp: 2020-01-01 00:00:00 AEST (1577800800)
Finished dry run import
44640 records were processed and 44640 unique records would have been imported.
```
[Import_2019_12.txt](https://github.com/weewx/weewx/files/4145784/Import_2019_12.txt)

closed 2020-02-03 by tkeffer

FredSau, 2020-02-07:
Hello,
Some days had passed since our last exchanges.
I've just tried your workaround procedure and for me, the problem is still present.
BUT !
I've found another solution.
I compute all values on my PC (windows, excel), then I extract data from excel in .ods format.
Then (via onedrive) I download the file on RPI4, open it with 'LibreOffice', save it as .csv and ... TADA .... the loading of decembre 2019 is now in progress.
Thanks for your research ... 
Fred.