weewx/weewx #496: wunderground "certificate" problems....

RPeters52, 2020-01-30:
At about 4 pm yesterday (29 Jan 2020) my station (knmalbuq335) went offline at wunderground and has not returned.  It has been on wunderground for over 3 years!  The station is running and connected to the internet; I can upload the weewx html pages to Dropbox and view them with firefox.  I did a reboot!  No change!  So, I looked in the log file (/var/log/syslog) and found the following "certificate" error:

Jan 30 09:20:20 raspberrypi weewx[1002]: *** CertificateError: hostname 'weatherstation.wunderground.com' doesn't match either of '*.prod-pws-ng-546567-997b58a668d15d562a6bed58ea7c5f9e-0000.us-south.containers.appdomain.cloud', 'prod-pws-ng-546567-997b58a668d15d562a6bed58ea7c5f9e-0000.us-south.containers.appdomain.cloud', 'prod-pws-ng-546567.us-south.containers.appdomain.cloud'

Jan 30 09:20:20 raspberrypi weewx[1002]: restx: Wunderground-PWS: Thread exiting. Reason: hostname 'weatherstation.wunderground.com' doesn't match either of '*.prod-pws-ng-546567-997b58a668d15d562a6bed58ea7c5f9e-0000.us-south.containers.appdomain.cloud', 'prod-pws-ng-546567-997b58a668d15d562a6bed58ea7c5f9e-0000.us-south.containers.appdomain.cloud', 'prod-pws-ng-546567.us-south.containers.appdomain.cloud'

So it appears that wunderground has changed something -- do you (the readers) agree?
Any ideas on a fix or .....?

Drop me a note if you think that I can provide and further info!

Thanks!  
Ralph Peters

RPeters52, 2020-01-30:
I am running python 2.7.9 and the latest weewx 3.9.2 on a raspberry pi.  I have made no changes, except "normal" updates for quite a while.  
I searched the web for this sort of error and found:
https://community.pagerduty.com/t/certificateerror-hostname-proxy-doesnt-match-either-of-pagerduty-com-pagerduty-com/1002

Why this should be a problem now is puzzling...

tkeffer, 2020-01-30:
This is a known issue. See issue #417 

Fix is to appear in Version 4, due any day now.

RPeters52, 2020-01-30:
So, for today, to get running can I add?
    retry_certificate = 7200

to weewx.conf near the top ~~line 131 in the section

[StdRestful]
......
  [[Wunderground]]
    retry_certificate = 7200

Thanks!
Ralph

RPeters52, 2020-01-30:
And, of course, stop and restart weewx

tkeffer, 2020-01-30:
> So, for today, to get running can I add?
> retry_certificate = 7200

Unfortunately, this feature is only in V4. If that's what you're running, you should not have to do anything. It's on by default with a retry of 3600 seconds.

RPeters52, 2020-01-31:
.Hello again,

I am trying to figure out my problem.  I am still "offline" even though I have rebooted the computer several times and re-started weewx several times also.

Please correct me if I am wrong!  It seems that your fix (discussed in #413 and will show up in v4 ) will retry instead of the thread failing.  Yes?

If this is the case, then stopping and starting weewx should temporarily fix the problem also?

Am I mis-understanding anything?

THANKS for your time!
Ralph

tkeffer, 2020-01-31:
It doesn't fix anything. It just prevents the WU thread from terminating and, instead, keep retrying forever.

closed 2020-01-31 by tkeffer