weewx/weewx #500: Temp off by approx 4C

hrpunio, 2020-02-02:
Everything works as described in   https://github.com/weewx/weewx/wiki/gw1000-recipe
but the temperature reported to local server is wrong (lower approx 4C)
It is OK at ecowitt.net, at PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/interceptor.py --device=fineoffset-bridge --port 8000 --debug and as reported by WS View.
Why correctly interecepted data are mangled by weewx?
Any suggestions?

--tomek

tkeffer, 2020-02-03:
Hello, Tomasz

This list is for bug tracking for WeeWX. The interceptor driver is not actually part of WeeWX. You will have a better chance of having your question answered at the user's forum, [weewx-user](https://groups.google.com/forum/#!forum/weewx-user).

closed 2020-02-03 by tkeffer

hrpunio, 2020-02-06:
OK but I suspected problem with WeeWX not the driver (as the interceptor extracted valid data). Anyway I modified the config, restart the service, and the readings seems OK now.