weewx/weewx #504: Please add support for wind stats

kacnje, 2020-02-05:
It would be very nice if this skin can show wind gust statistics ... please.

tkeffer, 2020-02-05:
I don't know what skin you are referring to, but it's  easy enough to add yourself. Something like

    <p>The max wind gust today is $day.windGust.max</p>

See the Customizing Guide.

closed 2020-02-05 by tkeffer

kacnje, 2020-02-05:
Oh, sorry, my mistake. I was on xstats website in one tab in the browser and on this page ... 
Is something similar for all time? Like $alltime.windGust.max
I'm a noob, sorry.

tkeffer, 2020-02-05:
This issues list is for bug reporting and tracking. You would be better off asking your questions on [weewx-user](https://groups.google.com/forum/#!forum/weewx-user). That way, others can answer, and learn from the responses.

kacnje, 2020-02-05:
Understand. Thank you for your effort. 

Janez