weewx/weewx #506: Consider adding distutils to Python3 in docs/setup.htm

rezin8, 2020-02-06:
I was installing 4.0.0b11 today using the instructions in docs/setup.conf for Python3 in a RPi, and ran into ModuleNotFoundError: No module named 'distutils.core'. Once I installed the module, the setup ran fine. I would recommend adding that as a step in setup.htm to help out any confusion for other users. Added a screenshot for clarity. 

[image]

tkeffer, 2020-02-06:
@gjr80 discovered this earlier.

What OS are you using, and can you recall whether you used a regular, or minimal, build?

rezin8, 2020-02-06:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

It's Raspbian Lite, which is a minimal build.

tkeffer, 2020-02-06:
We can conclude this is common enough it's worth including in the instructions.

Commit https://github.com/weewx/weewx/commit/aaa3eb065110a56491b8e4d52848efe9700c927b

Note to @matthewwall: this means we will have to add it to `pkg/debian/control`, no?

closed 2020-02-06 by tkeffer