weewx/weewx #508: qos config, possible?

europacafe, 2020-02-14:
I'm not sure there is an option to config the qos level when publishing?

tkeffer, 2020-02-14:
Don't know what you mean. What QOS level? And, what do you mean by "publishing?" Uploading to a RESTful site, or publishing to your own host? Or, something else?

matthewwall, 2020-02-14:
> I'm not sure there is an option to config the qos level when publishing?

i assume that you are referring to the weewx-mqtt extension.  the qos option was added at commit 96d9bfc in version 0.20.

if you need help with that extension (or any other weewx extension), please post to the weewx user group.

if you find a bug in the weewx-mqtt extension, please file an issue at the weewx-mqtt repository:

https://github.com/matthewwall/weewx-mqtt

closed 2020-02-14 by matthewwall

europacafe, 2020-02-14:
Sorry I posted at a wrong place. You are right. I mean the weewx-mqtt extension. I am not sure how to specify qos for a particular topic. For me, I am using topic weewx/loop.